gin66/FastAccelStepper #1: Moving a set distance or angle.

vangalvin, 2020-06-14:
Do you by chance have an example showing how to implament this for distance movement's?

I have a prety smart hybrid driver and a high torque stepper from Novotech. This library seems to work well with that combination although I am trying to split out the encoder signals so I can also get the positioning of the shaft rather then rely on limit switches.

in my case I am using micro 3200 steps on the driver. The project I have only needs to motor to turn one rotation forward and backward however I need to do this as fast as possible. Essentially thats 8.888 steps per degree. 

I would like to be able to vary the velocity and speed of the movement over the 360 degree's of rotation but not quie sure how to implament this based on the examples.

Thanks in advance for any advice.
Andrew

gin66, 2020-06-21:
How is the velocity and speed varying over the 360 degrees ? Will this be always the same ? If so, then I recommend to use the raw version with filling the stepper queue as fast as possible with the static ramp. 

What do you mean with distance movement ? Moving to an absolute position or moving a number of steps ? The high level interface support both:

```
void move(int32_t move);
void moveTo(int32_t position);
```

For raw mode the absolute position can be retrieved, for the case after the queued commands are done:
```
getPositionAfterCommandsCompleted()
```

Based on this (future) information, new commands can be en queued.

Nevertheless, I do not completely understand your use case. Could you please explain bit more ?

gin66, 2020-06-21:
micro 3200 steps on the driver .... 8.888 steps per degree

Sounds like a stepper with 1.8° per step (200 steps per 360°) and 16 micro steps ?

Have you tried the RawAccess.ino example ?

vangalvin, 2020-06-21:
I have had to put the project on hold for a couple of weeks until a new micro arrives. Some.. Most likley me but im not admitting to it, did not tighten up the step wire properly on the controller and when it came out of the hole as luck wild have it somehow it made contact with the 48V line on the power supply. The poor Teensy is no more.

I will delve back in to the RawAccess.

What i was hoping to do is while the motor is going from 0 to 360 (or set distance) deg be able to adjust the speed and velocity. So the motor starts out slow for the and ramps up to full speed to get to the end of the travel then when its there it stops. This puts a bit of a shudder in things with the inertia of the motor but that is kind of what I am after. or in another mode that is reversed so the motor around the 0 (or what ever the minimum limit is set to) is running at full speed and maximum acceleration acusing a bit of a thump as it changes direction but then it gets to the max limit after changing direction this is reduced by a slight ramping of the speed. 

I hope that makes a little more sense. 

I have found that with the teensy I was able to get faster steps so I may be able to increase the length or rotations so as to give me a bit more time but and inirtia but its that little thump caused by the sudden change in direction that I am wanting on one dise of the stroke or the other as well as the ability to change the speed on the fly while the motor is turning.

For the speed if the motor has been told to rotate say 240deg it can start out at say 100rpm and ramp up as the speed control is turned up.

gin66, 2020-06-22:
If you control the speed and acceleration only in order to reach maximum speed, then it is easier to use the high level interface. This contains all the needed functionality.

Current high level implementation uses this formula to calculate the needed steps s to reach max speed v with acceleration a:

         s = 1/2 * v² / a

For your case, 360° and 3200 steps, the max speed is based on 1600 steps for acceleration and deceleration

         v = sqrt(3200 * a)

If the acceleration is set too low, then the maximum speed cannot be reached. Which max speed your stepper/system can reach before it looses steps, you need to try out.

Assume the 360° shall be done in 0,2s (180° => 1600 steps in 0.1s), then the needed acceleration can be calculated like this:

        s = 1/2 * a * t²

        1600 steps = 1/2 * a * (0.1s)²

       a = 3200 / 0.01 s² = 320000 steps/s²

The maximum speed at 0.1s is then:

       v = a * t = 320000 steps/s² * 0.1s = 32000 steps/s

So the parameter to setSpeed should be

       1000000 us/step / 32000 steps/s = 31.25 => 31

vangalvin, 2020-06-22:
Awesome, I will give that a try when the new teensy arrives, 
The stepper I am using is a hybrid stepper with encoder feedback, to be honest with the SC5708 it acts more like a servo than a stepper and seems to well and truly out perform any other stepper and driver conbination I have used. 

Thak you for your assistance on that. I will let you know how it goes.

gin66, 2020-06-25:
while waiting, I will close the issue

closed 2020-06-25 by gin66