gin66/FastAccelStepper #2: Stop and SetCurrentPosition methods

WTM-Jarryd, 2020-09-07:
Hi Gin66,

Great library that you have here.

Do you have an update on when the SetCurrentPosition method will be done and do you have plans of having a Stop method in place to stop all current operations for the queue?

These operations are important when it comes to homing processes to find end of travel limits.
I am using the library with a nano (atmega328p)

Kind regards,
Jarryd

gin66, 2020-09-10:
Hi Jarryd,

thanks for your feature request. As I have not yet needed those until now, I have not worked on them.

Here questions back:
* SetCurrentPosition: This should act like a command, which is appended to the queue, or set the current position while the motor is running ?
* Stop shall act immediately, means aborting the current command , or just remove the not yet started commands from the queue ?

Regards
Jochen

WTM-Jarryd, 2020-09-11:
Hi Jochen,

The SetCurrentPosition will act as a command , i.e. appended to the queue.
This will usually be used for homing purposes similar to what 3D printers do to find their physical zero positions for axes.

The Stop should act immediately. This will cover the cases for homing and emergency stop conditions.
During homing you will give the stepper queue a move action, usually the calculated max travel steps for the axis length, but as soon as it detects a limit switch it will need to stop, otherwise it could damage the switch of ride into a hard stop.

The Stop would need to do a deceleration stage to prevent abrupt stops in motion as you could do physical harm to a device.
During homing this isn't as much of an issue as you generally do homing movements at slow speeds, so deceleration happens very quickly.

Let me know if you need further information or example cases.

With these two features, I believe you will have a very good library.
The step rates achievable even with the avr chips are great.

Kind regards,
Jarryd

gin66, 2020-09-11:
Hi Jarryd,

now I have implemented setting the position with two functions:
* setCurrentPosition(int32_t new_pos)
   This will set the momentary position as good as possible while running.
* setPositionAfterCommandsCompleted(int32_t new_pos)
   For low level queue control. 
   While stepper is running with ramp control by the library, then
   the change will take place at the end of the current queue commands (whatever this is)

So in general it is best to set the position while stepper is not moving.

This is implemented in 0.4.3

If Stop shall decelerate and happen after current queue commands being executed, then this would be equivalent to moveTo(getPositionAfterCommandsCompleted() +/- stepsForDeceleration). The stepsForDeceleration is known to the driver.

Not sure, if this is what you want, because the queue can be quite filled (at least for esp32). Consequently I assume, you would assume to empty the queue and perform immediate deceleration. Please comment

BR
Jochen

gin66, 2020-09-11:
In the meantime have pushed out version 0.4.4
This implements stopMove()

For testing please use example DualStepper, which has reworked a lot.

gin66, 2020-09-11:
In the meantime have pushed out version 0.4.4
This implements stopMove()

For testing please use example DualStepper, which has been reworked a lot.

gin66, 2020-09-13:
Apparently the 0.4.x and the initial 0.5 version have an issue with Stepper B for AVR.
Hopefully can fix it today.

gin66, 2020-09-13:
Finally, just released the 0.6.0
Now it works, based on my tests.

WTM-Jarryd, 2020-09-14:
Hi Jochen,

Thank you very much for the effort you are putting into this.

For the stop, it would need to empty the queue and then do the deceleration/stop.
It is to abort all current operations and come to a halt.

I am currently working on another controller project, currently implementing a pause system, so I won't be able to test this today.
As soon as I am able to get back on this project, I will test and let you know what the results are.

kind regards,
Jarryd

devrim-oguz, 2020-09-22:
@gin66 Thank you very much for this library. This is the best stepper library I have used so far. I can drive stepper motors really fast using an ESP32 thanks to this. However, there is a slight issue on the auto enable function while using the setCurrentPosition().
For example, if I try this code:
```
    homeAxis();

    fastStepper.init();
    axisStepper = fastStepper.stepperConnectToPin(AXIS_MOTOR_STEP_PIN);

    if( axisStepper ) {
        axisStepper->setDirectionPin(AXIS_MOTOR_DIRECTION_PIN);
        axisStepper->setEnablePin(AXIS_MOTOR_ENABLE_PIN);
        axisStepper->setAutoEnable(true);
    }
    else {
        ESP.restart();
    }

    axisStepper->setCurrentPosition(0);

    delay(1500);
    axisStepper->setSpeed(30);
    axisStepper->setAcceleration(5000);
    axisStepper->moveTo(550400);
```

It homes the steppers using directly turning on and off the pins and then it sets the position to 0. After that, it moves to a given position which it can't reach correctly. It always misses it with some margin. I think this is due to starting driving the motors as soon as you enable them, because some drivers takes some time to adjust before actually pushing steps. If I change the code like this:

```
if( axisStepper ) {
        axisStepper->setDirectionPin(AXIS_MOTOR_DIRECTION_PIN);
        axisStepper->setEnablePin(AXIS_MOTOR_ENABLE_PIN);
        //axisStepper->setAutoEnable(true); <-Commented this line
    }
    else {
        ESP.restart();
    }

    axisStepper->enableOutputs(); //Manually enabling the outputs and waiting a little after that
    delay(200);
    axisStepper->setCurrentPosition(0);

    delay(1500);
    axisStepper->setSpeed(30);
    axisStepper->setAcceleration(5000);
    axisStepper->moveTo(550400);
```
it works without a problem. I think you need to give some time to the drivers before actually pushing steps to them. I don't know if you'd have the same issue with other drivers, but this is the case with my HPD-HBS57 closed loop stepper driver. It never misses any step due to its closed loop nature, however it misses steps with autoEnable turned on. I think you should wait a little before disabling the stepper motors between the commands and wait a little before starting them. This way they can have some time to adjust, and they wouldn't get turned on and off when there isn't any command in the queue immediately.

Thank you

WTM-Jarryd, 2020-09-22:
Hi Devrim.

Please create this request as a new issue as this is an autoEnable issue.

Thanks

WTM-Jarryd, 2020-10-06:
Hi @gin66 

I have finally been able to get back on this project... for the moment.

The `stepper->stop()` method is working very well and liking the new sample code example.
One issue that I am seeing though is that it is causing a sudden stop and then doing the deceleration when moving at higher speeds. The sudden stop is causing the stepper to bind before continuing with the deceleration.
The speeds at which this will start to happen will depend on the stepper motor being used and what torque limits it has.

I've used my oscilloscope to see what the pulses are doing at this point.
I am seeing that as soon as the interrupt trigger and stop() happen, the step pin is being held HIGH for about 330us and then there is a 3.788ms LOW gap directly afterwards. See below images.

Begin of stop - 50us grid resolution
[image]
Duration of delay period (4.118ms total) - 0.2ms grid resolution
[image]

As a side note, the timings otherwise are perfect. Getting the 10us HIGH pulses and 50us LOW gaps 👍 

The time values for the HIGH and LOW delays are always the same.
Could this be due to the overhead of resetting the queue?

This is my case:
Stepper driver: Wantai DQ542MA
Stepper motor: Hetai Electric 57BYGH420 2.0A 1.8deg/step

Microstepping: 1:10 = 2000 steps/rev

Interrupt method is attachInterrupt on pin2 on FALLING condition

Test results of the steps lost for my setup are as follows.

Stepper = Stepper 1
Acceleration = 80000
Range  = 0 to 80000 steps (using the P command from the sample)
These are the step differences for a series of tests. 
```
V = 50µs
1: 120 steps short
2: 80 steps short
3: 120 steps short
4: 40 steps short
5: 120 steps short
6: 160 steps short
7: 40 steps over
8: 40 steps over
9: 160 steps short
10: 40 steps over

V = 60µs
1: 1440 steps short
2: 1400 steps short
3: 1360 steps short
4: 1440 steps short
5: 1320 steps short
6: 1360 steps short
7: 1360 steps short
8: 1440 steps short
9: 1400 steps short
10: 1360 steps short
```
Interestingly the step differences are always multiples of 40

Below is my main.cpp (doing the project with PlatformIO)

```cpp
#include <Arduino.h>
#include "FastAccelStepper.h"

#define DIN3 2

[... 198 lines cut ...]
    }
  }
  else
  {
    stopped = !(stepper1->isRunning());
  }
  // interrupt debugging
  if (isInterrupted)
  {
    // count to visually separate multiple interrupt displays
    interrupt_count++;
    Serial.print(interrupt_count);
    Serial.println(" - Interrupt triggered");
    if (interrupt_count == 255)
    {
      interrupt_count = 0;
    }
    isInterrupted = false;
  }
}
```
Let me know if you need anything further.

Kind regards,
Jarryd

gin66, 2020-10-06:
Hi Jarryd,

thanks for testing and your comments. Only this evening I have time to check for this issue on the hardware. BTW I normally use platformio, too. But from the code, I have already a good guess for the reason.

The lowHardLimit() is called in an interrupt context and blocks all other interrupts, this leads to the stepper aka timer interrupt getting delayed and no steps coming out, This is easily visible, because the high->low step transition is done inside the interrupt routine. The stopMove() is a very slow operation and needs approx 300us.

There are two solutions:
1. enable the interrupts at the beginning of lowHardlimit(). Here need to make sure, that the interrupt flag is reset before so.
2. move the stopMove out of the interrupt routine

gin66, 2020-10-06:
This critical interrupt situation could be improved, by changing the implementation to use OC1A and OC1B to drive only one stepper motor and not two in parallel. Then OC1A could do the rising edge and OC1B the falling edge.

WTM-Jarryd, 2020-10-06:
Hi @gin66 

Okay that makes sense.
I'll try adding the interrupt enable within the limit interrupt.

I'm going to be moving to an ESP32 in the future for this task as the performance from the ESP32 is needed.

Thanks for this.

Kind regards,
Jarryd

gin66, 2020-10-07:
Better implementation of StopMove would be setting a flag, which in the ramp generator leads to switch into immediate deceleration. This could then be called from an interrupt routine

gin66, 2020-10-09:
In branch isr_rework, this flag is implemented and stopMove() can be called from an interrupt.

closed 2020-10-09 by gin66

WTM-Jarryd, 2020-10-12:
Hi @gin66,

You are a champion.
I will try have this tested for you as soon as possible.

Thank you for all the effort.

Kind regards,
Jarryd