gin66/FastAccelStepper #3: Error compling example in Arduino IDE

jangrik, 2020-09-07:
Hi,
I get an error when trying to compile the simple avr example in Arduino IDE.
Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Arduino Nano, ATmega328P"
In file included from C:\Users\joerg\Documents\Arduino\sketch_sep07c\sketch_sep07c.ino:1:0:
C:\Users\joerg\Documents\Arduino\libraries\FastAccelStepper\src/FastAccelStepper.h:188:39: error: conversion from 'FastAccelStepper*' to non-scalar type 'FastAccelStepper' requested
 #define stepperA() stepperConnectToPin(9)
C:\Users\joerg\Documents\Arduino\sketch_sep07c\sketch_sep07c.ino:8:40: note: in expansion of macro 'stepperA'
 FastAccelStepper stepperA = fas_engine.stepperA();
                                        ^~~~~~~~
C:\Users\joerg\Documents\Arduino\sketch_sep07c\sketch_sep07c.ino: In function 'void setup()':
sketch_sep07c:13:12: error: base operand of '->' has non-pointer type 'FastAccelStepper'
    stepperA->setDirectionPin(dirPinStepperA);
            ^~
sketch_sep07c:15:12: error: base operand of '->' has non-pointer type 'FastAccelStepper'
    stepperA->setEnablePin(enablePinStepperA);
            ^~
sketch_sep07c:16:12: error: base operand of '->' has non-pointer type 'FastAccelStepper'
    stepperA->setAutoEnable(true);
            ^~
sketch_sep07c:18:12: error: base operand of '->' has non-pointer type 'FastAccelStepper'
    stepperA->setSpeed(1000);       // the parameter is us/step !!!
            ^~
sketch_sep07c:19:12: error: base operand of '->' has non-pointer type 'FastAccelStepper'
    stepperA->setAcceleration(100);
            ^~
sketch_sep07c:20:12: error: base operand of '->' has non-pointer type 'FastAccelStepper'
    stepperA->move(1000);
            ^~

exit status 1
base operand of '->' has non-pointer type 'FastAccelStepper'

CODE
--------------
#include "FastAccelStepper.h"

#define dirPinStepperA    5
#define enablePinStepperA 6
//#define stepPinStepperA   9  // OC1A

FastAccelStepperEngine fas_engine = FastAccelStepperEngine();
FastAccelStepper stepperA = fas_engine.stepperA();
//FastAccelStepper stepperB = fas_engine.stepperB();

void setup() {
   fas_engine.init();
   stepperA->setDirectionPin(dirPinStepperA);

   stepperA->setEnablePin(enablePinStepperA);
   stepperA->setAutoEnable(true);

   stepperA->setSpeed(1000);       // the parameter is us/step !!!
   stepperA->setAcceleration(100);
   stepperA->move(1000);
}

void loop() {
}

WTM-Jarryd, 2020-09-07:
The error is in your stepper declaration.
You have to declare your stepper using the * (pointer) reference.

I.e. FastAccelStepper stepperA = fas_engine.stepperA();
must be FastAccelStepper *stepperA = fas_engine.stepperA();

You can only access the steppers via their pointer to the library.

Hope this helps.

jangrik, 2020-09-07:
Thanks a lot. 
I had to change another thing: fas_engine has to be renamed. The example that compiles now without error:
----------------
#include "FastAccelStepper.h"

#define dirPinStepperA    5
#define enablePinStepperA 6
//#define stepPinStepperA   9  // OC1A

FastAccelStepperEngine fast_engine = FastAccelStepperEngine();
FastAccelStepper *stepperA = fast_engine.stepperA();
//FastAccelStepper *stepperB = fast_engine.stepperB();

void setup() {
   fast_engine2.init();
   stepperA->setDirectionPin(dirPinStepperA);

   stepperA->setEnablePin(enablePinStepperA);
   stepperA->setAutoEnable(true);

   stepperA->setSpeed(1000);       // the parameter is us/step !!!
   stepperA->setAcceleration(100);
   stepperA->move(1000);
}

void loop() {
}

WTM-Jarryd, 2020-09-08:
Okay great stuff.

It is generally a good idea to give a unique name to your objects and not reuse the internal names used by the library.
Can cause things to get a bit confusing when trying to trace references in your IDE.

If the problem has been solved, please close the issue.

gin66, 2020-09-10:
Thanks to WTM-Jarryd for your great support.

For the fas_engine problem: I need to update the README, because the example avoids the name clash and does not use fas_engine but only engine.

gin66, 2020-09-11:
version 0.4.3 addresses the issue.
1.) fas_engine defined as static inside the library
2.) the global fas_queue kept as this. added a warning in the README.
3.) the variables fas_ledPin and fas_debug_led_cnt defined as static, too

jangrik, 2020-09-11:
thanks a lot

closed 2020-09-11 by jangrik