gin66/FastAccelStepper #4: setPinsInverted() support for different modules

devrim-oguz, 2020-09-21:
Hello, are you going to add setPinsInverted() support in the future? This might help the usage with different drivers or different connection configurations. This property exists in the AccelStepper library like so:
`    stepper.setPinsInverted(false, false, true);`

Thanks in advance.

gin66, 2020-09-24:
Hi, the implementation for DIR and Enable is quite easy, while for Step this is hardcoded with direct timer register settings. Is it for you sufficient, to have this for DIR and Enable only ?

devrim-oguz, 2020-09-24:
Yeah, I guess so. Step would work either way anyways. However, I didn't like this approach a lot since it is a bit ambiguous which pins are which. I guess you can overload the setEnablePin() and setDirectionPin() functions, so they would both accept these kind of calls:

```
stepper->setEnablePin(5); <- Default way and compatible with older versions
stepper->setDirectionPin(5, PIN_INVERTED); <- Invert the pin while adding
```

You might need to create a definition for PIN_INVERTED and PIN_NORMAL though.
For example:

#define PIN_INVERTED true
#define PIN_NORMAL false

Thanks a lot for your help.

gin66, 2020-09-24:
Actually I prefer to add these functions:
* setEnablePinLowActive() and setEnablePinHighActive()
* setDirectionPinLowActive() and setDirectionPinHihghActive()

The former setEnablePin() and setDirectionPin() would be deprecated.

Advantage: 
* clear interface
* can be independently assigned, so one stepper drive can output Direction with low and high active simultaneously e.g. two steppers with two drivers, which need to step synchronously but with opposite direction

devrim-oguz, 2020-09-24:
Makes sense. But I hope there will be no typos in the function call haha 😄

Thanks again

gin66, 2020-09-24:
Actually I prefer to add these functions:
* setEnablePinLowActive() and setEnablePinHighActive()
* setDirectionPinLowActive() and setDirectionPinHighActive()

The former setEnablePin() and setDirectionPin() would be deprecated, because the „defaults“ are for one driver type.

Advantage:
* clear interface
*can be independently assigned, so one stepper drive can output Direction with low and high active simultaneously e.g. two steppers with two drivers, which need to step synchronously but with opposite direction

Hmm. Actually I enjoy to not need looking up the driver spec. Something like configureA4988() would be an idea, too.

Opinions ?

devrim-oguz, 2020-09-24:
Yeah, but I think we should both have manual and automatic adding. It wouldn't hurt. Since some drivers don't have a single configuration. For example my driver have optocouplers on its inputs so I can either put VCC to the + pin and pull the - to ground in order to signal it or I can connect the - pin to GND and pull the + pin high in order to signal the same thing. It works both ways, but the input gets inverted. I chose doing the first one the first time I tried, that's why I needed inverted pins, but then I changed it. So there are two ways to configure some drivers.

gin66, 2020-09-24:
Just have pushed out the version 0.6.4.
The change is API backward compatible using C++ default parameters :-)
  ```
void setDirectionPin(uint8_t dirPin, bool dirHighCountsUp = true);
  void setEnablePin(uint8_t enablePin, bool low_active_enables_stepper = true);
```

gin66, 2020-09-24:
Actually this overloading has been proposed from your side already.
Hope the parameter naming is self explaining and no need for PIN_INVERTED.

I wonder, if your defines (accordingly modified) to DIR_HIGH_COUNTS_UP, DIR_LOW_COUNTS_UP,.... should be added for clarity?

devrim-oguz, 2020-09-24:
That would actually be nice. I can try this tomorrow and close the issue after testing.

gin66, 2020-10-30:
stale. if does not work, just open a new issue

closed 2020-10-30 by gin66