gin66/FastAccelStepper #6: Can't use pins after ESP.restart()

devrim-oguz, 2020-09-24:
I didn't wanted to open a new issue for this, but I gotta ask. Do you know why I can't drive a stepper manually after I use the ESP.restart() function? Do the pins stuck attached to something after restart? What I mean is digitalWrite functionality doesn't work on those pins. Maybe the library also gets broken, idk.

devrim-oguz, 2020-09-24:
For example try adding an ESP.restart() to the singlestepper example, it probably won't turn after it restarts.

gin66, 2020-09-24:
The doc says:

‘‘‘
This function can be called both from PRO and APP CPUs. After successful restart, CPU reset reason will be SW_CPU_RESET. Peripherals (except for WiFi, BT, UART0, SPI1, and legacy timers) are not reset. This function does not return.
‘‘‘

Could you please try ESP.reset() ?

devrim-oguz, 2020-09-24:
I actually tried, but ESP.reset() does not compile. Maybe it's because I'm using it on PlatformIO. Will try again tomorrow and notify you.

devrim-oguz, 2020-11-27:
I see that you moved it to the idea list, maybe you can add a Stepper.detach() function to disable the library before restarting.

gin66, 2020-11-27:
Yes. The esp32 is currently not available on my bench, that's why I have not yet digged into it. But I plan to do so.
The Stepper.detach() is a good alternative.

gin66, 2020-12-03:
in the meantime the StepperDemo has a command "r" to initiate ESP.restart() in version 0.12.0
With this I have tested it and the stepper continue to run after a restart().
Eventually one of the earlier bugfixes around interrupt management has fixed it.
Anyway, please give it a try.

BTW: Stepper.detach() would be complicated to implement. The initialization relies on sequentially allocating the stepper. If a stepper would be first attached and then attached again, this would scramble the initialization sequence. In the moment I prefer to not implement detach(). As restart() seems to work, the need for detach() may not be there

devrim-oguz, 2020-12-04:
Thanks a lot. I don't have a stepper at the moment, but I will try. detach() is not necessary if the esp is able to restart properly anyways.

gin66, 2020-12-13:
stale issue

closed 2020-12-13 by gin66