gin66/FastAccelStepper #7: Real Time System Latency

etempm, 2020-10-06:
Hi,
This is very nice and seems that very powerfull. I want to use this library on my realtime motion simulation project. I am currently processing commands from pc and generating pulses with arduino with interrupt timers, like you. But I dont have acceleration / decc routines, and I am very happy to found your library. Also nano/uno is not enought mcu power for 4 stepper. So ESP32 is also very vital for me.

My question is latency. I am asking only latency of library after run command. On docs, I saw 4ms for ATMEGA and 10ms for ESP, but I am not sure.

Thanks for help.

gin66, 2020-10-06:
The implementation on avr and esp32 are quite different, so I focus only on esp32. As I have no specific test case for this, my answer is based on design only.

Shortest latency is for this case:
* stepper is NOT running
* auto enable is not active

Then the time till the first step comes out is:
1. execution of move/moveTo, which call _calculate_move()
2. execution of _calculate_move(), which sets the variables for the cyclicTask
3. The next cyclicTask (period Time 10ms), will inits hardware registers to output the first step
4. The hardware will output first step after the time t0. This t0 equals the slowest step frequency at the beginning of the ramp

Example: If the step frequency at the beginning of the ramp is 10Hz, then the latency would be worst case: 10ms+100ms+several us=110ms and best case 100ms.

If the step frequency at the beginning of the ramp is 1kHz, then the latency would be worst case: 10ms+1ms+several us=11ms and best case 1,0xx ms

Hope this helps

etempm, 2020-10-06:
Very clear. Thanks again.

gin66, 2020-10-06:
In case the stepper is already running (queue contains commands), then a change in commands could take for slow stepper movements even seconds. Reason is - as mentioned under ISSUES, that the queue is always filled completely and the esp32 queue is rather deep (32 entries). Currently I am thinking about an implementation to fill the queue only for approx 20ms as lookahead. Just this is a bit complicated for implementation, that’s why not yet done.

etempm, 2020-10-06:
If I am wrong, please correct me. When I send new position command while previous command is not completed, I will wait queue to complete ? Is that possible to clear queue and re calculate route instantly ?

gin66, 2020-10-06:
if next move/moveTo changes direction, then you need to wait till the motor stops. Initiating a stop can be done with stopMove(). Current implementation of stopMove() is just a moveTo(end_pos) with end_pos being the position, which is reached with immediately starting deceleration. 

If changes shall be done in the same direction as the motor is already running, then the result depends on the command. A moveTo will replace the end position of the current running movement. A move will be added to the end position of the current running movement. Any move/moveTo will consider any updates in speed or deceleration.

gin66, 2020-10-06:
and I think, the moveTo for a running motor has the risk of a race condition in case of a running motor....
Explanation: the current end position of a command queue is read and used for calculation. If between getting the end position and update of the control variables the task/interrupt routine gets executed and commands are added, then the end position may be missed.

Sounds like a bug, which I need to look in further.

etempm, 2020-10-06:
Thanks for your time. I am watching your project and I will write my results.

gin66, 2020-10-07:
After some more analysis and tests by injecting virtual interrupts during testing, I could not found any bug in regard to wrong end position and rereading the code, confirms this.

Nevertheless, those tests revealed a possible divide by zero error, which is quick fixed now until a better rework is done. So it was work the exercise.

For this comment, I think this issue can be closed, because I reopened it due to a possible bug in move/moveTo, while there isn’t.

closed 2020-10-07 by gin66