gin66/FastAccelStepper #10: Usage example code not doing anything?

TwinLizzie, 2020-11-27:
I didn't know where else to post this. I've tried using the example code on the Readme, and it doesn't do anything, but does seem to activate the stepper motor.

```
#include "FastAccelStepper.h"

#define dirPinStepper    2
#define enablePinStepper 4
#define stepPinStepper   3  // OC1A in case of AVR

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {
   engine.init();
   stepper = engine.stepperConnectToPin(stepPinStepper);
   if (stepper) {
      stepper->setDirectionPin(dirPinStepper);
      stepper->setEnablePin(enablePinStepper);
      stepper->setAutoEnable(true);

      stepper->setSpeed(1000);       // the parameter is us/step !!!
      stepper->setAcceleration(100);
   }
}

void loop() {
    stepper->moveTo(1000);
}
```

Does the FastAccelstepper library have some kind of stepper.run() function like accelstepper does?

I'm interested in using this library for a robotics project. It would be great to have a simple sweep code to begin with (stepper move clockwise.. stop. Stepper move anti-clockwise. Stop)

gin66, 2020-11-27:
Do you use atmega328 aka AVR or an esp32 ?

In case of atmega328, then the step pin is restricted as per Readme:
avr: This must be connected for stepper A to Pin 9 and for Stepper B to Pin 10.

The FastAccelStepper library does not need a stepper.run(). For AVR all is done via interrupts and for esp32 interrupt and tasks.

In order to check the HW, I recommend to first use the StepperDemo. Need to adapt the HW configuration first.
Then via serial you can issue commands. For example you could:
1. M1 - select the stepper 1
2. N - enable the output driver, if enable pin is connected. After this, the motor axle should be fixed.
3. T - drive motor with direct port access. Motor should run 3200 steps in one and then the other direction.
4. Try the ramp function with e.g. A1000 V1000 R1000

TwinLizzie, 2020-11-27:
Thanks for the heads up. Wouldn't it be better to be able to customize this, though? Say, if I have a PCB that is already wired a certain way for certain pins?

gin66, 2020-11-27:
For atmega328, this is hardwired into the uC. The timer output compare pins are fixed like this. And the high speed comes from using those hardware timers.

If you need the freedom of pin assignment, then you should consider AccelStepper, which does not have this restriction.

TwinLizzie, 2020-11-27:
Thanks, this makes perfect sense now. Works perfectly!

closed 2020-11-27 by TwinLizzie