gin66/FastAccelStepper #9: Idea: add forwardStep() and backwardStep() functions for homing

devrim-oguz, 2020-11-27:
As far as I can tell, the library does not include a stop function which would stop the stepper immediately without acceleration. However, this is needed for homing purposes in very precise devices. The homing function runs the stepper step by step until it hits an end stop or triggers a sensor. As soon as the end stop detected, the controller needs to stop the motor in order to get a very accurate position. It doesn't seem to be possible right now to do this using the library(Correct me if I'm wrong please). In fact I am doing it manually by toggling the step pin. What I'm suggesting is maybe we can add forwardStep() and backwardStep() functions to move the motor one step at a time to be able to check the sensor status at each step. What do you think about this? Thanks in advance.

gin66, 2020-11-27:
forwardStep() and backwardStep() can be realized by move(1) and move(-1). If forwardStep() and backwardStep() are expected to return after the step, then a blocking wait for isRunning() would be needed after move().

The current StopMove() has been added based on #2 and request was _with_ deceleration. Reason was, that an abrupt stop may have adverse effect on the mechanics. 

Besides, I would assume homing is run with slow speed. And here the current implementation is not ok, because the queue is filled too much. Related idea is: https://github.com/gin66/FastAccelStepper/projects/1#card-46938149
With the latest change due to #8, the planned time in the queue can be easily calculated. Just haven't done it yet.

The interface could be changed to StopMove(withoutDeceleration = false) to allow both versions.

Related to this issue is: https://github.com/gin66/FastAccelStepper/projects/1#card-47125120

devrim-oguz, 2020-11-27:
Maybe you can bypass the queue and move the steppers directly for this functionality. It would allow for the devices to home initially then start moving according to the queue. I don't know how exactly your code works though.

gin66, 2020-11-27:
forwardStep() and backwardStep() is implemented. forwardStep(true)/backwardStep(true) will block until stepper is stopped.
Haven't measured the time yet. I would expect blocking time in the magnitude of 40µs without need to auto enable the driver

These functions push a single step to the command queue, which allows goodies like auto enable/position counting still working - even though the blocking version would wait a bit.

stopMove() with emptying the queue is not yet done

gin66, 2020-11-27:
The problem with stopMove() with emptying the command queue is: emptying the command queue and reverse calculating the position takes time. This time is most likely too much to be called from an interrupt routine.

One possible solution could be to introduce immediateStopWithPositionSet(uint32_t new_pos) or similar
This function should
1.) empty the queue
2.) if possible stop any running command
3.) set the positionAtQueueEnd to <new_pos>
4.) stop ramp generator

If this is ok, then this is much easier to implement

gin66, 2020-11-28:
Have called the function forceStopAndNewPosition().
With StepperDemo use 'X' to initate abort
Only tested with StepperDemo and avr

gin66, 2020-12-10:
stale issue

closed 2020-12-10 by gin66