gin66/FastAccelStepper #11: Problems with motors 4-6 on ESP32

billieblaze, 2020-11-29:
Found your library and mostly its working great for me.   Super glad to see a few today too!  

I'm using an ESP32 (NodeMCU) and I cannot get the last 3 motors working.   I've verified my wiring by switching positions (last 3 -> first 3, vice-versa)  in the config array.   Not real sure whats up, so I thought I'd reach out for help.    Entirely possible I'm doing something dumb.  

Config: 
```
const struct stepper_config_s stepper_config[6] = {
    {
      step : 4,
      enable_low_active : 32,
      enable_high_active : PIN_UNDEFINED,
      direction : 5,
      direction_high_count_up : true,
      auto_enable : true,
      on_delay_us : 5000,
      off_delay_ms : 10
    },
    {
      step : 12,
      enable_low_active : 32,
      enable_high_active : PIN_UNDEFINED,
      direction : 13,
      direction_high_count_up : true,
      auto_enable : true,
      on_delay_us : 5000,
      off_delay_ms : 10
    },
    {
      step : 14,
      enable_low_active : 32,
      enable_high_active : PIN_UNDEFINED,
      direction : 15,
      direction_high_count_up : true,
      auto_enable : true,
      on_delay_us : 5000,
      off_delay_ms : 10
    },
    {
      step : 17,
      enable_low_active : 32,
      enable_high_active : PIN_UNDEFINED,
      direction : 18,
      direction_high_count_up : true,
      auto_enable : true,
      on_delay_us : 5000,
      off_delay_ms : 10
    },
    {
      step : 19,
      enable_low_active : 32,
      enable_high_active : PIN_UNDEFINED,
      direction : 21,
      direction_high_count_up : true,
      auto_enable : true,
      on_delay_us : 5000,
      off_delay_ms : 10
    },
    {
      step : 22,
      enable_low_active : 32,
      enable_high_active : PIN_UNDEFINED,
      direction : 23,
      direction_high_count_up : true,
      auto_enable : true,
      on_delay_us : 5000,
      off_delay_ms : 10
    },
    
};
```

gin66, 2020-11-29:
Thanks for looking at FastAccelStepper. These days I am working on the library for more functionality and less bugs.
On my desk is in the moment an arduino nano (avr), while my esp32 project is currently is on the shelf - waiting.

From your code, I am surprised, that you hook all six steppers on the same enable pin. Even this makes perfectly sense from usage point of view, I have not had this use case in mind. If one stepper is auto disabling, it will shut off all steppers. I do not think, that is the expected behavior.

In order to have a central auto disable/enable - via the FastAccelStepperEngine - I would need to update the code in several areas. Consequently in your case, please disable auto enable for all steppers and try again.

Second point, I have never tested more than two steppers. Even though the code is written quite generic and I could not spot an error, I have never used the second pcpwm module. Actually easy to be done by defining few dummy steppers, but haven't thought about before.

As I am not sure, if I can pull out my esp32 project before next weekend. Perhaps you could check with the auto enable disabled and give me feedback.

billieblaze, 2020-11-29:
I'll check that out and report back.    I'm not actually using the enable pins on my drivers, just have them shorted at the moment.

gin66, 2020-11-29:
If the enable pin does not switch off the motors, then this assumption of mine is wrong.

So I need to check on my hardware and I will try to squeeze it in before next weekend.

gin66, 2020-11-30:
I have tried with my board (just one stepper) and I have configured all 6 stepper drivers to use the same Step/Enable/Direction Pin. Auto enabled turned off. Result: it works as expected.

Another try was to change the definition in StepperISR_esp32.cpp to first use mcpwm1 module and then mcpwm0 module. Again, it works as expected.

Could you please try with the latest library ? Currently I have no clue, what the root cause could be

TwinLizzie, 2020-12-01:
@gin66 have you tested those 6 steppers using different step/direction pins? 

I've just bought an ESP32 for testing, and will try running the FastAccelstepper library on it too. I will report back to this thread and let you know how it went.

gin66, 2020-12-01:
In version 0.11.2 I have fixed an old bug in the esp32 code. If two steppers are running in parallel, the two motors' interrupt requests could eventually collide and leading to a deadlock. This has been fixed hopefully. Currently I have one stepper at mcpwm module 0 and one at mcpwm module 1. Both run well in parallel. If two are connected to mcpwm0, seems ok, too.

Hope with this version, you will have good results

gin66, 2020-12-03:
Correction 0.12.0

gin66, 2020-12-05:
Direct drive is now working for esp32.

BTW: central auto enable pin should work

billieblaze, 2020-12-05:
will check this out this weekend, thanks for the updates

gin66, 2020-12-13:
stale issue

closed 2020-12-13 by gin66