gin66/FastAccelStepper #14: Problem with ESP32_DEVKIT V4  WROOM & StepperDemo

Francois71828, 2020-12-03:
Hello,
I want to run a single stepper (200 steps per turn; 1A current spec) using an DRV8824 powered at 12V for the motor (DRV8824 trimpot adjusted). 
I am using a win10 laptop and an Arduino environment.
When using an Arduino mega2560 hooked to a RAMPS card with A DRV8824, I get correct motion of the motor using a StepperDriver library (with acceleration/deceleration capability)

I want now to use a and an ESP32_DEVKIT V4  WROOM , and my goal is to use 6 steppers at 200 steps/turn, to rotate concurently each with an eventual different speed in the range 20RPM-800RPM

I have started to test only one motor; I am encountering problems;  and would be happy to find a way to fix the problem:
I am using the "StepperDemo" from  FastAccelStepper
I have used GPIO5 for stepping, GPIO27 for enable (enable_low_active), GPIO32 for direction; also I have hooked up a led in series with  a 200 ohm resistance between pin GPIO5 and ground, so as to monitor step activity. 

Here is what I get on the console:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Demo Stepper
    F_CPU=40000000
    TICKS_PER_S=16000000
Enter commands separated by space, carriage return or newline:
     M1/M2/..  ... to select stepper
     A<accel>  ... Set selected stepper's acceleration
     V<speed>  ... Set selected stepper's speed
     U         ... Update selected stepper's speed/acceleration while running
     P<pos>    ... Move selected stepper to position (can be negative)
     R<n>      ... Move selected stepper by n steps (can be negative)
     K         ... Keep selected stepper running in current direction
     @<pos>    ... Set selected stepper to position (can be negative)
     E<us>     ... Set selected stepper's delay from enable to steps
     D<ms>     ... Set selected stepper's delay from steps to disable
     N         ... Turn selected stepper output on (disable auto enable)
     F         ... Turn selected stepper output off (disable auto enable)
     O         ... Put selected stepper into auto enable mode
     S         ... Stop selected stepper with deceleration
     X         ... Immediately stop motor and set zero position
     I         ... Toggle motor info, while any motor is running
     W         ... Blocking wait until selected motor is stopped (will deadlock if the motor will never stop)
     +         ... Perform one step forward of the selected motor
     -         ... Perform one step backward of the selected motor
     T         ... Test selected motor with direct port access
     Q         ... Toggle print usage on motor stop
     ?         ... Print this usage

   M1: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
Select stepper 1
Set acceleration to 1000
Set speed (us) to 1000
Output driver on           {>>> here the axis of the motor is not anymore  free to rotate , which is OK}
Move to position 100
returncode=0
M1: AUTO Curr=0 QueueEnd=1/0us Target=100 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=0 QueueEnd=1/0us Target=100 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=0 QueueEnd=1/0us Target=100 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=0 QueueEnd=1/0us Target=100 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
[... 187 lines cut ...]
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
(.......)
(.......)
(.......)
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=1 QueueEnd=5/9984us Target=1000 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
ets Jun  8 2016 00:22:57
* * * 
Similarly the led on GPIO5 is NOT turned on at all (no led activity)
I also tested the "T" command: it does apparently nothing that I could detect (axis of motor free to rotate or not, eventual noise of the motor, led on, etc...); Upon typing "T" and hitting return, I just get "nothing", and no feedback on the console either.

gin66, 2020-12-03:
Thanks for the lots of information. Unfortunately, from the log I do not see, which version you are using.

Anyway, the issue sounds familiar, which has been fixed in e32b48e5c156e2e8c8e38f367c3146c319f0f53f. Some commits before (while revising the code for introduction of pause commands), I have managed to make an endless loop in the StepperTask, which is stated in your log (watchdog not triggered)

Please try version 0.12.1. This I have currently on my bench running and looks quite ok.

With github actions, I can test the ramp generator quite well. But for anything hardware related, I have no systematic test approach. Especially I do not always check on both supported platforms, all features. So the blame is on me.

The "T" command works very well on avr. Only today I have recognized, that on esp32 this does not work. That's why, the command is disabled in the latest StepperDemo - for now. The reason is, that on avr, the stepper pin is disconnected, if the pulse generator is not working. The same is not implemented on esp32, besides it is a bit more complicated to do. So T will control enable/direction as expected, but no steps come out. The next days I will check, if I can enable this command for esp32 again, because on avr it is quite helpful.

gin66, 2020-12-04:
but was introduced on cbf3c43134b0cd0782b104e2ccda3f6a0d2dfc2c

Francois71828, 2020-12-04:
Hello,
thanks for your lightning fast feedback  yesterday! :)
(and sorry for my previous lengthy pasting of the console output)
I have now just tested  the version FastAccelStepper-0.12.1 on my setup.
I get a similar behavior, see next.
note1: I am using the Arduino IDE 1.8.13
note2: what value for acceleration and velocity do you recommend for plain-testing your library without pushing to limits for a generic stepper motor  ?

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Demo Stepper
(...)
M1: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE 
   
Select stepper 1
Set acceleration to 1000
Set speed (us) to 1000
Output driver on
Move to position 100
returncode=0
M1: AUTO Curr=0 QueueEnd=0/0us Target=100 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
(...)
M1: AUTO Curr=0 QueueEnd=1/0us Target=100 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=0 QueueEnd=1/0us Target=100 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
E (76934) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (76934) task_wdt:  - IDLE0 (CPU 0)
E (76934) task_wdt: Tasks currently running:
E (76934) task_wdt: CPU 0: StepperTask
E (76934) task_wdt: CPU 1: loopTask
E (76934) task_wdt: Aborting.
abort() was called at PC 0x400d51f3 on core 0

Backtrace: 0x4008b81c:0x3ffbe170 0x4008ba49:0x3ffbe190 0x400d51f3:0x3ffbe1b0 0x40084869:0x3ffbe1d0 0x400d272c:0x3ffb86a0 0x400d18e9:0x3ffb86c0 0x400d1bfc:0x3ffb86e0 0x400d1d9c:0x3ffb8710 0x400d1ead:0x3ffb8750 0x400d1f61:0x3ffb8780 0x400884d1:0x3ffb87a0

Rebooting...

gin66, 2020-12-04:
Hi Francois,

cool. Just have tried again with your commands and it works fine on my side. So I need to extend the library to capture more info to get down to the root cause. Hope you have sufficient patience to support on this path. What I want to do (list will change/extend eventually):
1. Add version info to StepperDemo output (hopefully can do something automatic, e.g. linked to make fmt)
2. Make T command work
3. Find out, what if the PC value/Backtrace can support in identifying the culprit 
4. Add some Serial.prints at specific code areas, linked to a debug switch.

Perhaps more. Give me some time to come up with this.

Francois71828, 2020-12-04:
Hello Gin,
I will be happy to try out eventual test versions whenever you come up with them.
Best,
Francois

gin66, 2020-12-05:
First two points done:
1. Add version info to StepperDemo output (hopefully can do something automatic, e.g. linked to make fmt)
2. Make T command work

Direct drive for esp32 should work since commit b14b32536188218500e8b6db57d28dced823a6b4.

Francois71828, 2020-12-08:
Hello Gin,
could you prepare a downloadable version of b14b325  as a zip format library ?
I tried to reassemble a modified zip library using the modified files of you last commits; I named it "FastAccelStepper-0.0.1" and then under the Arduino GUI, I specify "add zip library"; But then I do not succeed in getting that "FastAccelStepper-0.0.1" to be selected when I want to prepare a sketch and specify the library version to be used in that sketch: the list proposed under the name "FastAccelStepper" does not contain the new "FastAccelStepper-0.0.1" 
What is the way to escape from this silly trap ?

gin66, 2020-12-08:
Hello Francois, no need. I have made a new official release (0.12.2), which incorporates the fix. Hope it shows up in Arduino library manager already

Francois71828, 2020-12-08:
Hello Gin,

the version 0.12.2 does indeed show up in the Arduino IDE, but it does not load correctly: when specifically selecting the version 0.12.2 and clicking OK, the IDE stays with the 0.12.1

gin66, 2020-12-08:
Strange behavior. The IT support may say: restart Arduino IDE or restart the computer.

Anyway, the nice thing of an official release is, that github provides nice packaged [zip-file](https://github.com/gin66/FastAccelStepper/tags)

Francois71828, 2020-12-09:
Hello Gin,
the problem at hand is not that simple.
It resists to simply restarting the IDE or rebooting the PC.
So it might be a problem within the 0.12.2 package, that the Arduino IDE refuses to load

gin66, 2020-12-09:
I have tried now with my Arduino IDE 1.8.12 in Linux and it works as expected. 
Select 0.12.2 and download is complete. IDE reports 0.12.2 installed and in the drop down list for versions to be installed, the 0.12.2 is missing (because is installed).

Checking in the library folder this file: ~/Arduino/libraries/FastAccelStepper/library.properties
Inside is stated, that 0.12.2 is present.

Francois71828, 2020-12-09:
OK, my bad.
I had a problem with a duplicate folder with libraries.
I removed the old library, and loaded the recent one.
Now thing look fine, the bug appears solved with the 0.12.2 library and the ESP32-DEV 4 WROOM;
Select stepper 1
Set acceleration to 1000
Set speed (us) to 1000
Output driver on
Move to position 100
returncode=0
M1: AUTO Curr=0 QueueEnd=0/0us Target=100 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=10 QueueEnd=12/6432us Target=100 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=30 QueueEnd=34/3824us Target=100 RUN  ACC   M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=58 QueueEnd=62/3616us Target=100 RUN  DEC  M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=82 QueueEnd=84/5600us Target=100 RUN  DEC  M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: AUTO Curr=95 QueueEnd=96/11200us Target=100 RUN  DEC  M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
M1: MANU Curr=100 QueueEnd=100/0us Target=100 STOP  IDLE  M2: MANU Curr=0 QueueEnd=0/0us Target=0 STOP  IDLE  
Enter commands separated by space, carriage return or newline:
     M1/M2/..  ... to select stepper
(...)
* * *
The motor runs now  ok!
I have fiddled a bit with the parameters while there is a sizeable inertial load on the axis of the motor. I am using a Nema 17 low profile, 13Ncm, I_avg~= 0.15A on the DRV8825, and 15V supply.
(motor hQQps://www.aliexpress.com/item/4000035468189.html)
(A2000, V300, N, P20000), or even (A2000, V200, N, P20000) are still ok, while for (A2000, V150, N, P20000), I get stalling after the start has occurred.

I have noticed a little problem with the demo program though: after a fist set of command has been issued and the corresponding motion is terminated, the demo program will do nothing if trying to re-issue the same set of commands. One has to reset the ESP to get going again.

Francois71828, 2020-12-09:
Some additional questions: 
- when specifying V200, does it strictly mean 200 microseconds per step, or is there also some additional overhead time  ?
-is the behavior strictly linear for the parameter V ? (e.g. upon measuring an effective RPM_0 when using a low speed V0: for a speed V1, the RPM_1 will be RPM_0*V1/V0 ?
- for the acceleration (say A2000), how does it relate to step/sec^-2 ?
-does the esp run effectively at 240Mhz, or at the basal clock rate of 40Mhz ? (How to test that ?). 
- if one wants a concurrent serial communication, should one be weary of potential interference with the interrupt-driven motor operation? An upper limit in the serial com speed ?
- How to test (without having yet 6 motors and their drivers) that the system is still behaving nicely (timing-wise)?
-how to get the relative average time fraction  spent for the motor's interrupt servicing with respect to the esp32 other tasks ?

gin66, 2020-12-09:
Happy to hear, that the bug has not reappeared.

For the “after first set of commands issue”. Which commands do you use ?
Repeating e.g. “A2000, V200, N, P20000” for a selecting motor will not do anything. Because at the second command, the motor is already at position 20000. For repeating commands better use A2000, V200, N, R20000

If the motor is coasting with V200, then it will be exact 200us step time (+/-0.0625us @ 16MHz and +/- offset of quarz/uC-clock). There is no overhead. So it is strictly linear for the pulse time and reciprocal for the speed (ticks/s)

A2000 means, that within 1s after motor start the stepper will run at 2000steps/s. 
v[ticks/s]=A[ticks/s^2] * ramp up time[s]

There is no interference with serial communication. Runs smoothly in parallel. On esp32 the interrupt load from the stepper driver is quite low.

The esp32 does not know, if the stepper is attached or not. Just configure the I/Os as if there would be 6 steppers and you could test the performance accordingly (e.g. one stepper at a time). For esp32 you can assume minimum time between two interrupts is around 1ms.

Francois71828, 2020-12-11:
Hello Gin,
the motor is running very smoothly under the tests I have done.
I have a few additional remarks:

1) I realized that in the demo program, some of the built-in commands are missing in the user's menu, such as f, K,S,X,U. That could be easily  fixed-up in a next version.

2) To my surprise :-), I have been able to ramp-up my motor to nearly double its initial max speed, by incrementing progressively the V parameter: 
initially: A2000 V200 N f
then V190 U, few seconds,
then V180 U, etc... down to V110 U, with the motor still running at this much higher speed  :-)

3) in view of the pin-number limitation of the ESP32 WROOM for driving say 6 motors while still having a bunch of I/0 pins for other tasks, I was toying with the eventual idea of using an spi I/O expander.
The idea is that the step pins certainly needs to be fast GPIO pins, while the enable and direction controls do not really  have serious timing constraint: it would be nice if your library offered a simple way of NOT using GPIO pins for the enable and direction of the motors (i.e. interfaced via user-defined means, such as an spi I/O expander). Would you be ready to introduce such possibility in the library ? 

Best,
Francois

gin66, 2020-12-11:
Hi Francois,

just have checked the code and the output in the monitor from code running on esp32. All builtin commands are listed. So I have no clue, what's wrong. Could you please check again with the latest version ?

The direction pin is managed from inside the stepper interrupt routine(s). That's why I do not intend to change that. For the enable pin, this is much easier. But even here I would vote for a callback routine and then it is up to the application to turn it on or off. Basically I want to avoid the burden in the library to support many configurations (different SPIs on different pins) and which I even cannot even test. Hope for your understanding.

For reducing the amount of used pins:
- As result of #11, enable pins can be shared. Perhaps your app needs actually only one global enable pin for all motors ?
- Direction pins can be shared, if the motors never run simultaneously.

Could you please check your application for that kind of solution ? If this does not work, I will check for the callback.

Francois71828, 2020-12-12:
Hello,
- sorry I thought that some commands were missing, but I was wrong, they are indeed in the list printed on the monitor.

- for the idea of reducing the number of pins and specifically for the enable and direction that do not have strict timing requirement (except that the step pulses should not begin until those pins are set):

I actually would like to control each motor separately and concurrently, each with an enable pin (when a particular motor is not moving I would like to be able to specifically disable that motor), and each with a possibly different velocity and direction;
so that none of the pins enable /direction /step can be shared.

But I understand also your point of view of seeking simplicity

Nevertheless, if it is not too much re-work for you, a "callback" mode would be nice if it would allow either simple straight pin addressing or also the possibility to use an spi-I/O expander for the enable and direction controls (and allowing possibly some delay for the completion of the "slow" pin settings on the spi-I/O expander )

gin66, 2020-12-15:
Function has been implemented. Please try:

```
  // If the enable pins are e.g. connected via external HW (shift registers),
  // then an external callback function can be supplied.
  // This will be called for defined low active and high active enable pins.
  // If both pins are defined, the function will be called
  // twice. The supplied value is either LOW or HIGH. The return value shall be
  // the status of the pin (either LOW or HIGH).
  //
  // In auto enable mode, this function is called from cyclic task/interrupt
  // with 4ms rate, which creates the commands to put into the command queue.
  // Thus the supplied function should take much less time than 4ms.
  // Otherwise there is risk, that other running steppers are running out of commands in
  // the queue. If this takes longer, then the function should be offloaded and
  // return the new status, after the enable/disable function has been successfully
  // completed.
  void setExternalEnableCall(bool (*func)(uint8_t enablePin, uint8_t value));
```

Francois71828, 2020-12-21:
Hello Gin,
great !

I have therefore just ordered on aliexpress  breakout boards with MCP23S17 Bidirectional 16-Bit I/O Expanders with SPI  interface.

It might however take 3 to 4 weeks to arrive; I will do the test when I will receive them. 

For the callback technique, do you confirm that the timing requirement << 4ms is on the response time of the callback function upon being called (via the interrupt),  and therefore independant of eventual delays in the "loop" section of the arduino program ?

gin66, 2020-12-21:
yes. if you return the old status, the callback will be called again and again until the new (expected) status is returned. If it was an enable request, then the motor will start.

If I member well, for the disable request, it will not be repeated....

gin66, 2020-12-23:
have updated master for repeating disable request until succeeded.

gin66, 2021-01-03:
feature is implemented. If there is a bug, please open a new issue.

closed 2021-01-03 by gin66