gin66/FastAccelStepper #15: Synchronise steppers

Speedjf37, 2020-12-04:
Howto synchronise multiple steppers ?
For start move 2 steppers
 send a command wait Extern volatile int Synch ?

JF

gin66, 2020-12-04:
This is a bit tricky. That's why X-Y-axes control, which requires synchronized motors, is not recommended.

The are actually two issues to be solved:
1. Both steppers should start simultaneously (if run up to e.g. 50kSteps/s, a start time delay between two steppers of 20µs would already one delta step).
2. The ramp should be synchronized, which is currently not the case.

Ín regard to issue 1:
For avr, this is relatively easy to control, because both steppers depend on the same timer.
For esp32, every stepper has its own counter. I would need to check, if all timers can be "zero"ed with one command or at least within a short time frame.

In regard to issue 2:
I do not see a chance to make the ramp generator for two steppers sync well. That's why a different approach would be needed. In the idea list I have a task to come up with an XY-ramp generator (which could be extend to more steppers). In the moment I would think, the avr is not powerful enough for this. This XY-ramp generator would then create raw commands directly to the stepper queues, which will stay in sync as long as the queues are filled. A delta time between steppers due to issue 1 will stay constant. Anyway, this is future, I work currently on the quality and my biggest problem in the moment is regression test on HW. Just in process to order PCB to support 6 steppers for esp32 and 2 steppers for avr, and not too many cables on a breadboard.

Anyway, could you please describe your use case, so I better understand your requirement. Eventually the current library fulfills your need, and i only think too complicated ? And which platform you are looking at: avr or esp32 ?

gin66, 2020-12-13:
stale issue

closed 2020-12-13 by gin66