gin66/FastAccelStepper #16: Compiler flag for other Timers

ixil, 2020-12-10:
Hello, I've been using the library for a project at my Uni - and forked it to so I can use the `TIMER4` on a MEGA2560. At the moment it is only functionally I just use `sed -i -f regexes.sed ./src/*ISR_avr.cpp` to do so. See ixil/FastAccelStepper. (The other difference were just when I was testing it).

Would you be interested/would it be possible to use some preprocessor magic (or better, modern c++) so I do not need to follow your fork?
Thanks for the useful library

gin66, 2020-12-10:
Please check out [atmega2560 branch](https://github.com/gin66/FastAccelStepper/tree/atmega2560)

Here the timer module 4 is hard coded.

I am searching standard way, how the application programmer can configure a library in arduino and platformio.

In platformio I can define preprocessor variables in platformio.ini. Not sure, how to do this in arduino and what is the standard way of doing so.

Background:
For mega2560 I like to be able to use all four 16bit timers. If this is hardcoded in the library, for every timermodule the respective ISRs are defined. This yields a collision, if the application or another library wants to use another timer module and defines its own ISR.

gin66, 2020-12-10:
BTW: apparently the atmega2560 can support even three steppers per timer due to a third compare unit.

gin66, 2020-12-10:
Released 0.13.0 with atmega 2560 support and three steppers, linked to timer 4
Hope it works for you

ixil, 2020-12-10:
That was super quick of you! I will have access to the robot on Monday (I hope) so I can test my existing stuff with that then - I can't really test from home sorry. 

> In platformio I can define preprocessor variables in platformio.ini. Not sure, how to do this in arduino and what is the standard way of doing so.

The Arduino platform is regrettable really. 
From the other libraries I surveyed, they normally just require a flag `#define USECHANNEL1` in the `.ino` or a combination of class templates combined with more preprocessor magic to make the register/bit/ISR strings.

In pseudo cpp, maybe something like:
```
#define STRING(T) "T" /*my macro magic skills are poor */
enum class ChannelEnum { A, B, C};
template <size_t timer> 
AVRAccelStep { AVRAccelStep(ChannelEnum channel){ switch(channel){}}; };
```

edit: looking at your code, I think you handle it better than what I could come up with. I don't really know how to fix the Scoping with the Macros and have it encapsulated.

ixil, 2020-12-10:
> Background: For mega2560 I like to be able to use all four 16bit timers. If this is hardcoded in the library, for every timermodule the respective ISRs are defined. This yields a collision, if the application or another library wants to use another timer module and defines its own ISR.

Looking at the avr library, you could do a `#ifdef _VECTOR(41)` and through an `#error` I'm not sure that's particularly scalable though.

See `avr/sfr_defs.h` and the `avr/io.h`

gin66, 2020-12-10:
The issue is still, that the library would need to source a header file from the application. Just have opened arduino IDE and I could not find any place to add a compiler define, which would be applied to the whole project including libraries. In platformio, this is feasible.

And for sourcing an application header file into the library, I do not know, if there is any kind of standard like this. Or if the library manager (managing all the libraries) would reject this. I have had this before and the culprit was, that symbolic links are forbidden.

The avr/sfr_defs.h and the avr/io.h is part of the avr ecosystem. FastAccelStepper is a library, which is not part of that ecosystem and not part of the application.

gin66, 2020-12-10:
The version 0.13.1 will try to include an application defined header-file using this:

```
#if __has_include("fas_config.h")
#include "fas_config.h"
#endif
```

Now I wait till it shows up in library manager.

gin66, 2020-12-15:
This header file inclusion has not worked.

So 0.14.0 evaluates a preprocessor flag called FAS_TIMER_MODULE.
With platformio just add it to build_flags in platformio.ini.

For Arduino-IDE, I don't know.

ixil, 2020-12-15:
Robots were unavailable, so I'm not sure when I'll be able to test (covid + Christmas etc) - possibly tomorrow

> For Arduino-IDE
I imagine with Arduino-IDE enduser just needs to do `#define FAS_TIME_MODULE` at the top of .ino sketch.

ixil, 2020-12-17:
The pin mappings look incorrect to me, I've sent a pr: #19

ixil, 2020-12-17:
In the pull-request I have updated the build target to the arduino atmega, rather than plain ATMega2560

gin66, 2020-12-17:
Thanks for the pull request. Already integrated.

So this issue can be closed ?

ixil, 2020-12-17:
I think the issue can be closed, I am having problems using the newer version of the library, but that's not this issue

ixil, 2020-12-17:
Maybe this should be reopened: I seem to be hitting some under behavior (board is continuously resetting, and I'm not sure why yet)
So until I've tested this should probably be open

gin66, 2021-01-03:
stalled. if the reset issue is related to FastAccelStepper, then please open a new issue.

closed 2021-01-03 by gin66