gin66/FastAccelStepper #17: Interrupt  other PCNT

Speedjf37, 2020-12-12:
Hi ,

Use 3 PCNT for Quadrature Encoder with ISR for overflow. (run ISR Overflow OK)

Use your LIB with succes for 2 steppers and 1 frequency module.

Lost somes Interrupts on PCNT encoders overflow when FastAccelStepper  -> engine.stepperConnectToPin

Howto fix it ?
(change priority or insert the ISR Quadrature Encoder  in your LIB )

Thanks JF

gin66, 2020-12-12:
Hi,

what comes to my mind:
1.  isr service is installed twice and it is not clear, which one is valid and which config is applied. Which parameters are you handing over to `pcnt_isr_service_install(ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_IRAM);` ?
2. Run time of all pcnt-ISR services is too long, if your and mine interrupts occur on same time. BTW your overflow interrupts happening how often ?

BtW: you mention, that you loose interrupts. Only during call of stepperConnectToPin or permanently ? And you loose all or only some ? Steppers are working in this condition ?

Speedjf37, 2020-12-12:
FastAccelStepperEngine engine = FastAccelStepperEngine();
 FastAccelStepper *Fstepper[MAX_STEPPER];
OK
 before FastAccelStepper -> engine.stepperConnectToPin no Int loose.

After one engine.stepperConnectToPin  some Int loose .
Int for PCNT in Half_encoder mode with 176 step count (rotate axe).
more loose when speed >=1000 rot/mn .

Time count =(30 to 100ms )1T loose for 2T or 5T (1 to 5 five Interrupt)

gin66, 2020-12-12:
Thanks for explanation. Apparently the interrupts do not cooperate so well. 

Do you use a lib for the other three pulse counters, or do you manage those interrupts in your app ? In the latter case, how do you call pcnt_isr_service_install() ?

Speedjf37, 2020-12-12:
I use this lib:
https://github.com/madhephaestus/ESP32Encoder

gin66, 2020-12-12:
Thanks. This lib and FastAccelStepper are not compatible.

ESP32Encoder takes over the whole PCNT-module and installs an own interrupt handler.
If you call afterwards FastAccelStepper's connectToStepperPin(), then the ESP32Encoder will stop working.

Solution: ESP32Encoder author should not take over the PCNT-module interrupt. Espressif offers a cooperative solution: use pcnt_isr_service_install() and with pcnt_isr_handler_add() routines ca be registered to manage one channel's interrupt request.

Speedjf37, 2020-12-12:
Ok thanks for this explain!
It's a little bit hard to modifiethe .
I don't now howto do !

gin66, 2020-12-12:
It is not rocket science to change that. Perhaps you can file an issue with the ESP32Encoder's github and he will implement this change.

closed 2020-12-12 by gin66