gin66/FastAccelStepper #20: "Unsupported board" error for Uno and Pro Mini

arion-p, 2020-12-20:
Using the library on Arduino Uno or Arduino Pro Mini fails with "Unsupported board" error.

gin66, 2020-12-20:
The check for the two boards nano and atmega2560 have been removed in 0.15.1
Please check, if it works now for you

arion-p, 2020-12-20:
Now I get:
>.pio\libdeps\Uno\FastAccelStepper\src\StepperISR_avr.cpp:41:25: error: 'stepPinStepper1A' was not declared in this scope

In StepperISR_avr.cpp, line 11 should be:
`#if defined(__AVR_ATmega328P__)`

ixil, 2020-12-20:
I didn't modify the flags for the other board definitions than the 2560, sorry @gin66 
`__AVR_ATmega328P` should be  `ARDUINO_AVR_ATmega328P` as per https://github.com/platformio/platform-atmelavr/blob/master/boards/ATmega328P.json

It is hard to tell from my phone however if that's everything

arion-p, 2020-12-20:
`ARDUINO_AVR_ATmega328P` doesn't seem to work.

gin66, 2020-12-20:
Updated to 0.15.2
Now the code does not check for specific board types, but only for CPU

arion-p, 2020-12-20:
Working fine here, both Uno and Pro Mini

closed 2020-12-20 by gin66