gin66/FastAccelStepper #22: (suggestion) Include setDelayToEnable and setDelayToDisable in usage example to fix TMC5160/etc. not moving

laminarturbulent, 2020-12-27:
I am using a TTGO T-Display ESP32 and a TMC5160 SilentStepStick v1.5.

It took me a while before I found out what was making the [StepperDemo](https://github.com/gin66/FastAccelStepper/tree/master/examples/StepperDemo) example work while the [usage example](https://github.com/gin66/FastAccelStepper#usage) didn't (it would enable the driver but the motor did not move).

The answer was actually right under the usage example, but I missed it:
> Some drivers need time to e.g. stabilize voltages until stepping should start. For this the start on delay has been added. See issue #5.

While trying to find the problem, I found that getCurrentPosition() would stay at 0 after a move() command when I didn't have the enable delays set. After adding
```
stepper->setDelayToEnable(50);
stepper->setDelayToDisable(1000);
```
to the usage example, everything worked as expected. I think it'd be good to add this to the usage example so it will work with more stepper drivers without needing modification. 

It was confusing when getCurrentPosition() stayed at 0 because it seems like the ESP32 was actually aware that the driver didn't move the motor, but I thought that this shouldn't be possible since the driver doesn't send any information back to the ESP32 (at least not to FastAccelStepper, I should also mention that I'm using TMCStepper with software SPI).

gin66, 2020-12-27:
Thanks for the suggestion. Just what you mention, that the getCurrentPosition() stayed at 0 is unexpected. So I prefer to call it a bug. Now I have checked this on my hardware setup and can confirm, it does not work as expected without the delays.

The bug has been fixed in the 0.16.1 and a new example called UsageExample.ino has been added.

Sorry for the inconvenience

laminarturbulent, 2021-01-01:
Awesome, thanks for fixing this so quickly!

closed 2021-01-01 by laminarturbulent