gin66/FastAccelStepper #23: Deceleration in applySpeedAcceleration() don't work.

FP2K-Minske, 2020-12-31:
I'm working with a potentiometer, setSpeed and applySpeedAcceleration. While the acceleration works perfect (by adding speed), there isn't an existing deceleration when decreasing the speed. It's jerky and jumpy getting slower.
With the methods "runForward" and "runBackward" it decelerates perfectly. Is it meant to be like that or isn't it implemented yet?

gin66, 2021-01-01:
speed changes are applied in a similar way to runForward()/runBackward() and move/moveTo(). So there should be no difference and jerky/jumpy getting slower should not happen.

Questions back:
- which library version and uC are you using ?
- you read the potentiometer, translate it in speed and forward to setSpeed() followed by applySpeedAcceleration(). Right ?
- what is your acceleration value ?

Today evening I can check on my hw.

gin66, 2021-01-01:
btw: which values or range for v do you use ?

FP2K-Minske, 2021-01-01:
- I'm using the 0.16.2 from the arduino library with an arduino nano (atmega328).
- Right. I'm reading the potentiometer, save the value in the variable "val" and calculate my desired speed with the formula "mySpeed = myMaxSpeed / ((double) val / 1023)". Basically getting a double between 0 and 1, so I can divide my maxSpeed to get a value between my maximum speed and infinity. My maxSpeed is 150 us/step, so I should get around 400rpm. My range for the speed is about 150-5000 (which I set as a max. value for the speed - after that I turn off the motors).
- My acceleration value is 500. I tried to lower it to 100, but it doesn't matter.

Thank you for the fast support.

gin66, 2021-01-01:
Have tried on my setup and can confirm this bug. After further investigation have found out, that this is related to the ramp generator and similarly impacting esp32.

The fix is implemented in 0.16.3

My test suite did not catch it, because only speed increase while running has been checked.
=> Now I have added a test case for speed deceleration while running.

My manual testing for release 0.16.1 and 0.16.2 has not identified the issue, because my build system is not reliable for changes in library code and used old, still working code.
=> Now the pio_dirs build system avoids use of symbolic links to refer to the library code. Similarly the tests build system with Makefile avoids symbolic links, too. Hope this way, `pio run` and `make` correctly build related object files on library file changes.

FP2K-Minske, 2021-01-01:
Works perfect, thank you!

closed 2021-01-01 by FP2K-Minske