gin66/FastAccelStepper #26: getCurrentPosition causes a block

TarekTaha, 2021-01-06:
It seems that using getCurrentPosition() causes the program to freeze/block. This is the code I am using:

 ```
#include "FastAccelStepper.h"

// As in StepperDemo for Motor 1 on AVR
#define dirPinStepper    9
#define enablePinStepper 10
#define stepPinStepper   8  // OC1A in case of AVR

// As in StepperDemo for Motor 1 on ESP32
//#define dirPinStepper 18
//#define enablePinStepper 26
//#define stepPinStepper 17

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {
  engine.init();
  stepper = engine.stepperConnectToPin(stepPinStepper);
  if (stepper) {
    stepper->setDirectionPin(dirPinStepper);
    stepper->setEnablePin(enablePinStepper);
    stepper->setAutoEnable(true);

    stepper->setSpeed(1000);  // the parameter is us/step !!!
    stepper->setAcceleration(10000);
    //stepper->move(7000);
    stepper->runForward();
  }
  Serial.begin(115200);
  Serial.print("    F_CPU=");
  Serial.println(F_CPU);
  Serial.print("    TICKS_PER_S=");
  Serial.println(TICKS_PER_S);  
}

void loop() 
{ 
    delayMicroseconds(100);
    if (stepper->isRunning()) 
    {
      Serial.print("@");
      Serial.print(stepper->getCurrentPosition());
    } 
}
```

Any idea why this might be happening?

gin66, 2021-01-06:
No idea. How quickly, does this problem occur ?
I am using your code and it runs couple of minutes without an issue.

What I have observed: if I close and open the USB serial interface, the uC will reset. But this should be expected.

Could you please share more details about this ?

TarekTaha, 2021-01-07:
basically if I comment out the Serial.print(stepper->getCurrentPosition()); the motor spins, otherwise it doesn't move at all.

I do understand that opening and closing the USB interface resets the uC, but as you mentioned that's expected.

Any reason why the stepper->getCurrentPosition() could cause a block?

I am using Arduino Mega (Atmega 256 board) btw

gin66, 2021-01-07:
Unfortunately I do not own an Atmega 256 board. With the arduino nano board it runs well.
So I am a bit left out in the dark. In order to exclude the Serial interface. Could you please check this ?
```
void loop() 
{ 
   delayMicroseconds(100);
   if (stepper->isRunning()) 
   {
     stepper->getCurrentPosition();
   } 
}
```

And in addition, how is the StepperDemo example working for you (after adapting the pin assignments to your setup) ?

ixil, 2021-01-07:
FWIW I've started to suspect that Serial breaks various things when interrupts get (correctly) switched on and off - There was a bug I saw in (Arduino/ArduinoCore-avr) but I can't find it again that may be relevant.

gin66, 2021-01-07:
Any change in behavior on your hardware without Serial.print or using a temporary variable as you have proposed ?

ixil, 2021-01-07:
I don't have a motor to test with (just the mega), but I'll put together a sketch now and see if I can test it.
Digging a bit deeper in the code, I don't think my previous thoughts should be the underlying problem - so I'm leaning towards bugs in the Serial libraries of ArduinoCore.

gin66, 2021-01-07:
Thanks for feedback. Please let me know, how it works with a motor

ixil, 2021-01-07:
Please see the updated comment above, unless @TarekTaha sees something different, I would assume it's just the stepPinStepper.

Investigating the `stepPinStepper 8` case more, by putting a:
`Serial.println("pre engine"); Serial.flush();` before `engine.init();`
shows the micro is constantly resetting, i.e. the serial monitor shows "Pre engine" being printed repeatedly. 

Maybe some `static_assert()`s could help with users debugging the problem?

gin66, 2021-01-07:
Hi ixil,

haven't noticed, that you have jumped in. That pin 9 is not working, is understandable. In the sketch is defined:
```
#define stepPinStepper4A 6  /* OC4A */
#define stepPinStepper4B 7  /* OC4B */
#define stepPinStepper4C 8  /* OC4C */
```
So pin 9 is not known to the driver. If the assignment is correct, I am not sure. I too have problem to understand the Arduino way of naming pins. For esp32 is more straightforward.

What I too do not know, if the timer 4 or those pins have specific use in the board/arduino core SW. 

The support of compare OCxC should work from the data sheet, from what I can see. And the code is pretty generic in regard to exchange A and B and C.

ixil, 2021-01-07:
Afaik timer4 is free by default ([Serial2 uses the same port and timer](https://www.arduino.cc/reference/en/language/functions/communication/serial/).
However, Timer1 is used for the Serial and delay() counter.
I made a quick PR that doesn't solve the mystery of pin8 on the Mega, but may help clear up some confusion for users.

gin66, 2021-01-08:
Now I am using simavr for testing. Perhaps over the weekend I can look into test of the atmega2569 derivate and especially timer C.

Thanks for the PR. I haven't had time to look at it. In the moment it generates under github actions a build error.

ixil, 2021-01-09:
I had tried to use simavr to work out what I was doing wrong, but didn't get very far, I think I have quite a bit to learn from you! The PR is a very minor thing but should build now.

gin66, 2021-01-09:
Now all timers and channels are checked in the simulator, which includes getCurrentPosition(). @TarekTaha: please check again, if the latest version works for you

gin66, 2021-01-14:
stalled: if any issue comes up again. Please open a new issue

closed 2021-01-14 by gin66

TarekTaha, 2021-01-18:
@gin66  I just had the chance to test it and it works!
Thank you