gin66/FastAccelStepper #27: Stepper is not run smooth at slower rpm

bkapadia01, 2021-01-07:
When comparing the stepper motor movement between Arduino Nano using the Accelstepper library and the use of ESP32 with the FastAccelStepper library the motor does not as smooth with the FastAccelStepper library. Is this because this library is optimized  for use with faster speeds? We plan to use speeds between 1-5rpm.

gin66, 2021-01-07:
Please be more specific about "not smooth", because in my setup I currently cannot see, what you mean. 

The library should be processor tick exact in regard to the generated pulses during coasting. So the reason is not about being optimized for faster speed. The optimization compared to AccelStepper is the use of the processor timers to generate the pulses.

Questions back:
* Which library version do you use 
* Can you share some code, revealing the problem ?
* Have you crosschecked with StepperDemo in the examples, if steppers still do not run smooth ?
* What the driver is not able to cope with: if interrupts are disabled for too long time or other tasks prevent the StepperTask from running. Could this be the reason in your application ?

gin66, 2021-01-07:
But what I do observe, that at low speed 200steps/s, the running sound is not as steady as I would expect it.

gin66, 2021-01-07:
Crosschecked with AccelStepper:
```#include <AccelStepper.h>

AccelStepper stepper(AccelStepper::DRIVER, 17,18); // Defaults to AccelStepper::FULL4WIRE (4 pins) on 2, 3, 4, 5

void setup()
{  
  pinMode(26,OUTPUT);
  digitalWrite(26,LOW);
  
   stepper.setMaxSpeed(1000);
   stepper.setSpeed(200);	
}

void loop()
{  
   stepper.runSpeed();
}
```

no difference compared to running with StepperDemo using `M1 A10000 V5000 f`

bkapadia01, 2021-01-09:
Thank you for the quick response and for validating on your end as well.
After doing some more testing we noticed that the behaviour using fastaccelstepper with ESP32  is similar to that of accelstepper library with the Arduino. 
It appears that this not an issue and the fastaccelstepper is working as expected. 👍 

---
Regarding your crosscheck running with accelstepper on the ESP32, I do observe intermittent pause/stuttering while the motor is running, using the same code you provided at `setSpeed(200)` and more noticeable at `setSpeed(26.6667)` for 1rpm...just curious if you observed this? Not an issue though, we suspect that accelstepper library does not work as well with ESP32. 

We do notice some odd behaviours with accelstepper library with the esp32 but the fastaccelibrary 
> Which library version do you use?

Using version 0.16.4 

> Can you share some code, revealing the problem ?

For test purpose I'm using the stepperDemo  @ `V=35000`, we plan to implement code that allows the motor to run between 1-5RPM at constant speed.
   
> Have you crosschecked with StepperDemo in the examples, if steppers still do not run smooth ?

Yes,I am using the stepperDemo example and Setting A=100, V=35000 and continuous running forward and comparing it to the Accelstepper with the Arduino Due also running at 1rpm. They work as expected now.
    
> What the driver is not able to cope with: if interrupts are disabled for too long time or other tasks prevent the StepperTask from running. Could this be the reason in your application ?

I was using stepperDemo but we will be using multitasking to read the quadrature encoder attached to the motor while the motor is running to read the angle of the motor as it spins. Would you happen to  have any insight/example code for multitasking with the ESP32 or know of any issues you've come across in the past with multitasking using the fastaccelstepper library? Thanks!

gin66, 2021-01-09:
Nice, that it works for you. While using simavr, I have identified, that even while running constantly, the queue has run out of commands. Due to bugs on restarting the queue, this has lead to some longer pauses and may cause stutter. As solution for now I have increased the required time in the queue to be 20ms instead of 10ms - for the esp32, too. And I still plan to increase the minimum amount of commands in the queue to be 2. Even though, with rework of the interrupt code on esp32/avr, an temporarily empty queue should now managed much better without impact on smooth running.

For multitasking on esp32, the stepper task is run with highest priority. But I have not tested on a fully loaded machine. Perhaps I even should tie it to the app core. Current task creation is this:
```
#define STACK_SIZE 1000
#define PRIORITY configMAX_PRIORITIES
  xTaskCreate(StepperTask, "StepperTask", STACK_SIZE, this, PRIORITY, NULL);
```

As there is currently no issue here to work on, I close this.

closed 2021-01-09 by gin66

bkapadia01, 2021-01-10:
Thank you for the response and further testing with the simavr.
Regarding increasing the required time in the queue and increasing the amount of commands is that a change that is already implemented or is there a separate ticket that will be created for that? 
Thanks for clarifying the multitasking of stepper using the ESP32.

gin66, 2021-01-10:
That change is already implemented. For the esp32, this is fine. For the avr, the high interrupt rate (every pulse one interrupt) still causes an issue, which I am trying to identify the root cause.