gin66/FastAccelStepper #29: Direct control of the acceleration independent of speed and position

SorenBNielsen, 2021-01-08:
Can the interface to the stepper driver be expanded so it become possible to directly control the acceleration of the motor independent of it's speed and position. 

This will enable to use the steppers as acceleration devices (like DC-motors), which is used in many closed loop control systems.

gin66, 2021-01-08:
In my opinion, it is implemented like this. Just use setAcceleration() and call applySpeedAcceleration(). 
Or I do misunderstand your requirement.

SorenBNielsen, 2021-01-09:
The requirement for the acceleration device is: No matter of what the actual position, speed and direction of the speed is, shall the device accelerate (decelerate) as specified in the "setAcceleration". 
This correspond to what happens in a DC motor, where the current determine what the acceleration of the motor is (independent of the position speed and direction of the speed).

SorenBNielsen, 2021-01-09:
You are right, that it probably may be implemented with the present setAcceleration, but you shall then handle the speed direction (and acceleration direction). It will be very usefull just to specify that the device shall accelerate (or decelerate) no matter what the speed and direction of the speed is.

gin66, 2021-01-09:
The acceleration is in effect only during ramping up or down. If during this time  the acceleration is modified, then it is immediately put into effect upon call of applySpeedAcceleration(). During manual testing I often use accelerations around 10 to 100 to observe the increase or decrease of the step rate. If in this time, the acceleration is changed, it has immediate effect, as I would expect.

Please check, if this works for you.

SorenBNielsen, 2021-01-09:
Yes, right, I understand and it works during ramp up and down from/to zero speed. This is fine for position - and speed controllers. 

What is needed to have an acceleration device is that it is possible to set both a positive and a negative value to setAcceleration that will accelerate (or decelerate) and keep on doing that no matter what the speed and speed direction is. 

It shall of course be possible to set a maximum speed.

gin66, 2021-01-09:
What I understand is, that you need the possibility to have negative acceleration values.

The point is, that the acceleration is _always_ towards a maximum target speed, which limits the speed. If the acceleration stays positive, it will ramp up till maximum speed. If the acceleration turns to negative, then the target speed needs to change, too. If the direction does not changed, then this sequence in the application should fulfill your need:
```
if (new_acceleration > 0) {
   stepper->setAcceleration(new_acceleration);
   stepper->setSpeed(MIN_DELTA_TICKS);
   stepper->applySpeedAcceleration();
}
else if new_acceleration < 0 {
   stepper->setAcceleration(-new_acceleration); // only positive allowed
   stepper->setSpeed(0);
   stepper->applySpeedAcceleration();
   }
else {
   stepper->setSpeed(stepper->getPeriodAfterCommandsCompleted());
   stepper->applySpeedAcceleration();
} 
```

ixil, 2021-01-09:
If I may jump in here with a comment. The behaviour you are describing
sounds a bit like what I would expect from something like a motor with
current control (a motor being a DC or AC motor - *not* a stepper.
Only to some extent can you replicate this in a stepper motor - only up to
a point as you can command individual steps.

i.e. Stepper motors use position offset command which is limited by the
control bandwidth - you can't accelerate forever with a stepper motor as
the *controller and pulses have a maximum frequency they can be run at* -
you could continue to accelerate until that maximum frequency - but then
your micro will be unable to do anything else but that (assuming the
motor's dynamics can handle that speed - but that's a different matter).

In summary, what I'm trying to say is, from the behavior you describe it
sounds like you would be better off using a different type of motor
altogether. But you can also just use an integrating loop as gin66 shows to
get what you want, just that you should be aware the controller run into
timing issues at very high speeds.

SorenBNielsen, 2021-01-09:
@gin66 : Yes, it will work. I will try it out and verify it in practice. 
@ixil: You are right, that it may be a better case to use another motor, but the steppers are cheap, precise and can be used as a good and stable replacement for DC-motors. We use them as "force devices", when we have constant masses that shall be moved.

gin66, 2021-01-10:
The API could be extended with:

`moveByAcceleration(int32_t acceleration, bool allow_reverse=true)`

This will simplify the not so obvious call `stepper->setSpeed(stepper->getPeriodAfterCommandsCompleted());` and would either go to stop or reverse.

gin66, 2021-01-10:
Please check version 0.17.0 for the implementation.

The code above has had the drawback, that the max_speed was lost on acceleration=0.

In StepperDemo the new feature can be tested. Try this sequence:
```
M1 V100
a10
a20
a100
a0
a-10
a-100
a0
S
```
First line will set the maximum speed. Then the accceleration is set to +10 and the motor starts moving. With a20 it accelerates faster and with a100 even more. With a0 will keep the speed. With a-10 will slowly decelerate and with a-100 even more. Eventually motor will stop and run in negative direction towards max speed. With a0 the current speed will be kept again and with S the motor decelerates from any current direction to motor stop

Hope this is the appropriate feature for you

SorenBNielsen, 2021-01-10:
Wooow ... you are a wizard. I am impressed - not only with this extension - but with the whole project you have implemented here. Thank you for this. 
I will try it out tomorrow and send some feed back.

SorenBNielsen, 2021-01-11:
I have downloaded the latest code and tried to use the UsageExample and modified it to be used on my esp32 system. 

The problem is that the motor is not running smoothly, see the attached video, so I cannot try the new acceleration functionality. 

I am using this code: 

#include "FastAccelStepper.h"

// As in StepperDemo for Motor 1 on AVR
//#define dirPinStepper    5
//#define enablePinStepper 6
//#define stepPinStepper   9  // OC1A in case of AVR

// As in StepperDemo for Motor 1 on ESP32
#define dirPinStepper DIR_PIN
#define enablePinStepper PIN_UNDEFINED
#define stepPinStepper STEP_PIN

#define ARDUINO_ARCH_ESP32
#define DIR_PIN           33 // Direction
#define STEP_PIN          25 // Step
#define N_EN_PIN          27 // Enable pin
#define N_RESET_PIN       12 // Reset pin 
#define N_SLEEP_PIN       26 // Sleep pin 
#define MODE0_PIN         13 // Mode0 input for micro stepping on DRV8825
#define MODE1_PIN         14 // Mode1 input for micro stepping
#define MODE2_PIN         15 // Mode2 input for micro stepping

// Define IO pins on ESP32 / DRV8825
void defineIO() {
  pinMode(N_EN_PIN, OUTPUT); digitalWrite(N_EN_PIN, LOW);         // Enable stepper driver 
  pinMode(STEP_PIN, OUTPUT); digitalWrite(STEP_PIN, LOW);         // Set step pin low
  pinMode(DIR_PIN, OUTPUT); digitalWrite(DIR_PIN, LOW);           // Set direction pin low
  pinMode(MODE0_PIN, OUTPUT); digitalWrite(MODE0_PIN, LOW);                                    // Set MODE pins to output
  pinMode(MODE1_PIN, OUTPUT); digitalWrite(MODE1_PIN, LOW);
  pinMode(MODE2_PIN, OUTPUT); digitalWrite(MODE2_PIN, LOW);
  pinMode(N_RESET_PIN, OUTPUT); digitalWrite(N_RESET_PIN, HIGH);  // Set not_reset pin high
  pinMode(N_SLEEP_PIN, OUTPUT); digitalWrite(N_SLEEP_PIN, HIGH);  // Set not_sleep pin high
}

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {
  defineIO();
  engine.init();
  stepper = engine.stepperConnectToPin(stepPinStepper);
  if (stepper) {
    stepper->setDirectionPin(dirPinStepper);
    stepper->setEnablePin(enablePinStepper);
    stepper->setAutoEnable(true);

    // If auto enable/disable need delays, just add (one or both):
    // stepper->setDelayToEnable(50);
    // stepper->setDelayToDisable(1000);

    stepper->setSpeed(1000);  // the parameter is us/step !!!
    stepper->setAcceleration(100);
    stepper->move(1000);
  }
}

void loop() {}

https://user-images.githubusercontent.com/39032923/104185243-00d5f380-5415-11eb-87b0-a56383b72208.mp4

gin66, 2021-01-11:
Please change to:
```
pinMode(MODE0_PIN, OUTPUT); digitalWrite(MODE0_PIN, HIGH); // Set MODE pins to output
pinMode(MODE1_PIN, OUTPUT); digitalWrite(MODE1_PIN, HIGH);
pinMode(MODE2_PIN, OUTPUT); digitalWrite(MODE2_PIN, HIGH);
```

Please check [Issue 2](https://github.com/gin66/FastAccelStepper#lessons-learned) as reference

At 0:02 the motor is driving backwards. You can measure with an oscilloscope the direction pin/line, but you will not find a reverse polarity on the direction line. Or to convince you more, comment this out:
```
// stepper->setDirectionPin(dirPinStepper);
```

If there is no reverse polarity, then how can the stepper run backwards from the library code ?

If you measure the step pin/line, then you will not see any glitches from the pulse generation. At least I don't see any on my cheap oscilloscope.

On my setup, the motor is running without microsteps at some speeds even more erratically than your video, the more microsteps the better. For this phenomenon I have no good and convincing explanation till now. But I am sure, this is not from the library.

SorenBNielsen, 2021-01-11:
Thank you for your fast feed back. 
You are right. I have also checked on my scope, and it seems as nothing is wrong with the library and I can verify that it runs ok with higher micro stepping (>2). 
My best guess for the cause of this problem is that there is a resonance in the steppers when running full step that causes these "missing steps", we see as a backward movement. I can verify that the "resonance" changes when I use a NEMA23 instead of a NEMA17. 
I will be back when I have tested the acceleration device.

SorenBNielsen, 2021-01-11:
One more issue: After the UsageExample is started, the library is sending 3 pulses to the stepperdriver, see the attached picture (sorry for the lack of quality)

[image]
. 
Not a big issue but not very nice.

gin66, 2021-01-11:
agree. Will check later on my side.

gin66, 2021-01-11:
Found the root cause and fixed it. Not so happy about this, because the mcpwm/pcnt modules of the esp32 often behave not as I would expect. So I am not 100% sure, that a future change will break it again.
Anyway, for the future I plan to use an available counter of the esp32 as up/down-reference counter during test. Then I should be able to catch it.

Thanks for mentioning, because lost/too many pulses should not happen in this driver.

I will make the fix available in 0.17.1 soon, but maybe not today.

SorenBNielsen, 2021-01-11:
I have verified that the the StepperDemo works as expected for control of the acceleration. 
It is excellent and very promising and I am looking forward to integrate FastAccelStepper into one of our controllers/models and verify that is keep on being stable and precise when pushed to the limits (I will try our reverse pendulum model for it). 
It will take some time before I can do this, so I will close this issue for now and send some feed back from the practical test results. 
Thank you for your excellent support and a great library.

SorenBNielsen, 2021-01-11:
There is a problem with the acceleration. 
Try the sequence: 
V30
a30000
a-30000
a40000

It seems as the latest acceleration is ignored. 
If you then give a a30000, the acceleration control works again. 
So something seems to be wrong.

gin66, 2021-01-12:
Can confirm this.
Alternative sequence is
```
M1 V60 A40000 f w2000 A40000 b S
```
But this works:
```
M1 V60 A40000 f w2000 b S
```

SorenBNielsen, 2021-01-12:
Do you confirm that my sequence should work, and that there is a bug in the library?

gin66, 2021-01-12:
Yes. Still working on a fix for this.

gin66, 2021-01-14:
Finally. Completely reworked the esp32 ISR code and checked with oscilloscope. There have been a couple of issues linked to the pulse counter and mcpwm module - at the beginning and even while running. From my testing all runs pretty stable and I think, the steppers even run smoother. The version 0.18.0 is worth a try.

Thanks for pointing these issues out. Has helped a lot to improve the driver

closed 2021-01-16 by gin66

SorenBNielsen, 2021-01-30:
Sorry for my very late feed back. 
I have verified the acceleration control and it works like a charm. 
I am impressed. 
Thank you for an excellent lib.

gin66, 2021-01-30:
Thanks for the feedback.