gin66/FastAccelStepper #33: ESP32 forceStopAndNewPosition

pavel-b-kr12, 2021-01-14:
often forceStopAndNewPosition(0) does not set position to 0. In that cases also something breaks and library stop to respond on moveTo function. Need more testing.

gin66, 2021-01-14:
which version and uC have you used ?

pavel-b-kr12, 2021-01-15:
few-days ago version from git, ESP32 like this https://www.cytron.io/image/catalog/products/NODEMCU-ESP32/NODEMCU-ESP32.jpg
 @ DOIT devkit v1

gin66, 2021-01-15:
Thanks for the feedback. 

First I recommend to use the latest 0.18.1 or 0.18.2. The previous versions apparently have lost steps while running. In the 0.18 path the esp32 ISR routine and mcpwm/pcnt interface have been completely revised and cross checked with an oscilloscope.

In course of this, I have made a modification to the stop routine and perhaps accidentally fixed the mentioned bug. In any case I plan to have more esp32 tests and as per your issue, will add tests for forceStopAndNewPosition(). In case the issue is not fixed, I plan to find them in my test setup.

pavel-b-kr12, 2021-01-15:
Seems now this code working: move motor between 2 limit switches.
`

       #include "FastAccelStepper.h"

      #define btn_limit_near_p			12
      #define btn_limit_far_p			14
      
      #define dirPinStepper		2	// 16, 17 are bad
      #define enablePinStepper	13			
      #define stepPinStepper		4	
      
      FastAccelStepperEngine engine = FastAccelStepperEngine();
      FastAccelStepper *stepper = NULL;
      
      void setup() {
        Serial.begin(115200);  while (!Serial);  Serial.println(F("Type any character to begin."));  while(!Serial.available()); delay(500);
        pinMode(btn_limit_near_p, INPUT_PULLUP);
        pinMode(btn_limit_far_p, INPUT_PULLUP);
      
        Serial.println("Starting");
        Serial.flush();
        engine.init();
        stepper = engine.stepperConnectToPin(stepPinStepper);
        Serial.println("Starting");
        Serial.print("Stepper Pin:");
        Serial.println(stepPinStepper);
        Serial.flush();
        Serial.println((unsigned int)stepper);
        Serial.println((unsigned int)&engine);
        if (stepper) {
          stepper->setDirectionPin(dirPinStepper);
          stepper->setEnablePin(enablePinStepper);
          stepper->setAutoEnable(true);
      
          stepper->setSpeed(500);  // the parameter is us/step !!!
          stepper->setAcceleration(10000);
          // stepper->move(7000);
          //stepper->runForward();
	      stepper->runBackward();
        } else {
          Serial.println("Stepper Not initialized!");
          delay(1000);
        }
        Serial.print("    F_CPU=");
        Serial.println(F_CPU);
        Serial.print("    TICKS_PER_S=");
        Serial.println(TICKS_PER_S);
        Serial.flush();
      }
      
      void loop() {
        delay(100);
        if (stepper) {
          if (stepper->isRunning()) {
            Serial.print("@");
            Serial.println(stepper->getCurrentPosition());
          }
        } else {
          Serial.println("Stepper died?");
          Serial.flush();
          delay(10000);
        }
        
        
        if(!digitalRead(btn_limit_near_p) )
        {
	      stepper->forceStopAndNewPosition(0);
	      Serial.println("btn_limit_far_p");
	      delay(200);
	      stepper->runForward();
	      while(!digitalRead(btn_limit_far_p));
	      Serial.println("runForward");
	      delay(200);
        }  
        if(!digitalRead(btn_limit_far_p) )
        {
	      stepper->forceStopAndNewPosition(stepper->getCurrentPosition());
	      Serial.println("btn_limit_near_p");
	      delay(200);
	      stepper->runBackward();
	      while(!digitalRead(btn_limit_near_p));
	      Serial.println("runBackward");
	      delay(200);
        }
      }
```

`

pavel-b-kr12, 2021-01-15:
also  in previous version this case lead motor to completely turn off

gin66, 2021-01-16:
Happy to hear, that it works. Thanks for sharing your test code.

Looking at your test code, I would recommend to replace the digitalRead() of the switches with a gpio-interrupt, in which forceStopAndNewPosition is called.

Rationale: the code is checking the switches in the loop and the loop starts with a delay(100). Thus every 100ms the switches are checked. With speed of 500us/step, the motor can overshoot max. 200 steps the switch trigger point.

As the original issue is fixed, the issue can be closed.

pavel-b-kr12, 2021-01-16:
hello again, last working version was 0.18.4 Now this issues come back gain.
Code above outputs:

..............
@-10231
@-10431
btn_limit_far_p
runForward
@1
btn_limit_far_p
runForward
@1
@1
@1
@1
@1
@1
@1
@1
@1
@1
@1
@1

and after call runForward motion became very slow (but it perhaps moving and working, except speed and position)

pavel-b-kr12, 2021-01-17:
more testing told that issue related to speed. It was not solved at all. But something changed in 0.18.x that cause test above passed successful with stepper->setSpeed(500); 
but it failed with speed 200 and faster

gin66, 2021-01-17:
The improvement for step loss at high speed as broken the code.
Version 0.18.11 fixes the problem.

I have created in StepperDemo a test sequence for testing. Start with `t M1 09 R I`

Nice is: the executed steps after forceStop() are reported to be zero (even though 1  or 2 are possible at higher speeds and depending on the timing).

closed 2021-01-20 by gin66