gin66/FastAccelStepper #36: How can we use library without setAcceleration?

bkapadia01, 2021-01-17:
Hi Gin66,
The motor does not run without `setAcceleration` defined but we do not require acceleration for our project. 
Is this set acceleration required? 
What is the maximum value we can set the acceleration to so the motor run immediately to the desired speed (1 to 3rev/min)?
We are trying to run the motor precisely 1 revolution in 60seconds, at a very low rpm and we find that there is very small discrepancy (0.1degrees) error with acceleration set so we are trying to remove the acceleration as a possible source of error.

gin66, 2021-01-17:
It is always the problem to define, what is a good default for a parameter. For the one set up, the default may be too high and for the other, way too high. So I have opted for being invalid until initialized.

The acceleration is an uint32_t. So the max value is 4294967295. Using this value, there will not be any ramp up or ramp down.

bkapadia01, 2021-01-18:
Thank you @gin66. I will give that a try tomorrow and will let you know how it works for us.

closed 2021-01-20 by gin66

AntoineGR, 2022-11-07:
Hello @gin66, I'm also interested in running without any accel/decel.
Unfortunately using `setAcceleration=4294967295` doesn't seem to work : the motor stands still, without any noise being heard.
Would you have any clue on this ? Thanks