gin66/FastAccelStepper #37: Motor stops running executing runBackard/runForward after calling it couple times.

bkapadia01, 2021-01-17:
Hi gin66,

Using executing runBackward() method a few times the motor stops running and does not respond to runBackward(). 

To explain our code:
We have a while loop and a timer for 30seconds. 
Every time I enter a command to `Start` the motor, the motor will runBackward() for 30seconds and then stop (forceStopAndNewPosition) and wait for the next command again. 
I will enter "Start" command again and the motor will run for 30seconds again and stop. 

The problem is that after I "Start" the motor after the 2nd or 3rd time the motor will not run. 
The timer is still running as expected and other parts of code are working as expected. 
runBackward or runForward do not respond after 2nd/3rd time calling the method.

bkapadia01, 2021-01-17:
On further testing and reviewing other #33  it seems that the issue has to do with forceStopAndNewPosition(0).
I updated the code to use `stopMove()` instead and I was able to run "Start" multiple times without any issues. 
There is definitely some issue with `forceStopAndNewPosition()`. 
I'm using `version 0.18.10`.

I've updated the title to reflect this.

gin66, 2021-01-17:
Looks like a duplicate to #33.

In the test_seq_09.cpp I constantly start the motor with runForward()/runBackward() followed by forceStopAndNewPosition(). But I have never observed, that it does no respond to a new run...ward() command. Nevertheless I hope, that the fix for #33 let this problem disappear, too.

closed 2021-01-20 by gin66