gin66/FastAccelStepper #40: setAcceleration cause wrong deceleration calculation. v0.18.11

pavel-b-kr12, 2021-01-17:
loop
{
stepper->setSpeed(variableSpeed);
//stepper->setAcceleration(9999); // uncomment this cause motor position will be over than target and motor have to go backward (same in both direction). In most bad cases it act like drunk car driver to chive target position in waves-like motion.
//stepper->setAcceleration(variableAcceleration); //same situation

long target = analogRead(A0)*5;
stepper->moveTo(target); 
}

I test on ESP32.

gin66, 2021-01-17:
Sounds interesting. Could you please add a Serial.println(target) at the end and perhaps a delay(10) ? Hope you can provide the log with timestamp. This way I can try to produce it on my side.

pavel-b-kr12, 2021-01-18:
Sorry, finished project, so no hardware. This is looks like absolutely the same as overshoot examples in google. 
The code above produce NO waves but only one return, but if acceleration is low , i think waves will appear.
The output, i got with one of the settings  with waves:
Green - target, Red - current position.
[image]

gin66, 2021-01-18:
Thanks for additional info. What is the meaning of the other three lines between 0 and 10000 ?

With this input I have tried to check on my HW (esp and avr) and could not reproduce anything close to what you get. So I have changed one of my tests to constantly issue moveTo() and setAcceleration. There has been a finding, which I have fixed, and I have added another precaution in setAcceleration. The code is uploaded to github, but not released yet.

Just I cannot explain that behavior with those changes.

Could you provide a log with timing included and your setting of speed/acceleration for the pic above ?

pavel-b-kr12, 2021-01-18:
forgot the picture above. The issue in 1st message looks like this: 
[image]

with any code that updates setAcceleration

loop
{
stepper->setSpeed(200);
stepper->setAcceleration(9999);
stepper->moveTo(20000);
delay(10);
}

gin66, 2021-01-18:
Thanks for the code. Based on this the issue came apparent and I have added a test case to my test suite.

The issue is fixed in v0.18.12, but may reappear, if the setAcceleration() would jump back and forth between 999 and 1000. Perhaps later add a test case for this.

gin66, 2021-01-18:
added a test case with alternating acceleration: no failure

closed 2021-01-20 by gin66