gin66/FastAccelStepper #43: Issue on moveTo() to exactly same target after a completion of stopMove()

macedolfm, 2021-01-26:
If one issues any moveTo i.e. "_moveTo(9999);_" then, while on the way, issue a "_stopMove()_" and wait for a **full stop**, a re-issue of "_moveTo(9999);_" will **NOT** move the motor, any other value but previous one (9999) will work just fine. If the re-issue of the same "moveTo(9999);" occurs while the stepper is still moving (deceleration phase) everything runs fine. A current workaround this issue is to double call moveTo() with +1 value, i.e. "_moveTo(value + 1);_" followed by "_moveTo(value);_"

gin66, 2021-01-26:
Thanks for the report. 
Easy to reproduce. So I have added a test case for this failure and fixed it in version 0.19.0

macedolfm, 2021-01-26:
Thank you so much for your fabulous work, best Stepper lib around, no doubt !!!

closed 2021-01-26 by macedolfm