gin66/FastAccelStepper #44: Glitches at  at some speeds

gmoneti, 2021-01-31:
Hi, 
Thank you very much for this code. I am a newbie. I am testing it on a Clearpath motor:
https://www.teknic.com/model-info/CPM-SDSK-N0563P-RLN/?model_voltage=230VAC3ph . 
It has integrated power supply + driver; it accepts standard step/dir or Quad A/B inputs. 

I'm using an Arduino Uno 16bit, and attached a digital encoder knob. 
The code takes speed commands from the knob and from monitor input.
The motor currently runs at 1000 steps/rev. I am testing at 0-1000rpm.
Motor accelerates to various speeds and reverses elegantly with decel/accel as commanded.
Knob is set to 20rpm increments per tick (count).

Problem: When adjusting speed with knob, at a few speed transitions the motor jolts (glitches) but recovers; motor software MSP warns "torque saturation" condition. Perhaps a rapid stop/start event?  Something in the frequency timing comes to mind, but I have no experience.

Examples (at 1000 steps/rev):
 Encoder cnt: 27,    from 520--540rpm,    speedInUs: 111
 Encoder cnt: 37,    from 720--740rpm,    speedInUs: 81
 Encoder cnt: 47,    from 920--940rpm,    speedInUs: 64
	
Tested: In MSP software, I changed the steps per rev from 1000 to 4000: there still were glitches but at different speeds than before.

I would like to know if someone else has experienced such glitches? And what may be the cause? Thanks.

gin66, 2021-01-31:
Thanks for the feedback and this is now a really big motor and not a tiny stepper :-).

I have tried all speed changes from 64us to 81us (64->65->64->66->64->....) with acceleration 1000 and acceleration 10000 using StepperDemo and Arduino Nano. Similarly test around 81 and 111. With this I could not reproduce the glitches.

The only thing, which comes close are few, different speed ranges, at which the motor shows some resonant noise, not caused by glitches in the generated steps.

On the other side, you mention, that you change the MSP software from 1000 to 4000 steps/revolution. I assume, this introduces microstepping. Here I have similar experience with the tiny NEMA-17: [In full step mode at specific speed ranges, the stepper even can run several step backwards](https://github.com/gin66/FastAccelStepper#lessons-learned). Here I am pretty sure, this cannot be the library, because the direction pin was not touched at all. With 2 or 4 micro steps, this reversing is gone, but still not super smooth. With 16 microsteps the stepper runs very smooth. Till now I have no good explanation. Perhaps some resonance !?

gin66, 2021-01-31:
This video shows similar [erratic reversing behavior](https://youtu.be/DsYgw3GFHZo)

[Resonance measurement](https://youtu.be/Rv8NVF6ZLlQ)

gmoneti, 2021-01-31:
Thank you for the kind response.  I do not see the erratic/resonance effects you mention, even at 1min/rev speed. I should note that this a not a stepper motor but a servo motor, a 3-phase PMSM motor (with integrated power supply and driver). It can be (is currently) configured to accept normal step/dir commands.  Also, the motor has a native 4000cpr encoder (configurable from 200 -- ~12800). And the ClearPath Step Input is “positive edge-triggered”.

The glitches arise as I move up and down the speed range by turning the digital knob (i.e., many commands in rapid succession). When setting speed via monitor command (i.e., one command) I see no glitches. 

Here is a scope image (from motor software MSP) showing speed changes (blue) and torque (red). Note that downward blue line = rising speed. The glitches are associated with a sudden drop in speed and a huge torque spike.

[image]

gin66, 2021-01-31:
Cool picture. 

From the right-most slope to the limit, the rpm increases by ~600rpm within ~1s (2,000 ms = 2000 ms!?). With 1000steps/rpm, the increase is 10.000steps/s within 1ms. This gives an acceleration of 10.000 steps/s².
Is this roundabout your acceleration setting ?

The time between speed changes is around 100ms+. Not so fast. 

Actually my test suite has one test case with speed being changed between values in order to detect lost steps. If run with the motor, all ok.

In the moment I speculate about two ideas (loud thinking):
- The ramp generator generates wrong raw commands.
- The timer compare interrupt is blocked too long (e.g. 20us). The ISR contains a mechanism to recover, but this could cause one step being late. Would look like speed drop followed by a speed increase over three steps. A normal stepper may even ignore this with bit of noise.

Anyway, I need to reproduce this.

gmoneti, 2021-01-31:
The right-most slope is a decrease to a stop (from ~600rpm)  after executing stopMove(). (Remember falling blue line = speed increase.)  Acceleration was: setAcceleration(8000); at 1000cpr = 480rpm/s. Correct? Rather relaxed. Yes, in USA, 2,000 = 2000.

Well, the speed drop and recovery is there. I wish I could be more helpful. I'll poke around some more and see if I can discover other potentially useful details.

gin66, 2021-01-31:
I have tried in my PC test suite to repeatedly apply a sinus variation as input to setSpeed() and all looks plausible. With high acceleration (1000000) and high speed (~50us) I have seen spikes in the calculated acceleration. Just this is normal, if the speed jumps from one value to another, which happens with high accelerations.

The PC test suite cannot detect an issue in the interrupt timing, so I just have done some tests using the StepperDemo and arduino nano. With a command like this:
`M1 A10000 V100 f w100 V30 U w100 V100 U w50 V80  U w50 V40 U w50 V100 U w50 V80 U V170 w50 V80 U w160 V580 U w1000 V100 U`

This sets the speed to 100us, then wait 100ms, then set speed to 30us, wait 100ms, set speed to 100us, and so on. Bit lengthy, but easy to be done without programming.

Nothing special and I can only check with the sound or abnormal force from the stepper. So I am bit stuck.

Perhaps this could help:
- Could you please extend your SW and after the setSpeed(), applySpeedAcceleration() add `Serial.print(millis()); Serial.print(' ');Serial.println(speed);` and record the serial data ?
- Or try to use the StepperDemo command version to repeat this.
- Can the MSP provide a zoomed in version of this spike with e.g. 1ms resolution or better ?

gmoneti, 2021-01-31:
Here is output from: Serial.print(millis()); Serial.print(' ');Serial.println(speed);
I spun the knob fairly quickly and continuously, but had to stop and re-grip twice (multi-turn knob). Is this what you wanted?
```
9876 20
9996 40
10129 60
10254 80
10366 100
10520 120
10648 140
10723 160
10815 180
10891 200
10988 220
11063 240
11143 260
11227 280
11307 300
11406 320
11511 340
11636 360
11759 380
11881 400
12037 420
12170 440
12241 460
13092 480
13211 500
13332 520
13477 540
13575 560
13688 580
13809 600
13937 620
14050 640
14205 660
14373 680
14503 700
14668 720
14817 740
15001 760
15133 780
15977 800
16130 820
16286 840
16492 860
16602 880
16693 900
16818 920
16887 940
17013 960
17158 980
17453 1000
17737 1000
```
I'll get to the StepperDemo findings a little later.

gin66, 2021-01-31:
Yes, That‘s the data, I am looking for. Have you encountered glitches with this data ?

gmoneti, 2021-02-01:
Yes, same as in my first post, during 20rpm jump to: 540, 740, 940 (these are the more consistent ones).

gmoneti, 2021-02-01:
I played with StepperDemo; I could not replicate the glitches.
I added a debug function to my code; it ramps the speed up then down as with StepperDemo; I set different rpm jumps and different periods between jumps; and I used the same speed setting code of my original code; but I could not replicate the glitches. Still, when using the knob, the glitches are there. I tried a new knob (same type), but same result.

Could still be some peculiarity of the knob, anyway?

I heard some interesting resonances when ramping up 1rpm per period (100ms). For each 100rpm increase, the pitch rose for 50rpm then fell back the next 50rpm, so 10 cycles to 1000rpm. Above 500rpm the pitch cycling was only faintly audible. At 10rpm jumps, less audible. At 20rpm jumps not audible (perhaps because of continuous state of acceleration at that rpm jump and period). See attachment: [debug_ramp_speed.txt](https://github.com/gin66/FastAccelStepper/files/5901635/debug_ramp_speed.txt)
BTW, the encoder knob is a KY-040.

gin66, 2021-02-01:
Thanks for the deep investigation and the data file. Hope tomorrow can check with my hw and your data file to reproduce the glitches.

It is interesting, that without rotary encoder the glitches are absent. For sure the rotary encoder itself is not the problem, but the sw reading the encoder may be. Could you please share the code of your sketch ? Do you use the RotaryEncoder library ?

For the pitch changes, in older version of the library was a bug, perhaps related to what you mention. If you use the latest one, it should be fine. Last time I have checked using acceleration of 1, the pitch is always increasing. Even though the stepper driver itself using a chopping frequency and there could be some modulation ongoing.For your motor not sure, if there is the possibility of similar effect of modulation.

gin66, 2021-02-01:
Please check again with StepperDemo v0.20.1. I have added for avr two commands (`r` and `e`), which allow to generate error causes. See for reference: (https://github.com/gin66/FastAccelStepper#error-investigation)

Would be interesting, if one of the two issues causes similar glitches on your MSP scope image

gmoneti, 2021-02-01:
I upgraded lib to 20.1. No change in my app.

I ran StepperDemo with "M1 A10000 V100 f", then tried the new"r" or "e". So, "r" generated no ouput; "e" generated ~3 motor glitches/sec. Interesting. FWIW, I don't use delay() anywhere in my code. The encoder ISR() does have a digitalRead().

I attached the encoder code---for your viewing pleasure... [rotary_sm.zip](https://github.com/gin66/FastAccelStepper/files/5907310/rotary_sm.zip)

gin66, 2021-02-02:
Thanks for the code. I have integrated the Rotary Encoder code into StepperDemo and could not find any anomaly. Could you please try on your side with the faster ISR version ? I hope, that the code change works
```
/*
 *      rotary_sm.ino   --  State machine implementation of rotary encoder driver
 *                          Interrupt driven; supports 2 types of interrupts:
 *                            * Polled interrupts; enable by defining TIMER2INT
 *                            * Pin change interrupts: enalbe by defining PINCHANGEINT
 *                              (Do NOT enable both at the same time)
 *
 *          This program is developed from the code at:
 *              http://www.buxtronix.net/2011/10/rotary-encoders-done-properly.html
 *              https://github.com/buxtronix/arduino/tree/master/libraries/Rotary
 *          Since this code was in the form of an arduino library, and I had unresolvable
 *          references that I could not figure out, I just modified it as a single sketch.
 *          The only library support required is if you want to use interrupt polling,
 *          and then the MsTimer2 library needs to be installed.
 */
 // Note: removed optional #defined code mentioned above; using pin change interrupts.

#define DIR_NONE 0x00           // No complete step yet.
#define DIR_CW   0x10           // Clockwise step.
#define DIR_CCW  0x20           // Anti-clockwise step.

uint8_t state;
#define A 2                     //  encoder pin (CLK)
#define B 3                     //  encoder pin (DT)

// ------ My additions ------
unsigned int SW = 4;            // switch pin
         int count = 0;         // count each indent
         int old_count = 0;     // check for count changed
unsigned long int presstime = 0;// duration of button press (ms)
// --------------------------

/*
 * The below state table has, for each state (row), the new state
 * to set based on the next encoder output. From left to right in,
 * the table, the encoder outputs are 00, 01, 10, 11, and the value
 * in that position is the new state to set.
 */

// State definitions state table (emits a code at 00 only)
// states are: (NAB) N = 0: clockwise;  N = 1: counterclockwiswe
#define R_START     0x3
#define R_CW_BEGIN  0x1
#define R_CW_NEXT   0x0
#define R_CW_FINAL  0x2
#define R_CCW_BEGIN 0x6
#define R_CCW_NEXT  0x4
#define R_CCW_FINAL 0x5

const unsigned char ttable[8*4] = {
    R_CW_NEXT,  R_CW_BEGIN,  R_CW_FINAL,  R_START,                // R_CW_NEXT
    R_CW_NEXT,  R_CW_BEGIN,  R_CW_BEGIN,  R_START,                // R_CW_BEGIN
    R_CW_NEXT,  R_CW_FINAL,  R_CW_FINAL,  R_START | DIR_CW,       // R_CW_FINAL
    R_START,    R_CW_BEGIN,  R_CCW_BEGIN, R_START,                // R_START
    R_CCW_NEXT, R_CCW_FINAL, R_CCW_BEGIN, R_START,                // R_CCW_NEXT
    R_CCW_NEXT, R_CCW_FINAL, R_CCW_FINAL, R_START | DIR_CCW,      // R_CCW_FINAL
    R_CCW_NEXT, R_CCW_BEGIN, R_CCW_BEGIN, R_START,                // R_CCW_BEGIN
    R_START,    R_START,     R_START,     R_START                 // ILLEGAL
};

uint8_t *port_A;
uint8_t maskA;
uint8_t *port_B;
uint8_t maskB;

void setup( ) {
    pinMode( A, INPUT );
    pinMode( B, INPUT );
    pinMode( SW, INPUT );                   // My add - button switch

    portA = portOutputRegister(digitalPinToPort(A));
    maskA = digitalPinToBitMask(A);
    portB = portOutputRegister(digitalPinToPort(B));
    maskB = digitalPinToBitMask(B);

    attachInterrupt( 0, AB_isr, CHANGE );   // pin-change interrupts:
    attachInterrupt( 1, AB_isr, CHANGE );

    state = (digitalRead( A ) << 1) | digitalRead( B );     // Initialise state.
    old_count = 0;

    Serial.begin( 115200 );
    Serial.println( "\n___Rotary Encoder Tests___" );
}

void AB_isr( ) {
    // Grab state of input pins.
    unsigned char pinstate = *portA & maskA ? 2:0;
    pinstate |= *portB & maskB ? 1:0;

    state &= 0x07;
    state <<= 2;
    state |= pinstate;

    // Determine new state from the pins and state table.
    state = ttable[state];

    if( state & DIR_CW )    count++;
    if( state & DIR_CCW )   count--;
}

void loop( ) {
    if( old_count != count ) {
        Serial.println( count );
        old_count = count;
    }

    // --- My test of encoder button switch ---
    else if (digitalRead(SW) == LOW) {		                    // Button was pressed
      presstime = millis();
      while (digitalRead(SW) == LOW)	                // Wait for button release
      {}
      if (millis() - presstime > 500) {  	      // Long-press event
        count = -count;
        Serial.print( count ); Serial.println("  Reverse");
      } else {				                                                                          // Short press
        count = 0;
        Serial.print( count ); Serial.println("  Stop");
      }
      old_count = count;
    }
}
```

gmoneti, 2021-02-02:
Thank you. The code did not compile at first. I see that *port_A (B) is later referred to as *portA and portA.  I added an underscore (port_A) where missing. It compiled, but encoder knob generated no output. Button works.

gin66, 2021-02-02:
Thanks for correcting the typo. 

That the code does not work, could be related to this code:
```
    attachInterrupt( 0, AB_isr, CHANGE );   // pin-change interrupts:
    attachInterrupt( 1, AB_isr, CHANGE );
```
Reading the [spec](https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/), then it should be
```
    attachInterrupt( A, AB_isr, CHANGE );   // pin-change interrupts:
    attachInterrupt( B, AB_isr, CHANGE );

```

gin66, 2021-02-02:
but looks like digitalPinToInterrupt(A) and (B) translates into 0 or 1

gin66, 2021-02-02:
Guess the problem is here:
```
    portA = portOutputRegister(digitalPinToPort(A));
    maskA = digitalPinToBitMask(A);
    portB = portOutputRegister(digitalPinToPort(B));
    maskB = digitalPinToBitMask(B);
```

Should be Input and not Output:
```
    portA = portInputRegister(digitalPinToPort(A));
    maskA = digitalPinToBitMask(A);
    portB = portInputRegister(digitalPinToPort(B));
    maskB = digitalPinToBitMask(B);
```

gmoneti, 2021-02-02:
Ha! port_A = portInputRegister(digitalPinToPort(A));   That's the ticket! I'll run some tests.

gmoneti, 2021-02-03:
The code works fine alone with above corrections.

When I insert it into the motor code, it compiles but with a warning:
```
warning: invalid conversion from 'const unsigned char*' to 'uint8_t {aka unsigned char}' [-fpermissive]
     state = (ttable[state]);
```
After checking and rechecking, I cannot find a difference in the relevant code or another cause. I tried to recast the assignment of 'state', but so far found nothing that works. One potential clue is that when running the code (it compiles), the encoder counts only forward regardless of direction of turn. Also, to the point, there still were glitches at the usual frequencies.

Separately, I had already tried a fast digitalRead() replacement ([fast_digital_read_write2.txt](https://github.com/gin66/FastAccelStepper/files/5915618/fast_digital_read_write2.txt)). It tested at ~13X faster than digitalRead(). However, there still were glitches at the usual frequencies.

gin66, 2021-02-03:
Yes, the conversion warning comes from here:
`const unsigned char ttable[8*4] = {`
should be
`const uint8_t ttable[8*4] = {`

I prefer to use uint8_t, uint16_t and uint32_t over unsigned char, short, int, long. While switching machines, with the latter you always do not know, what you get from the compiler.

From my simulate long ISR tests, I already have doubts, that this is the root cause.

Currently I wonder, if the speed changes of a ramp or change of speed can be detected from the MSP as overload. If you for example start a ramp with `M1 V50 a1`, then the tiny speed steps can be heard. And those are due to limited amount of bits: 8bit mantissa put into a sqrt could end up with few significant bits leading two adjacent speed steps.

gin66, 2021-02-03:
if the encoder runs only one direction, try the second port too:
 `portB = portInputRegister(digitalPinToPort(B));`

gmoneti, 2021-02-03:
I had tried the different assignment above with no luck.
What is this?: 
`const uint8_t ttable[8*4] = {`
compared to this:
`const uint8_t ttable[8][4] = {`
Neither eliminates the warning when merging with motor code; and I get only one direction and skipped encoder counts.  I set this aside temporarily and tried something else...

gmoneti, 2021-02-03:
I

> if the encoder runs only in one direction, try the second port too:
> `portB = portInputRegister(digitalPinToPort(B));`

I didn't understand this. Would you elaborate?

gmoneti, 2021-02-03:
So, I merged the digitalRead() speedup code above (fast_digital_read_write2.txt) into the motor code. And it works. I stated that it still had glitches. But upon further testing, I noticed a few things:
1. glitches are I think gone.  (motor resolution @1000/rev, up to 1000rpm (60us/step)
2. With motor resolution @2000/rev, glitches returned. ( hit limit of 41us/step, 732rpm)
3. Get some extra glitches if I turn the knob back/forth really fast (more processing?).

Seems like all three to point a delay in processing that causes the glitch.

gin66, 2021-02-03:
In the optimized code the `const uint8_t ttable[8*4]` has been turned from a two dimensional array into a one dimensional.

This table has, as before 32 entries, just flat. That's why this calculation is needed to get the index in the one dimensional array:
```
    state &= 0x07;
    state <<= 2;
    state |= pinstate;
```

With this code I am sure, what the compiler will generate as code. Afterwards this calculated state is used as index into ttable.
```
    // Determine new state from the pins and state table.
    state = ttable[state];
```
 If the ttable is still the old `ttable[8][4]`, then you get the compiler error as above.

For the `portB = portInputRegister(digitalPinToPort(B));`: You said port_A = portInputRegister(digitalPinToPort(A)` has worked, but encoder only in one direction. So I thought, the change for portB from portOutputRegister to portInputRegister has missed. But now I assume, that actually the ttable was still being two dimensional and pointer values have been read...

Anyway, I doubt in the meantime, that the ISR is the root cause. And it is related to the speed steps in the ramp. Now I am thinking about a good border case and short sketch for test with the MSP.

gmoneti, 2021-02-03:
I had already made the changes to portA and portB originally. So, I thought you meant something else.

But Ah! Thank you for the explanation of  `ttable[8*4]`. Now, I understand better the changes you made and realized the brackets from the old multi-array were still there.  After removing them, everything compiles and works!

Test results: same as above: @1000counts/rev, smooth up to 1000rpm (60us/step), with possible glitch near top speed; @2000counts/rev, glitches at the usual speeds.

> Anyway, I doubt in the meantime, that the ISR is the root cause. It is more likely related to the speed steps in the ramp. Now I am thinking about a good border case and short sketch for test with the MSP.

Do you mean my 20rpm speed jumps? Or your ramp code? I'll keep poking around.

gmoneti, 2021-02-03:
Does the apparent 41us/step limit I saw have to do with the limits of the 16Mhz Uno?

gin66, 2021-02-03:
The max is 40us/step. As per README, the step rate is limited to 25 kSteps/s.

If you set a lower value, then setSpeedInUs() will return -1 and reject the provided value.

For sure it is not your 20rpm steps. Expectation is, that the application throws whatever value to the ramp generator and the ramp generator should make the best out of it. Now I am wondering, if the observed glitches, in my understanding detected by the MSP, are somehow by design in the ramp generator due to the limited precision in the calculation. Something a stepper may not react to, but the MSP does.

gmoneti, 2021-02-04:
I see. To be clear, MSP doesn't have to be connected while motor is running. It is used to configure some motor parameters, or to monitor it while running. 

ClearPath has been frustrating as a first introduction to motor control programming. On the one hand, it offers Step/Dir  or Quad A/B input (and their controller products, of course). On the other hand, certain parameters and "input modes" are set in MSP. It seems a partly "black box" and partly open box. They don't publish details of the interface that may be obscure to the programmer. It leaves one uncertain of what one is dealing with. They would probably attract a lot more business if the published more code and documentation.

Be that as it may, they have published some code that may be of interest to you:
https://github.com/Teknic-ClearPath
https://github.com/Teknic-ClearPath/ClearPath-Arduino-Step-And-Direction/tree/master/Arduino%20Uno/ClearPathStepGen (what I started with; but no continuous run or change command on the fly).
https://www.youtube.com/c/TeknicInc/videos 

As to what I can do, I'll put my motor at your disposal to run any sketches you wish to test.

gin66, 2021-02-05:
Could you please try the latest StepperDemo from github. I have added a new test sequence 10, which performs speed changes. This can be invoked by `t M1 10 I R`.

The sequence will:
1. ramp up to 64us.
2. vary between 64us and another lower/higher speed for 10 times and then check the next speed. The accerelation is set so high, that there is no ramp between two speeds. The current speed change is print to Serial
3. ramp down to stand still

Output to the terminal will be:
```
Run tests
Speed changes 64us <=> 65us
Speed changes 64us <=> 63us
Speed changes 64us <=> 66us
Speed changes 64us <=> 62us
Speed changes 64us <=> 67us
Speed changes 64us <=> 61us
Speed changes 64us <=> 68us
Speed changes 64us <=> 60us
Speed changes 64us <=> 69us
Speed changes 64us <=> 59us
Speed changes 64us <=> 70us
Speed changes 64us <=> 58us
Speed changes 64us <=> 71us
Speed changes 64us <=> 57us
Speed changes 64us <=> 72us
Speed changes 64us <=> 56us
Speed changes 64us <=> 73us
Speed changes 64us <=> 55us
Speed changes 64us <=> 74us
Speed changes 64us <=> 54us
Speed changes 64us <=> 75us
Speed changes 64us <=> 53us
Speed changes 64us <=> 76us
Speed changes 64us <=> 52us
Speed changes 64us <=> 77us
Speed changes 64us <=> 51us
Speed changes 64us <=> 78us
Speed changes 64us <=> 50us
finished
```

This test should reveal, how sensitive the torque measurement on slight speed jumps is.

gmoneti, 2021-02-06:
I ran the new test. Test ran flawlessly. Motor steps/rev was set at 2000; speed reached ~750rpm. 

Commands: t  M1  10  I  R

Enter test mode
Enter commands separated by space, carriage return or newline:
     M1/M2/..  ... to select stepper
     R         ... start all selected tests
     I         ... Toggle motor info, while test sequence is running
     01        ... select test sequence 01 for selected stepper
     :
     10        ... select test sequence 10 for selected stepper

   M1: @0 Acceleration [steps/s^2]=0 Speed [us/step]=0 
   M2: @0 Acceleration [steps/s^2]=0 Speed [us/step]=0 
Select stepper 1
Select test_seq: 10
Toggle motor info
Run tests
Speed changes 64us <=> 65us
Speed changes 64us <=> 63us
Speed changes 64us <=> 66us
Speed changes 64us <=> 62us
Speed changes 64us <=> 67us
Speed changes 64us <=> 61us
Speed changes 64us <=> 68us
Speed changes 64us <=> 60us
Speed changes 64us <=> 69us
Speed changes 64us <=> 59us
Speed changes 64us <=> 70us
Speed changes 64us <=> 58us
Speed changes 64us <=> 71us
Speed changes 64us <=> 57us
Speed changes 64us <=> 72us
Speed changes 64us <=> 56us
Speed changes 64us <=> 73us
Speed changes 64us <=> 55us
Speed changes 64us <=> 74us
Speed changes 64us <=> 54us
Speed changes 64us <=> 75us
Speed changes 64us <=> 53us
Speed changes 64us <=> 76us
Speed changes 64us <=> 52us
Speed changes 64us <=> 77us
Speed changes 64us <=> 51us
Speed changes 64us <=> 78us
Speed changes 64us <=> 50us
Speed changes 64us <=> 79us
Speed changes 64us <=> 49us
Speed changes 64us <=> 80us
Speed changes 64us <=> 48us
Speed changes 64us <=> 81us
Speed changes 64us <=> 47us
Speed changes 64us <=> 82us
Speed changes 64us <=> 46us
Speed changes 64us <=> 83us
Speed changes 64us <=> 45us
Speed changes 64us <=> 84us
Speed changes 64us <=> 44us
Speed changes 64us <=> 85us
Speed changes 64us <=> 43us
Speed changes 64us <=> 86us
Speed changes 64us <=> 42us
Speed changes 64us <=> 87us
Speed changes 64us <=> 41us
Speed changes 64us <=> 88us
Speed changes 64us <=> 40us
finished
>> M1: @3958679 Acceleration [steps/s^2]=1000 Speed [us/step]=64 
   M2: @0 Acceleration [steps/s^2]=0 Speed [us/step]=0

gin66, 2021-02-06:
Thanks for testing. No glitches observed by the MSP ?

gmoneti, 2021-02-06:
No glitches in MSP. 

My first run @1000steps/rev I stopped because the speed was going over the ~1000rpm motor current configuration limit (unchanged).  But, in the past, at 2000 steps/rev and more the glitches increased compared to 1000 steps/rev. So this was a good test.

gin66, 2021-02-06:
If this was a good test, then I am running out of ideas, what the root cause of the glitches could be.

gmoneti, 2021-02-06:
The glitches (really a torque spike from rapid drop and return of speed, not MSP exceptions) are most common at  41us/step, but more generally perhaps at points where the frequency changes?? (wild guess: skips a beat?). Unfortunately the glitches, despite occurring in usual places, are not consistent, only occasional. Which is a bit confounding. I sent a message to ClearPath (Teknic) support. There seems a slight chance of getting some possibly useful information. Won't know until Monday.

gin66, 2021-02-06:
Putting all puzzle pieces together, I think the issue is, that the encoder lines of KY-040 tend to bounce. This seems to be confirmed [here](https://www.best-microcontroller-projects.com/rotary-encoder.html):
_If you use the CLK signal as an interrupt you will get into deep trouble as the random bouncing of the input will trigger interrupts all the time (and not at the time when you want to read the data signal) so you will get incorrect data._

If the knob is turned, the lines are bouncing and every pin toggle generates an interrupt. If the bouncing is fast (more or less the definition of bouncing :-) ), then immediately after the first interrupt the next needs to be serviced and so on. Would be no problem, but the timer interrupt used by the library has a lower priority. Consequently, the step interrupt must wait and with sufficient long bouncing, the interrupt can be blocked for more than 40us causing the next scheduled step to be delayed by ~4ms. This is same to `e` in StepperDemo.

Solutions:
- Add capacitors to the input line
- Use a technique as described in the linked article (even has a sketch to judge the quality)
- Use a method of interlocked interrupts: 
  1. Start with Pin A interrupt enabled and Pin B interrupt disabled in setup()
  2. When Pin A interrupt occurs, disable Pin A interrupt and enable Pin B interrupt
  3. When Pin B interrupt occurs, disable Pin B interrupt and enable Pin A interrupt (back to step 2).
- Poll the encoder and not use interrupts for decoding
- Do not use KY-040

gmoneti, 2021-02-06:
I will test the different encoder coding solutions you suggested and report back. 

However, that page suggests using a state machine code design as a more sophisticated solution for the more troublesome encoders. I think I found the code I'm using via that page, indirectly.

I can report that I cannot generate any apparent bounce in turns or button presses no matter how fast I move.

On another page, I read that many people implement INPUT_PULLUP or external resistors on the KY-040, and that this is an error because the unit has them built-in (some may not, but mine do). In fact, I don't use  INPUT_PULLUP. When I tried using it, the output was "bouncy".

One thing I don't understand fully is your mention of possible problematic interaction between the stepper code and my interrupt driven encoder code. Is this a possible problem only in cases of bouncy encoders? Or independently of that?

Could the problem be that the encoder is inherently bouncy (causing interrupt delays) but the encoder code simply hides this well as it succeeds in generating the correct program output?

gin66, 2021-02-06:
If the timer interrupts are blocked for too long, then the issue will pop up.

The ttable/state implementation is a state machine and that state machine is quite robust against bouncing.  

If you only have the encoder and the stepper, then just poll the KY-040 in main loop and not use the interrupt. This is quickly done: Comment out the two attachInterrupt() calls and put a AB_isr() at the beginning of the loop(). This way the encoder is used in polling mode.

gin66, 2021-02-06:
The INPUT_PULLUP does not prevent (perhaps reduce) bouncing. 

If the encoder would not be bouncy, then the interrupts would not block the processor for too long. So the interrupt code would be fine, if the encoder would not cause multiple interrupts in sequence.

gmoneti, 2021-02-07:
I commented the `attachInterrupt()` calls and put a `AB_isr()` at the beginning of the loop(). The glitches were indeed eliminated at 1000 and 2000 counts/rev. 

Interestingly, at 4000 and 8000 counts/rev the ramping stopped working:hard reversals; hard stops; hard velocity changes. But also, at 8000, after some commands, speed stops responding to encoder turn or monitor commands.---The program seems to lose its way.

Tomorrow (Sunday) I'll try some code that should show how bouncy the encoder is, mechanically.

gin66, 2021-02-07:
Sounds good, that the glitches are happening anymore. If this holds true during the next days, then better to close this thread, which has become already quite long.

For the new observation, I have opened a new issue

gmoneti, 2021-02-07:
Yes, of course. I learned a lot. Thank you. 

To finish up, I tested the bounciness of the ky-040 using the code from the page you referenced above [here](https://www.best-microcontroller-projects.com/rotary-encoder.html). Interestingly, the results were very close to the "good" example.

Clearly there is some problem interaction with the encoder. But I think the question still remains: Is the encoder not good enough? Or, is using encoder via interrrupts generally incompatible (or imperfectly compatible) with the stepper code, at least on arduino Uno?

It also remains curious that the glitches (using interrupts) occur at predictable us/step frequencies.

gin66, 2021-02-07:
Hmmm, after rereading the code from the linked page, I am not sure, if it reliably detects the bouncing on one signal line alone, which causes on each transition an interrupt. Better detection could be using just a plain interrupt counter, which increments by one for each AB_ISR() invocation. If the knob is rotating continuously in one direction, then per encoder count there should be (I assume) four interrupts to be counted. If the interrupt counter counts much morem then this would be a better and direct proof for bouncing on the encoder signal lines.

The encoder seems to be mechanical. For a mechanical switch, bouncing can happen. So the switch is for itself not bad. Similarly, the interrupt solution is reliable working and there is no issue, if a signal line transitions several times between high and low. The encoder interrupt will work  reliable and it doesnot matter to the encoder, that the ISR is blocking the CPU for a longer time. It just does not interact well with the FastAccelStepper library, because this lib is designed to have interrupts happening with 25kHz/s and expects from the application to not block interrupts for >40us. Putting all three components together just does not work well, because the assumption from FastAccelStepper is not met by the encoder (or the application from FastAccelStepper's point if view).

Why the glitches (using interrupts) occur at predictable us/step frequencies is definitely a mystery. And speculation could be, that the step pin from the stepper via EMC/parasitic capacitances/... support to induce  glitches and if the right frequency is met, by chance in sync with encoder rotation, then the long ISR blockage have chance to happen more likely at specific stepper frequency. For sure a super long shot and probably wrong.

If the above mentioned interrupt counter only counts faster than the encoder counter at special frequencies of the stepper, then this could hint that my long shot eventually has some truth. 

Anyway, as the glitches have vanished by not using the ISR for the encoder, then I prefer to say, that the FastAccelStepper library appears to not cause the glitches and I cannot solve this by changing the lib code.

gmoneti, 2021-02-07:
Thank you for the analysis.  I added an interrupt count in the `AB_isr()` as you suggested. On each encoder change it prints the ISR count and resets the counter to 0. That's all. There is no other activity than the encoder and stepper code. 

I was able to generate a few glitches. They somewhat correlated with ISRs of 20+. But there were some ISRs=~50 that generated no glitch, and some ISRs=~15 that did.

A snippet of ouput:
```
cnt: 941    rpmR: 937    speedInUs: 64    ISRs: 6
cnt: 942    rpmR: 937    speedInUs: 64    ISRs: 7
cnt: 943    rpmR: 937    speedInUs: 64    ISRs: 7
cnt: 944    rpmR: 937    speedInUs: 64    ISRs: 21
cnt: 945    rpmR: 952    speedInUs: 63    ISRs: 6
cnt: 946    rpmR: 952    speedInUs: 63    ISRs: 8
cnt: 947    rpmR: 952    speedInUs: 63    ISRs: 9
cnt: 948    rpmR: 952    speedInUs: 63    ISRs: 11
cnt: 949    rpmR: 952    speedInUs: 63    ISRs: 5
cnt: 950    rpmR: 952    speedInUs: 63    ISRs: 13
cnt: 951    rpmR: 952    speedInUs: 63    ISRs: 12
cnt: 952    rpmR: 952    speedInUs: 63    ISRs: 6
cnt: 953    rpmR: 952    speedInUs: 63    ISRs: 4
cnt: 954    rpmR: 952    speedInUs: 63    ISRs: 5
cnt: 955    rpmR: 952    speedInUs: 63    ISRs: 8
cnt: 956    rpmR: 952    speedInUs: 63    ISRs: 8
cnt: 957    rpmR: 952    speedInUs: 63    ISRs: 14
cnt: 958    rpmR: 952    speedInUs: 63    ISRs: 15
cnt: 959    rpmR: 952    speedInUs: 63    ISRs: 5
cnt: 960    rpmR: 952    speedInUs: 63    ISRs: 6
cnt: 961    rpmR: 968    speedInUs: 62    ISRs: 7
cnt: 962    rpmR: 968    speedInUs: 62    ISRs: 9
cnt: 963    rpmR: 968    speedInUs: 62    ISRs: 15
cnt: 964    rpmR: 968    speedInUs: 62    ISRs: 9
cnt: 965    rpmR: 968    speedInUs: 62    ISRs: 10
cnt: 966    rpmR: 968    speedInUs: 62    ISRs: 4
cnt: 967    rpmR: 968    speedInUs: 62    ISRs: 6
cnt: 968    rpmR: 968    speedInUs: 62    ISRs: 8
cnt: 969    rpmR: 968    speedInUs: 62    ISRs: 15
cnt: 970    rpmR: 968    speedInUs: 62    ISRs: 12
cnt: 971    rpmR: 968    speedInUs: 62    ISRs: 23
cnt: 972    rpmR: 968    speedInUs: 62    ISRs: 6
cnt: 973    rpmR: 968    speedInUs: 62    ISRs: 8
cnt: 974    rpmR: 968    speedInUs: 62    ISRs: 8
cnt: 975    rpmR: 968    speedInUs: 62    ISRs: 8
cnt: 976    rpmR: 984    speedInUs: 61    ISRs: 5
cnt: 977    rpmR: 984    speedInUs: 61    ISRs: 8
cnt: 978    rpmR: 984    speedInUs: 61    ISRs: 28  (glitch here, I think)
cnt: 979    rpmR: 984    speedInUs: 61    ISRs: 8
cnt: 980    rpmR: 984    speedInUs: 61    ISRs: 10
cnt: 981    rpmR: 984    speedInUs: 61    ISRs: 14
cnt: 982    rpmR: 984    speedInUs: 61    ISRs: 8
cnt: 983    rpmR: 984    speedInUs: 61    ISRs: 9
cnt: 984    rpmR: 984    speedInUs: 61    ISRs: 4
cnt: 985    rpmR: 984    speedInUs: 61    ISRs: 11
cnt: 986    rpmR: 984    speedInUs: 61    ISRs: 6
cnt: 987    rpmR: 984    speedInUs: 61    ISRs: 8
cnt: 988    rpmR: 984    speedInUs: 61    ISRs: 11
cnt: 989    rpmR: 984    speedInUs: 61    ISRs: 10
cnt: 990    rpmR: 984    speedInUs: 61    ISRs: 7
cnt: 991    rpmR: 984    speedInUs: 61    ISRs: 7
cnt: 992    rpmR: 1000    speedInUs: 60    ISRs: 9
cnt: 993    rpmR: 1000    speedInUs: 60    ISRs: 6
cnt: 994    rpmR: 1000    speedInUs: 60    ISRs: 8
cnt: 995    rpmR: 1000    speedInUs: 60    ISRs: 8
cnt: 996    rpmR: 1000    speedInUs: 60    ISRs: 7
cnt: 997    rpmR: 1000    speedInUs: 60    ISRs: 12
cnt: 998    rpmR: 1000    speedInUs: 60    ISRs: 70  (none here)
cnt: 999    rpmR: 1000    speedInUs: 60    ISRs: 7
cnt: 1000    rpmR: 1000    speedInUs: 60    ISRs: 12
cnt: 1000    rpmR: 1000    speedInUs: 60    ISRs: 4
cnt: 1000    rpmR: 1000    speedInUs: 60    ISRs: 6
```

gin66, 2021-02-07:
Thanks for the experiment. Seems my guess with 4ISRs per 1count was not too bad, as I can see four times with only 4. 

On the opposite side, 28 or even 70 ISRs per count is strong indication for the bouncing of the encoder signal lines. That 70 ISRs does not create a glitch depends on the timing of the ISRs. If there is an e.g. µs-pause between one ISR end and the next ISR, the timer ISR can squeeze in :-)

gmoneti, 2021-02-07:
You are no doubt consistently on the money. I'm not surprised. I am the student, and I'm learning a lot from you. I am new to hardware programming, and I wonder about these, lets call them, "programming collisions". Are they commonly potentially lingering in the hardware programming world, whereby minimizing their occurrence---but not guaranteeing their elimination--- is the realistic strategy?

gin66, 2021-02-07:
As software developer for embedded devices you can expect, that hardware may give you the one or other surprise. So adding a capacitor to the input lines is for sure a good strategy and helps to avoid those spikes. On the other hand, if you have identified the root cause, then you still can be decide to eliminate it or live with it. And living with it is a well accepted strategy. Just along the lines: never change a running system. Perhaps the quick fix to the root cause, gives you later a even bigger headache. For example the capacitors, you have added to mitigate the bouncing, are working totally good, until you notice days later, that very fast rotation of the knob are not detected anymore. And typically this happens in exactly that moment, when you want to demonstrate your wonderful device to someone important to you.

If you have to use ISR, then you should always expect, that the wonderful ISR routine could totally block your µC. That's why, it is a good idea to use a port interrupt with a rate counter, to shut off the interrupt, if the rate is way too high due to noise coming in. For example your mobile phone next to your cool device, while sending an SMS or receiving a call, can generate spooky effects on unprotected inputs...

Anyway, the glitches topic appears to be closed and so can be this issue.

closed 2021-02-07 by gin66