gin66/FastAccelStepper #48: Question on Demo/Test Program to Use to Verify FastAccelStepper Running on ESP32 to Operate A STAC6 Stepper Driver

vepicurean, 2021-02-07:
Hi:

Just FYI background. We've been using the ESP32 MCU with the Easy Driver stepper driver till now. We have had some good results, but not perfect. The Easy Driver stepper driver modelues were just for the lab setup to develop our code, and now we're trying the FastAccelStepper out on a true industrial stepper driver (the STAC6-QE stepper driver) with a NEMA 34 1841 oz-in torque step motor. 

A application is straight forward (I'll leave the encoder our of things on this question for simplicity, just interested in running the STAC6 stepper 6 correctly and smoothly for this industrial large NEMA 34 step motor). 

Just need FastAccelStepper to run on an inductrialized ESP32 MCU (Using Industrial Shields hardware as a ESP32 based PLC to control a stepper driver and step motor). 

What is a good demo to test sketch to verify FastAccelStepper functionality and the set-up with a STAC6-QE stepper driver using FastAccelStepper running on an ESP32 using the Arduino IDE for the PLC code?  Sorry, I don't see a "question" designation on here. 

https://github.com/gin66/FastAccelStepper/tree/master/examples

Thanks for your help and time.

gin66, 2021-02-07:
The swiss army knife of the examples is StepperDemo. I use it for everything in order to test the functionality. It can even simulate two types of application errors, which can disturb operation of the stepper. As StepperDemo is nearly identical on avr and esp32, I haven’t come up with a sophisticated Wifi/Web/websocket solution to drive the stepper as an example. So just old-fashioned serial interface at 115200 Baud.

For building an application and integration: StepperDemo code could be partially reused. For example the code for table of steppers and generating the steppers would be worth, but with only one motor to drive definitely overkill. Which leaves the UsageDemo with only a handful of commands, which can be integrated in the application quite easily,

PrintPosition is primarily an example to print the position, while running.

If you plan to use the acceleration features, then Pulses and RawAccess* examples are not of help.

vepicurean, 2021-02-07:
Thanks. 

We'll get moving on setting up the STAC6 stepper driver/NEMA 34 step motor to run with the ESP32 MCU as a PLC using the StepperDemo as a test sketch (likely with some edits). 

We only need to test just going in the clockwise direction at very low speeds from 1RPM to about 4RPM, with the ability to vary the speed without stopping the rotation of the motor). 

As we are going at low speeds, acceleration and deceleration are not functions we need (but it seems they are needed for certain things from a coding perspective with FastAccelStepper). We want the target speeds (e.g. 1RPM, 2RPM, etc.) to be attained instantaneously, and we need to have the precise step position (step position/ID of the motor or exact position in degrees as of the motor shaft rotates).

We'll give you some comments and feedback here as we go along the next few days. 

Thanks for the information.

gin66, 2021-02-07:
NEMA34 has 200steps/revolution. This means:
- no microstepping => ~3.3 steps/s
- 256 microsteps (max of STAC6 from a quick internet search) => ~853 steps/s

Using 853 steps/s as max.frequency, then the minimum time between two steps is ~0.0012s. If the max frequency shall be reached after the first step, then the acceleration should be larger than 853steps/s/0.0012s = 710833 steps/s². So an acceleration value of 1,000,000 or 10,000,000 should jump directly to the requested speed without ramp up. And those acceleration values are ok for less microsteps, too.

gin66, 2021-02-07:
> ...and we need to have the precise step position (step position/ID of the motor or exact position in degrees as of the motor shaft rotates).

The FastAccelStepper is using a command queue, from which the low level stepper driver gets its commands from: how many steps, with how many ticks in between and which direction. The low level stepper driver does not maintain a separate counter for efficiency reasons. If the application needs to have the precise step position, then the driver has to calculate back from the end of command queue position (this position is maintained) all commands till the current command. And for esp32 the currently executed command is not included in this position as mentioned [here](https://github.com/gin66/FastAccelStepper#issues)

On the other hand, the esp32 has free pulse counters at your service and those can count every step (even with direction) and the current pulse count at the moment of readout is retrieved. There is only one drawback: This counter is only 16 bit and in the FastAccelStepper configured to count only up to 16383 (so 14bit). An example for the usage is [here](https://github.com/gin66/FastAccelStepper/blob/master/examples/StepperDemo/test_seq_09.cpp). It is included in StepperDemo. 

For implementation this needs to be done:
1. Attach a free pulse counter to the stepper in setup() and after stepper has been defined:
```
      if (!stepper->attachToPulseCounter(7)) {
        Serial.println("Error attaching to pulse counter");
      }
```
2. When the position is required:
```
int16_t current_position = stepper->readPulseCounter();
```

If you need the absolute 32bit position, then you can combine this value with `getPositionAfterCommandsCompleted()`. This combination is not trivial, but not too complicate either. If the application never sets the position or uses ` forceStopAndNewPosition()`, then those two counters stay in sync. Just while running, the `readPulseCounter()` value is momentarily and `getPositionAfterCommandsCompleted()` is ahead by a couple of ms. So something like this should work:
```
uint16_t current_position = (uint16_t) stepper->readPulseCounter();
uint32_t future_pos = (uint32_t)stepper->getPositionAfterCommandsCompleted();

int32_t combined_pos;
if ((future_pos & 0x3000) == (current_position & 0x3000) {
   combined_pos = (int32_t)((future_pos & 0xffffc000) | current_position);
}
else {
   // Try again later or deal with the wrap around
}
```

In StepperDemo a free pulse counter can be attached by `p7` (pulse counter 7) and from the moment on, the stepper position is followed by a number in brackets (14bit wrapping around).

gin66, 2021-02-07:
...actually, if the attachToPulseCounter() would take a second parameter to define the wrap around value (max. 32767), then this could be used to encode the current angle of the motor shaft. Just not for 256 microsteps and 200steps/revolution, as the wrap around value would be too high. But 128 microsteps should work with wrap around at 25600.

gin66, 2021-02-07:
If pulse counters 7 is used, this should adjust the wrap around value after attaching the pulse counter:

`PCNT.conf_unit[7].conf2.cnt_h_lim = 200*128;`

vepicurean, 2021-02-07:
The STAC6 motor we've configured for the default 1/128 microstepping or about 25000 microsteps (actually 25,600 microsteps/revolution).  That's plenty of steps for smooth and precise operation of the index table application. 

At the default setup (no PLC attached, just the stepper driver running the NEMA 34 motor), just the test set-up within the stepper driver, the movement of the NEMA 34 1841 oz-in motor is very quiet and extremely smooth in both directions. 

From the numbers in your response, I can see you're basing them on 1 revolution per minute for the index table movement cycle application.  1 revolution per minute (speed on the index table, we go continuously clockwise) or 25,600steps/60sec = 426.666667 steps/sec. 

We need to be able to configure the operation as the motor is running to go from a low speed of 1RPM, to 1.5RPM, to 2RPM, etc. up to about 4PM maximum. 

At maximum 4 RPM, then that would be 4 x 25,600steps/60 sec = 1,706.666667 steps/sec

So the the 4RPM setting, we're have a minimum time between each step of 0.00058594 seconds. To reach the 4RPM speed instantaneously then we need the acceleration to be 1,706.666667 steps/sec/0.00058594s = 2,912,698.68416 steps/s². So your calculation/suggestion of setting the acceleration to 10,000,000 steps/s² will achieve are range of speeds from 1RPM through to 4RPM for our index table application. 

Thanks for your detailed explanation and suggestions on using the pulse counter. I have to go through it and implement it fully/properly to understand the step ID/positioning counting technique as you describe it. 

The key to all this is finding our where the Z index pulse is on the incremental encoder (which step number) that Z index pulse occurs related the physical position of the shaft which can be done through proximity sensor on the index table (what step position the Z index pulse occurs at vs what step ID the proximity sensor gives a pulse at).  Once we have that relationship and between the shaft position using a proximity sensor and the Z index pulse on the encoder in the physical would, then we can use specific steps to trigger events as the index table turns at the respective step positions (or even the encoder count, given the encoder has 8000 counts per revolution = 2000 pulses per revolution on the encoder A and B signals).

gin66, 2021-02-07:
Just have extended the API to allow configuration of the pulse counter.
For your case, you could call now `attachToPulseCounter(7, -25600, 25600`.
Reading the pulse counter will give the angle at that moment (for a negative value just add 25600).

In StepperDemo the command would be `p7,-25600,25600`

vepicurean, 2021-02-08:
Thanks. Please leave this question topic up and let us try to implement your pulse (aka step) counter instructions over the next day or two.  We'll give you some details feedback and I'm sure we'll have a couple questions.  

I realized we probably don't need to involve the encoder in this. Let the STAC6 stepper driver's built in encoder closed feedback loop functionality use the incremental encoder on the shaft of the motor track where the shaft is vs where it should be (the ESP32 step count and the STAC6 step count should be in sync). The STAC6 stepper driver gives an error alarm if the actual motor shaft is not where (what step) the STAC6 believes the shaft should be (alerted by the incremental encoder)..... No need to duplicate the wheel. 

Monitoring the proximity sensor pulse via an input on the ESP32 as the index table turns.....  if we know what step that pulse from the proximity sensor occurs (once per revolution), then we know exactly what step in the physical world to trigger the other events (sensors, triggers, etc.) at (or can determine it) in relative or even absolute steps as the shaft turns. 

I'm hoping it's that simple, and this functionality you describe using `attachToPulseCounter(7,-25600, 25600)` enables/allows us to number each step from either 0 to 25599 or probably easier from 1 to 25600 in the clockwise direction and in the counter clockwise direction the counter goes from 0 to -25599 or from -1 to -25600.

gin66, 2021-02-08:
The pulse counter does unfortunately not work like this. You can experiment with it in StepperDemo.

vepicurean, 2021-02-08:
Thanks for the information. Yes, I'm a bit worried that FastAccelStepper is not the ideal stepper drive library for controlling precise positioning on an index table in our specific application with a STAC6 stepper driver and the NEMA34 step motor. 

We're going to try a few different approaches for our index table application using FastAccelStepper on the ESP32 based PLC, and we'll let you know how it goes as we're working on it over the next couple of days.

gin66, 2021-02-08:
OK. If any issue pops up or support is needed, just reopen this issue or raise a new one. For now I assume, the issue can be closed

vepicurean, 2021-02-14:
Hi:

We have worked out several software issues the last few days. Lesson learned. Must only use Arduino IDE on Linux with ESP32 MCU. There is serious issues uploading sketches to ESP32 on Windows 10 and MacOS 11. Created a little Linux VM and the Arduino IDE works as it should loading the sketches to the ESP32 now. 

We're on the last bit of investigative research to see of the FastAccelStepper library can work with our Industrial Shields ESP32 and Arduino Nano based PLCs. 

Just can't get FastAccelStepper and this industrial version (outputs at from 12VDC to 24VDC on the output PINs to drive the stepper driver) of our ESP32 MCU from Industrial Shields to work together for some reason. Something I'm doing wrong on implementation of the "digitalWrite(X, HIGH)" and "digitalWrite(X, LOW)" on the output pi of the ESP32 to the stepper driver to generate the steps.... unfortunately, it looks like this digitalWrite syntax is required with Industrial Shields ESP32 based PLC implementation. 

This is the little simple test sketch I use to verify the ESP32 can drive our stepper driver (STAC6) and the step motor. I can set the enable pin with the digitalWrite and I can set the direction pin with the digitalWrite. The stepper driver and motor status confirm these two pins from the ESP32 on the Industrial Shields device are set properly. 

But can't get the motor to step properly, and usually I can't get it to step at all. The below code, the motor won't step at all, but it's enabled, and the direction is set. 

Any suggestions on how to implement the FastAccelStepper library with this industrialised ESP32 PLC?

`#include "FastAccelStepper.h"

// As in StepperDemo for Motor 1 on AVR
//#define dirPinStepper    5
//#define enablePinStepper 6
//#define stepPinStepper   9  // OC1A in case of AVR

// As in StepperDemo for Motor 1 on ESP32 - Industrial Shields 10 I/O device
#define dirPinStepper Q0_1  //   pin Q0_1 configured as output at 12VDC with 800 Ohm resistor
#define enablePinStepper Q0_2 //  pin Q0_2 configured as output at 12VDC
#define stepPinStepper Q0_0  // pin Q0_0  configured as output at 12VDC with 800 Ohm resistor

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {

   pinMode(dirPinStepper, OUTPUT);
   pinMode(enablePinStepper, OUTPUT);
   pinMode(stepPinStepper, OUTPUT);

   
   digitalWrite(enablePinStepper,HIGH);  // set enable pin on stepper driver, confirmed OK
   digitalWrite(dirPinStepper, HIGH); // set direction CW or CCW
 
  
  engine.init();
  stepper = engine.stepperConnectToPin(stepPinStepper);
  if (stepper) {
    digitalWrite(stepPinStepper, HIGH); // not working.... motor enabled, direction set, but won't step
    //stepper->setDirectionPin(dirPinStepper);
    //stepper->setEnablePin(enablePinStepper);
    //stepper->setAutoEnable(true);

    // If auto enable/disable need delays, just add (one or both):
    // stepper->setDelayToEnable(50);
    // stepper->setDelayToDisable(1000);

    stepper->setSpeedInHz(426.666667);  // set speed.... when using Hz is steps/sec !!!
    stepper->setAcceleration(10000000); //go straight to target step with 10,000,000 or above
    stepper->moveTo(25600);  //one revolution of motor with 25,600 microsteps to position 25,600
    //stepper->runForward();
  }
}

gin66, 2021-02-14:
Could it be, that part of the I/Os are connected to an I/O-extender ? From what I have found in the Internet, the Q.0.5, Q.0.6 and Q.0.7 are marked as Digital Out/PWM Out. This sounds more like those three pins are directly connected to the uC. So switch the stepper pin to one of those three as a try.

For debugging this issue:
1. Do not use stepperConnectToPin()
2. generate first pulses by digitalWrite() and a sleep in a loop
3. If this works, then the question is, if that pin is really connected to the esp32 pin

Do you have a schematics of this Industrial ESP32 ?

BTW: The direction pin must too be on a processor pin, because the interrupt does not use digitalWrite but toggles bit directly on an esp32 port.

vepicurean, 2021-02-14:
I've seen information about Q0_5, Q0_6, and Q0_7, but I'm not sure it's for this specific Industrial Shields 10I/O ESP32 based device.

Here's the info on the [Industrial Shields ESP32 10 I/O device](https://www.industrialshields.com/web/content?model=ir.attachment&field=datas&id=208561&) that should help.  I don't seen anything in the user guide about the Q0_5, Q0.6, and Q0_7 in this 10 I/O ESP32 user guide being different from the others. 

All the pins have a corresponding ESP32 GPIO pin. You have to set it as an input or output,  and additionally there is this digitalWrite() function requirement that is proving to be a challenge. 

[image]

I'll try your suggestions and let you know in a bit how it behaves. 

Thank for this.

gin66, 2021-02-14:
Links has helped. It contains a reference to the board description, which links to another sdk file.

From the file pins_is.h it seems, that all the Q and I pins (except IO_5 and IO_6) are not directly connected to an esp32 port pin. There is mentioned an expanded GPIO via PCA9685 (https://www.nxp.com/products/power-management/lighting-driver-and-controller-ics/ic-led-controllers/16-channel-12-bit-pwm-fm-plus-ic-bus-led-controller:PCA9685)

From the file pins_arduino.h you would have more luck with the T0-T9 ports.

vepicurean, 2021-02-14:
I'm not sure what you're saying with regards to this NXP controller device. Are you saying this Industrial Shields 10 I/O ESP32 device (using their implementation of the ESP32 devkitC board) can't work with this STAC6 stepper drivers? 

Where do Industrial Shields reference this NXP device/controller that you sent me the link to? Not sure how that is going to help with this 10 I/O ESP32 device. I'm not sure if you were suggesting to add that NXP controller device to the Industrial Shields ESP32 10 I/O PLC setup or not. 

I have another sketch that I use for basic testing that works just fine with our other ESP32 stand alone devkitC boards (ESP32-WROOM-32), but the adjusted sketch for the Industrial Shields ESP32 10 I/O device behaves completely differently when driving the STAC6 stepper driver.  They show their 10 I/O device has the same/similar ESP32 devkit3 board in it. 

Changing to pins Q0_5,Q0_6, and Q0_7 made no different in behaviour with the our other simple test sketch I use. On loading this other test sketch, the motor actually goes a bit crazy, and gets very loud and spins much faster than on the stand alone ESP32 dev boards.  Something is not right, as it does not even change direction when toggling the direction pin between HIGH and LOW on their device. 

I have a note into the people at Industrial Shields to get them to explain why the stepper driver and motor behave so different when using the ESP32 within their setup vs when it's stand alone. They mentioned before they have never used the AccelStepper, no the FastAccelStepper libraries before in their setup.

Will let you know what they say. Thanks for the help and time.

vepicurean, 2021-02-18:
An update on happenings. 

These use of the optical isolation technique on the output pins (they do this to protect the devices I guess) is problematic for running a stepper driver properly. The maker of the industrialized PLC based on the ESP32 (or any other dev board) admitted that to me. You were right on that. I didn't understand what you were saying about that fact before.  It's a problem that there are a few options to get around this limitation, but that's outside the scope of FastAccelStepper itself. So I won't discuss it here. What we'll probably do is make our own custom PLC board and I/O system based on using the ESP32 dev board (devkitC board). 

The stand-alone ESP32 seems to be working well with regards to speed (steps/second).... I've timed the performance of the stepper driver and the step motor externally using the FastAccelStepper on a stand-alone ESP32-WROOM-32 dev board. Whatever we set the code speed to using FastAccelStepper to in Hz, we measure that speed exactly. And we know we have exactly 25,600 microsteps in one revolution of the step motor shaft. I've measured that externally many times using the motor vendor software for the step motor.  

What we're struggling with now is using the counter 7 you mentioned in before. With the way we're using the counter 7, it's not in line with the number of actual steps, so we can't use that counter for our positioning needs.  As you mentioned, the count counts to a maximum of 16,383. Neither of us quite understand that wrap around counting approach you mentioned before. But right now, when we let the counter continue past 16,383, it resets back to 0, then counts up again, when we add the 16,383+ the value of the counter when the motor stopped (we simply time 60 seconds) we get about 9,619 on the counter. So 16,383+9619+1+1= 26,004   -> The counter is not even close to the 25,600 value that we should be getting when running the speed of the motor at 1RPM for 60 seconds. The motor shaft rotates the exact one revolution, so we know we should be reading 25,600 for the counter value (or 25,599 if the steps start at zero instead of 1).

So the question is can we get a single counter using FastAccelStepper than can count up to the 25,600 steps we need for one rotation of the motor shaft? We need this for positioning?

The second part of that question is, assuming we can get a counter that can go to 25,600 steps, how accurate will it be to mark/ID each of the 25,600 steps? 

If we get this solved, I believe we're on our way to making this FastAccelStepper library work for our positioning application needs based on using the industrial stepper driver and step motor setup with the programming done using an ESP32 MCU.

Thanks for all your help and time. Appreciated.

gin66, 2021-02-18:
The optical isolation is not the main problem. It is the LED-IC driver, they have introduced to drive the optical isolation LED. This disconnects the output from the uC pin and the timer module is not able to drive the output pin after the LED driver.

For the pulse counter: Please note that I have recommended this call `attachToPulseCounter(7,-25600, 25600)`. This way the counter will call up to 25599 and then wraps to 0. if the motor only runs into positive direction, the pulse counter will be always (total_steps mod 25600) and as such in the range 0..25599. So exactly linked to your motor angle, no matter how fast or how slow or how many revolutions the motor is currently on.

vepicurean, 2021-02-19:
Thanks for all the info. 

I can get the attachToPulseCounter(7, -25600, 25600) to count from 0 to 25,599, but counter readings are not correct.  

It's counting pulses way too fast for a 1RPM speed. 

We are not getting how you do the adjustments to get the pulse counter to sync 1 to 1 with the actual 25,600 microsteps at the correct speed for the motor shaft. I know you go over it above, we're just not grapping your instructions on how to do the 1:1 sync between the counter and the actual microsteps as the motor shaft turns.

gin66, 2021-02-19:
The driver uses one mcpwm module to generate pulses and one pulse counter to count the generated pulses. Both are attached to the step pin. With attachToPulseCounter a second pulse counter is attached to the same pin. If the second pulse counter is counting too fast, then something is wrong.

Could you please replicate with StepperDemo  and provide the log file of one session, which includes connecting second pulse counter and let the motor run for one or two revolutions ?

vepicurean, 2021-02-19:
Thank you. 

I will adapt our little test app within Stepper Demo (will comment out all of the rest of StepperDemo we don't need) to use one motor and capture the logs. Should send to you in a couple days. 

We close now to figuring this out. 

Thanks for all your help.

bkapadia01, 2021-02-20:
Hi,
I've added the log of file of one of the sessions which includes running the stepper for 1 revolution at a rate of 2 revolutions/min. 
Any suggestions regarding the way this has been implemented or improvements we should make? 
I've pasted the code that we're using below.
I've also attached the logs to this message.

Just a note that the read pulse counter and steps appear to be in sync which is expected but when we print the pulses we have the same pulse printed couple times initially. We have set the acceleration to (10000000) to prevent any timing issues with acceleration.

[FAST_pulsecounter.pdf](https://github.com/gin66/FastAccelStepper/files/6013997/FAST_pulsecounter.pdf)

Code:
```
#include <pfodESP32Utils.h>
#include <pfodESP32BufferedClient.h>

#include <loopTimer.h>
#include <pfodBufferedStream.h>
#include <FastAccelStepper.h>
//#include <Encoder.h>

// // Define a stepper and the pins we will use ESP32
#define dirPinStepper 18
#define enablePinStepper 13
#define stepPinStepper 15

char user_input;

//timer init
unsigned long startMillis;
unsigned long currentMillis;
boolean counting;
boolean stopInput = false;

#define BAUD_RATE 115200

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

int pos = 1000;
long oldPosition = -999;

// Change these two numbers to the pins connected to your encoder.
//Encoder myEnc(12, 14); // (B+, A+)

//Example of using bufferedStream to release bytes at BAUD_RATE baud, buffer size 130
const size_t bufSize = 130;
uint8_t buf[bufSize];
pfodBufferedStream bufferedStream(BAUD_RATE, buf, bufSize, false);

millisDelay stepDelay;

void setup()
{
  Serial.begin(115200);
  engine.init();
  stepper = engine.stepperConnectToPin(stepPinStepper);
  stepper->setDirectionPin(dirPinStepper);
  stepper->setEnablePin(enablePinStepper);
  stepper->setAutoEnable(true);
  stepper->disableOutputs();

  bufferedStream.connect(&Serial); // connect buffered stream to Serial
  stepper->setSpeedInHz(853.333333333334);  //    2 revs/minute // the parameter is us/step for esp32 1 rev= 
//    stepper->setSpeedInHz(426.666666666667);  // 1 rev/minute 
//  stepper->setSpeedInHz(10);        // the parameter is us/step for esp32

  stepper->setAcceleration(10000000);
  stepDelay.start(0);  // Stepper msdealay between each cycle

  if (!stepper->attachToPulseCounter(7, -25600, 25600)) {
        Serial.println("Error attaching to pulse counter");  // discuss this attachment of pulse counter. Don't understand. 
      }
}

// the loop function runs over and over again forever
void loop()
{
  startMillis = millis(); //save the start time
  stepper->disableOutputs();

if(Serial.available())
  {
    user_input = Serial.read();
    stepper->disableOutputs();
    int16_t current_position = stepper->readPulseCounter();  
    if (user_input == 'S')
    {
      runStepperMotor();
    }
    else if (user_input == 'P')
    {
      stopStepper();
    }
    else if (user_input == 'Z')
    {
      runStepperToZero();
    }
    else
    {
//      Serial.println("Invalid option entered.");
    }
  }
}

void runStepperMotor()
{
  stepper->enableOutputs();
      stepper->moveTo(-25599);
      while(!((stepper->readPulseCounter() - 1) == (-25600))){
      Serial.print(" -> PULSE COUNTER -1: ");
      Serial.print(stepper->readPulseCounter() - 1);

      currentMillis = millis();
      if (stepper->pulseCounterAttached()) {
        Serial.print(" -> Pulse Counter: [");
        Serial.print(stepper->readPulseCounter());
        Serial.println(']');
        Serial.print("TIME MILLIS: ");
        Serial.print(currentMillis);
        }
    }
    Serial.print("Done Counting");

//      stepper->forceStopAndNewPosition(0);
}

void stopStepper()
{
  Serial.println("Stop!");
  stepper->stopMove();
}

void runStepperToZero()
{
  stepper->enableOutputs();
  Serial.println("Zero");

}

//without encoder --> latency uS =  5sec  max:28 avg:6
// with encoder + print --> 5sec  max:33 avg:8

gin66, 2021-02-20:
Thanks for sharing the code. Just see, that you call disableOutputs() for every received byte. For a running motor with auto enable, this will may go undetected (or cause short motor hickup), but for sure provoke step loss. For the next release, disableOutputs() will have no effect on a running motor with auto enable. Besides: As enableOutputs() and disableOutputs(), are used quite often, better to use `stepper->setAutoEnable(false);`

The comment to setSpeedInHz() is outdated. BTW: the setSpeedInHz() expects a uint32_t and not a float.

> Just a note that the read pulse counter and steps appear to be in sync which is expected but when we print the pulses we have the same pulse printed couple times initially. We have set the acceleration to (10000000) to prevent any timing issues with acceleration.

In the attached log, I do not see the "_same pulse printed couple times initially_". Could you please elaborate more about the problem ?

vepicurean, 2021-02-20:
Thanks for the information on the disableOutputs() causing step loss with a running motor with auto enable.  We'll clean up that bit of code. 

Sorry about the comment on the setSpeedinHz() statement. Yes that comment should be changed to microsteps/sec. If that parameter is a 32 bit integer now (and not a float), then in theory setSpeedInHz(427) would give us 1.00078125 rev/min . I just want to make sure that we can never achieve exactly 1.0000000 rev/min (RPM) level of accuracy in terms of significant digits with the setSpeedInHz() function as an integer.   It probably does not make a huge difference anyway, as we can simply use the moveTo(25599) {in either direction} command and that seems to stop the motor shaft in exactly correct position on a consistent basis.

Sorry for the confusion. When observing the output data on the positioning, when we start the program, we notice the println command prints out the [0] position multiple times, before printing out the next position. I don't think we captured all the initial println with the [0] position when we created the document in the previous note we posted. Why would this be do you think?  Something to do with the acceleration at the start? 

Also, I notice something about the output data. Please look at the line. 

`21:31:21.499 -> -> TIME MILLIS: 2169 -> PULSE COUNTER -1: -18 -> Pulse Counter:
[-19]`

Notice that PULSE COUNTER -1: value of -18 is actually greater than than the Pulse Counter value of -19 ?? Why do you think this occurs?  

I notice that sometimes the PULSE COUNTER -1 value is less than the Pulse Counter value, sometimes it is the same, sometimes in a few cases it is one more (greater than) the Pulse Counter value. 

With this code example we sent you, the motor speed is fairly accurate to 2 RPM (not exactly 2 RPM, but close), but the positioning is exactly right..... It stops at position 25999 physically. 

Note: We physically set the 0 zero start point, and then we have a physical set of markers to denote 25,999 and 0 ..... The marker stops at the same spot every time using the moveTO(25999) command regardless of the speed we set.  

I'm more concerned that we can trigger events at position XX (such as triggering a camera to capture at position XX which the index table is moving at 1RPM, 2RPM, 3RPM) while the motor shaft is moving.  Position control while the shaft is moving is more important versus being able to run the step motor shaft rotation at exactly 1.000000 vs 1.00078 RPM.

We almost there if you feel we are on the right track with this code. Not we just need to able to use a proximity sensor (gives off a pulse one per revolution, need to track the rising edge and then confirm the rising edge) to physically set/calibrate the zero [0] pulse counter value relative to physical space.  Using a proximity sensor on the first rotation of the motor. Once we can do this physical space calibration accurately (via the proximity sensor) to set the zero value of the pulse counter (an accurate calibration to physical space), then we well on our way.

gin66, 2021-02-20:
Exact 1 rpm you can achieve with setSpeedInTicks(37500).
Exact 2rpm with setSpeedInTicks(18750).

If you read the pulse counter twice in sequence, different value means, that a pulse has happened between the two calls.

For reset the pulse counter with your proximity sensor, you have at two choices: polling or interrupt. Polling would be something like:
```
while(digitalRead(proximityPin) == LOW) {}
stepper->clearPulseCounter();
```

vepicurean, 2021-02-20:
Hi, thank for the information.  Greatly appreciated. 

How do I equate the value of 37500 in ticks (which I assume is related to CPU clock speed in Hz) to 1 rev/min on the motor shaft for this 25,600 step/revolution stepper/motor system?  What is the calculation? I was looking for some info on the setSpeedInTicks(XX) function, but could not find it in the library. 

I did not do a good job of explaining what I'm noticing on with the println values on the PULSE COUNTER -1 value vs the PULSE counter value. 

I get that the value will change between sequence reads. What I don't get is why is PULSE COUNTER - 1 has a value of -18, when PULSE COUNTER has a value of -19?  If PULSE COUNTER reads -19 at that time, should I not see PULSE COUNTER - 1 be always one more value negative (e.g. -20) than what the PULSE COUNTER value reads at that reading of the counter? Or are we doing sometime wrong vs what we think should be showing?

We'll try the polling method you suggest with the proximity sensor to set the 0 value to calibrate the pulse counter 0 point to the physical world, so we don't have to keep doing it manually by physically moving the motor shaft, and restarting the ESP32 to clear the pulse counter to zero.

gin66, 2021-02-20:
ticks is based on 16MHz clock (160MHz/5/2). 
`60s/25600 steps * 16000000 ticks/s = 37500 ticks/step`

The stepper runs with approx. 2ms per step. 
```
Serial.print(stepper->readPulseCounter() - 1);

      currentMillis = millis();
      if (stepper->pulseCounterAttached()) {
        Serial.print(" -> Pulse Counter: [");
        Serial.print(stepper->readPulseCounter());

```
The serial interface runs with approx. 90 us/char. So with 2ms can be output 22 chars.  ` -> Pulse Counter: [` are 20 chars. Now output of the first pulse counter value are 1 to 5 chars, so one pulse will nearly always be in, sometimes two and quite rarely no.

vepicurean, 2021-02-20:
Got it on the ticks/step calculation for the setSpeedInTicks(XX) function. Thank you. Appreciated. 

Thanks for explaining the behaviour on the printLn output. It looks like we are good there also in terms of accuracy.

I'll let you know how it goes on using the proximity sensor pulse (setting the pulse counter to zero on a rising edge, once per revolution) to accurately reset (clear) the pulse counter to zero during calibration with the physical world. We only need to do this once during calibration on the first revolution. 

After that first rising edge reset (clear to zero) it's calibrated.... But we can always do a check every revolution on the rising edge pulse of the proximity sensor on the subsequent revolutions to see that the read pulse counter is on the value zero when that rising edge is seen. Make sure it's not drifting. 

I think we're on track..... Thanks.

closed 2021-02-20 by gin66