gin66/FastAccelStepper #49: Ramping stopped working at 4000 and 8000 counts/rev

gin66, 2021-02-07:
> Interestingly, at 4000 and 8000 counts/rev the ramping stopped working: hard reversals; hard stops; hard velocity changes. But also, at 8000, after some commands, speed stops responding to encoder turn or monitor commands.---The program seems to lose its way.

gmoneti, 2021-02-07:
Some more exploration.  Below is output of: speed at increment of 1 rpm (cnt); rpm derived from `getSpeedInUs()` (rpmR); and `getSpeedInUs()`) (us/step).  At higher speeds,  cnt increase several times without commensurate speed change . . . until the us/step changes. So, is this a case of us/step too low to resolve 1rpm change?

(It starts at ~233us/step (occasional double). Motor resolution @1000 counts/rev.)

```
cnt: 616    rpmR: 619    speedInUs: 97
cnt: 617    rpmR: 619    speedInUs: 97
cnt: 618    rpmR: 619    speedInUs: 97
cnt: 619    rpmR: 619    speedInUs: 97
cnt: 620    rpmR: 619    speedInUs: 97
cnt: 621    rpmR: 619    speedInUs: 97
cnt: 622    rpmR: 625    speedInUs: 96
cnt: 623    rpmR: 625    speedInUs: 96
cnt: 624    rpmR: 625    speedInUs: 96
cnt: 625    rpmR: 625    speedInUs: 96
cnt: 626    rpmR: 625    speedInUs: 96
cnt: 627    rpmR: 625    speedInUs: 96
cnt: 628    rpmR: 625    speedInUs: 96
cnt: 629    rpmR: 632    speedInUs: 95
cnt: 630    rpmR: 632    speedInUs: 95
cnt: 631    rpmR: 632    speedInUs: 95
cnt: 632    rpmR: 632    speedInUs: 95
cnt: 633    rpmR: 632    speedInUs: 95
cnt: 634    rpmR: 632    speedInUs: 95
cnt: 635    rpmR: 638    speedInUs: 94
cnt: 636    rpmR: 638    speedInUs: 94
cnt: 637    rpmR: 638    speedInUs: 94
cnt: 638    rpmR: 638    speedInUs: 94
cnt: 639    rpmR: 638    speedInUs: 94
cnt: 640    rpmR: 638    speedInUs: 94
cnt: 641    rpmR: 638    speedInUs: 94
cnt: 642    rpmR: 645    speedInUs: 93
cnt: 643    rpmR: 645    speedInUs: 93
cnt: 644    rpmR: 645    speedInUs: 93
Stop
```

gin66, 2021-02-07:
Yes, the pulses are based on a 16MHz clock. Perhaps I should add a getSpeedInTicks() function ?

gin66, 2021-02-07:
By the way: getSpeedInUs() retrieves the value of the last setSpeedInUs() or setSpeedInHz().
Not the actual speed, while ramp up.

Actually, the setSpeedInHz() should not be limited to us....

Still potential to improve the interface

gin66, 2021-02-07:
Added:

pre-0.21.2:
- Two new API functions: setSpeedInTicks() and getSpeedInTicks()
- setSpeedInHz() and setSpeedInMilliHz() uses higher resolution than us

gin66, 2021-02-07:
Which fits to ramp stop working:
`M1 H32000 a100000 w3000 A100 H25000 U`

It will just jump from 32000steps/s to 25000steps/s without any ramp down.

gmoneti, 2021-02-08:
Would you explain that a bit more? Also, I don't see "H" `H25000` in the StepperDemo.ino.

gin66, 2021-02-08:
The latest version offers the possibilities to give the speed in step/s aka Hz. There comes the H from.

The problem was, that there has been two wrong limit comparisons been in the code. While deceleration this has forced the speed to jump to the requested speed. In addition I have seen, that my “improved” calculation needs more work and testing, so I have reverted the code to old version. This all is done in v0.21.2.

Hope your observed behavior is now ok. Could you please check ?

gmoneti, 2021-02-08:
Updated and ran the commands. Everything is smooth as expected. 

It also fixed an unexpected jolt, yesterday, as I ran the motor at ~500rpm then directly a command to ramp up from 0 to 1000rpm. I didn't expect the jolt on the new command. I had temporarily inserted this code to resolve it:
```
stepper->stopMove();
while (stepper->isRunning()) {}
...ramp command...
```
I no-longer need it, and everthing is back to logical and consistent. Great work!

gin66, 2021-02-08:
Yes, the ramp generator once a while has created an invalid command. Hope this is now resolved. Only good thing is, that my test suite is growing with either issue. But it is always annoying to have those bugs in the first place.

Great, that now it works for you. So we can close this issue.

gmoneti, 2021-02-09:
To clarify, my last comment referred to a different test than the subject of this issue#49. The faults mentioned when running at 4000 and 8000 counts/rev remain unchanged. Ramping is lost. Indeed, the program "loses its way" at 4000 and 3200 as well. At 2000 and below, everything appears to work normally.

Another symptom: I can increase the speed in one direction +/- , but then turning opposite direction the speed doesn't change. . . until count=0, which causes a hard stop.

gin66, 2021-02-09:
Could you please come up with a sketch or a sequence of commands in StepperDemo, which demonstrates this problem ? Yesterday I have checked several sequences before release of 0.21.2 and I could not find this issue.

gmoneti, 2021-02-09:
Ok, currently, StepperDemo does not reverse direction `b`:
 `M1 A50000 V60 U f w1000 b w1000 f w1000 S` Not even at 1000counts/rev.

gin66, 2021-02-09:
Just have tried on avr and esp32 this sequence with the V0.21.2 works on my side.

Actually there is a similar test case in the test suite:
`test_sd_11_328p/platformio.ini:build_flags = -D SIM_TEST_INPUT='"M1 V40 a1000000 w1000 b w1000 f w2000 S W"'`

This will run forward 1s, backward 1s and forward 2s. The number of steps generated are 99652, but position is 50584.  So the stepper must have run backwards. Actually it runs ~25000 forward, then ~25000 back and ~50000 forward.

gmoneti, 2021-02-09:
I just re downloaded the code. Ran StepperDemo with:
`M1 A50000 V70 U f w1000 b w1000 f S`. It only turns CCW. . . as before. I don't know. It worked a couple days ago.

gmoneti, 2021-02-09:
It decelerates on change direction command, but then re-accelerates in the same direction with `b` and `f`.  With `P` move 25000 to -25000 or vice versa, it makes 50000 moves in same direction.

gin66, 2021-02-09:
Actually I have rebased the code to fix a typo in the commit message of the 0.21.2 
The 0.21.2 has been done yesterday.

Strange, that it works on my setup. Exactly the same commands. 
Does your motor run backwards at all ? BTW: A lower case `u` will put the motor in unidirectional mode (positive direction), until the next reset. I have added it to test motor without direction pin, so I do not need to change the hard coded assignment in StepperDemo

gin66, 2021-02-09:
Perhaps check your hw and try the low level direct drive with `T`. This should run 3200 steps in one direction and then 3200 steps in the opposite.

gmoneti, 2021-02-10:
The motor runs both directions with my sketch. At above 2000counts/rev, all else equal, the ramping is lost---as per this issue#49. 

I tried StepperDemo with `T`; it only goes one direction. Also tried `R` directional move. Nope. In all cases, motor moves only CCW.

gmoneti, 2021-02-10:
Solved single direction problem: my bad; I redownloaded StepperDemo so many times that, since yesterday, I forgot to adjust input pins settings. Apologies.

StepperDemo with `M1 V41 a50000 w1000 b w1000 f w2000 S W` works without issue to 8000 counts/rev (my motor limit). Will continue to seek why ramping is lost with my sketch using similar commands at 4000-8000 counts/rev.

gmoneti, 2021-02-10:
I tried the ClearPath sample step and direction code [SingleAxisDemo.ino](https://github.com/Teknic-ClearPath/ClearPath-Arduino-Step-And-Direction/blob/master/Arduino%20Uno/ClearPathStepGen/Examples/SingleAxisDemo/SingleAxisDemo.ino). The library is very limited, but it uses a ramping code as well. It works at high counts/rev. I wonder how fundamentally different it is from fastAccelStepper?

gin66, 2021-02-10:
Difference is for avr:
* ClearPath uses a 2ms rate interrupt (500Hz) and in every interrupt send precalculated impulse bursts to up to 6 steppers in parallel.
* FastAccelStepper produces only single steps with (during coasting) equal time distant pulses, but supports only two steppers

I assume, if the impulse bursts cover more than one step, this may cause problems for a running motor. With 32microsteps per step, then this would mean from >16kHz could be an issue. Another point: If the requested speed is e.g. 625 pulses/s, this would mean, ClearPath needs to send 2-1-1-1 (5 pulses every 8ms). FastAccelStepper will send 1-1-1-1-1 equidistant pulses within 8ms. onsequently I hope, that with FastAccelStepper the motor will run smoother. If you need more than two motors, then ClearPath can support this.

The ramp calculation is interesting and I will have a deeper look. Originally i have thought 8bit resolution for internal calculation is sufficient, but with slow ramps this resolution can be noticed.

gin66, 2021-02-10:
and FastAccelStepper supports super slow speed (<1 pulse/s) and has a better time resolution for the coasting speed, because ClearPath accepts only steps/s, while FastAccelStepper accepts milliSteps/s and us/Steps.

gmoneti, 2021-02-10:
Thanks for the insight. I still need to understand better timers and registers, etc., to read that code clearly. Here is another ramping sketch you might find interesting. It's  [sinusoidal](https://forum.arduino.cc/index.php?topic=605569.0). Implemented very simplistically perhaps, but easy to understand.

gin66, 2021-02-10:
ahead a quick look at the sketch. using double and sinus on avr plays not well along with high speed.

I currently think for few days, if I can afford a calculation in float needing ~80us+ or keep my 8bit mantissa calculation with takes only 8us...

And partly it is a waste of time, because the esp32 would not have this limitation, but still I like to keep the support of avr and as much code shared between the two as possible.

gmoneti, 2021-02-11:
> Difference is for avr:
> 
> * ClearPath uses a 2ms rate interrupt (500Hz) and in every interrupt send precalculated impulse bursts to up to 6 steppers in parallel.

"The ISR is set to 2KHz, nominally." So, it uses a 0.5ms rate interrupt. Or am I mistaken?

gin66, 2021-02-11:
no.  you are right, I have mixed the 2ms and 2kHz from memory. For me the key point is, that ClearPath uses bursts of steps at a fixed rate. In contrary to this, FastAccelStepper creates the steps with a variable frequency. Both approaches have advantages and disadvantages. For sure supporting six steppers on avr is a big plus. 

Only in order to bring the max. step rate on par with esp32, the concept of bursts of 1-16 may be interesting. Using esp32 allows me to drive the NEMA-17 steppers to the limits, which I cannot with the avr.

gmoneti, 2021-02-14:
I ordered a couple esp32 boards so that I can better test Clearpath motors with fastAcelStepper library.

gin66, 2021-02-14:
In the meantime, I have improved the poorman’s float implementation by one more bit and different math routines. Now the speed steps, I have observed with low accelerations, I cannot hear anymore. So, no need to use real float.

With esp32 the step frequency can go up to 200kHz. So the ClearPath can be driven to max. speed at 4000steps/rev.

Would be interesting to know, what application of the ClearPath you are working on.

Guess for now, this issue can be closed. Any problem with esp32, then please file a new issue.

gmoneti, 2021-02-14:
Currently, I'm just exploring CNC technology; instrumentation; micro-controllers; programming; machine chassis design/materials; bearing systems; the used motor is an opportunity to learn about high-end motors. The goal is to design and build a relatively low-cost, high-quality CNC/maual lathe... or possibly a multi-axis milling/turning center. We'll see.

gmoneti, 2021-02-16:
The esp32 arrived surprisingly fast. I just finished getting a basic orientation on esp32 and setting up the IDE for it; I swapped it in to my test code and voila`, it worked first time! I tested at 4000-8000 counts /rev and, as expected, no glitches or code "losing its way"; no issues to report.

Well, there is one thing we touched on before. The speed resolution isn't quite right at high speed---speed doesn't change for 2 or more counts of the encoder. Example output at 8000 counte/rev; doubling starts at cnt:26:

```
cnt: 15    rpmUs: 300    rpmTicks: 300    us/step: 25
cnt: 16    rpmUs: 326    rpmTicks: 326    us/step: 23
cnt: 17    rpmUs: 341    rpmTicks: 341    us/step: 22
cnt: 18    rpmUs: 357    rpmTicks: 357    us/step: 21
cnt: 19    rpmUs: 375    rpmTicks: 375    us/step: 20
cnt: 20    rpmUs: 395    rpmTicks: 395    us/step: 19
cnt: 21    rpmUs: 417    rpmTicks: 417    us/step: 18
cnt: 22    rpmUs: 441    rpmTicks: 441    us/step: 17
cnt: 23    rpmUs: 469    rpmTicks: 469    us/step: 16
cnt: 24    rpmUs: 469    rpmTicks: 469    us/step: 16
cnt: 25    rpmUs: 500    rpmTicks: 500    us/step: 15
cnt: 26    rpmUs: 536    rpmTicks: 536    us/step: 14
cnt: 27    rpmUs: 536    rpmTicks: 536    us/step: 14
cnt: 28    rpmUs: 577    rpmTicks: 577    us/step: 13
cnt: 29    rpmUs: 577    rpmTicks: 577    us/step: 13
cnt: 30    rpmUs: 577    rpmTicks: 577    us/step: 13
cnt: 31    rpmUs: 625    rpmTicks: 625    us/step: 12
cnt: 32    rpmUs: 625    rpmTicks: 625    us/step: 12
cnt: 33    rpmUs: 682    rpmTicks: 682    us/step: 11
cnt: 34    rpmUs: 682    rpmTicks: 682    us/step: 11
cnt: 35    rpmUs: 682    rpmTicks: 682    us/step: 11
cnt: 36    rpmUs: 750    rpmTicks: 750    us/step: 10
cnt: 37    rpmUs: 750    rpmTicks: 750    us/step: 10
cnt: 38    rpmUs: 750    rpmTicks: 750    us/step: 10
cnt: 39    rpmUs: 750    rpmTicks: 750    us/step: 10
cnt: 40    rpmUs: 833    rpmTicks: 833    us/step: 9
cnt: 41    rpmUs: 833    rpmTicks: 833    us/step: 9
cnt: 42    rpmUs: 833    rpmTicks: 833    us/step: 9
cnt: 43    rpmUs: 833    rpmTicks: 833    us/step: 9
cnt: 44    rpmUs: 833    rpmTicks: 833    us/step: 9
cnt: 45    rpmUs: 937    rpmTicks: 937    us/step: 8
cnt: 46    rpmUs: 937    rpmTicks: 937    us/step: 8
cnt: 47    rpmUs: 937    rpmTicks: 937    us/step: 8
cnt: 48    rpmUs: 937    rpmTicks: 937    us/step: 8
cnt: 49    rpmUs: 937    rpmTicks: 937    us/step: 8
cnt: 50    rpmUs: 1071    rpmTicks: 1071    us/step: 7
```

gmoneti, 2021-02-16:
FYI, here is a 1rpm per count version of above: [cnt_rpm_us.txt](https://github.com/gin66/FastAccelStepper/files/5990230/cnt_rpm_us.txt).
The first doubling appears at 92rpm.

gin66, 2021-02-16:
What's the difference between rpmInTicks and rpmInUs ?

I assume rpm 937 is the rotations/min. With 8000 counts/rev, this means 8000 * 937/60 = 124933 steps/s => 8us/Step.
rpm 1071 and 8000 counts/rev gives 8000 * 1071/60 = 142800 steps/s => 7us/Step.

With setSpeedInUs(), there is no chance to get any speed between 7us and 8us.
With setSpeedInHz(freq), there are plenty of values between 124933 and 142800.
The internal calculation is 16,000,000/freq. This gives for 124933 => 128 and for 142800=>112.
This means internally, there are between 124933 and 142800 still/only 16 speed steps: 112..118.
That comes from the 16 MHz clock, which gives only 16 ticks per us.

The issue is, that the API still is a bit too complex:
- setSpeedInHz/setSpeedInTicks: sets the maximum speed
- getSpeedInHz/getSpeedInTicks: read the maximum speed, but not(!) the current speed
- getPeriodAfterCommandsCompleted(): read the speed (couple of ms in the future) in us.

Now I have split getPeriodAfterCommandsCompleted() into getPeriodInUsAfterCommandsCompleted() and getPeriodInTicksAfterCommandsCompleted() in the 0.22.0.

The new function is used in StepperDemo:
With `M1 H124833 A10000 f` the result is ` M1: @1283882 nonstop QueueEnd=1286382/8us/128ticks COAST`.
With `M1 H142800 A10000 f` the result is ` M1: @1283882 nonstop QueueEnd=1286382/7us/112ticks COAST`.
With `M1 H130000 A10000 f` the result is ` M1: @1283882 nonstop QueueEnd=1286382/7us/123ticks COAST`.

gin66, 2021-02-16:
The esp32 is compared to avr much more powerful and the pulse generation is quite different, too. Especially the interrupt load is much less than with the avr.

gmoneti, 2021-02-17:
rpmInTicks refers to rpm computed from the output of `getSpeedInTicks()`
rpmInUs refers to rpm computed from the output of `getSpeedInUs()`
I wanted to compare the outputs of the two functions, but found none. Upon reflection, perhaps that is because I have continued to set speed with `setSpeedInUs()` ? I will try the other setSpeed functions.

> I assume rpm 937 is the rotations/min.

 Your assumptions are correct.

> The internal calculation is 16,000,000/freq. . .

To be clear, the output (both) was with the new esp32 (devkit V4)

> getSpeedInHz/getSpeedInTicks: read the maximum speed, but not(!) the current speed

OK, but still, the speed did not audibly nor via MSP display change for several counts of the encoder; changed only when micros changed.

I'll checkout the new StepperDemo. Thanks!

gmoneti, 2021-02-17:
I tested the new StepperDemo; not sure what I was supposed to see.

I tested getPeriodInTicksAfterCommandsCompleted() in another run of 1rpm steps to 1000. I found the output was the same as getSpeedInTicks(). Here is the output file with notes at end: 
[cnt_rpm2.txt](https://github.com/gin66/FastAccelStepper/files/5999122/cnt_rpm2.txt)

gin66, 2021-02-18:
getSpeedInTicks() and getPeriodInTicksAfterCommandsCompleted() are identical, if the target speed is reached.

What I have meant with reference to StepperDemo is this:

With M1 H124833 A10000 f the result is M1: @1283882 nonstop QueueEnd=1286382/8us/128ticks COAST.
With M1 H142800 A10000 f the result is M1: @1283882 nonstop QueueEnd=1286382/7us/**112ticks** COAST.
With M1 H130000 A10000 f the result is M1: @1283882 nonstop QueueEnd=1286382/7us/**123ticks** COAST.

Same 7us, but two different tick values.

From your cnt_rpm2.txt:
```
rpm = encoder count = 1rpm per count.
ticksToRPM = rpm calculated from getSpeedInTicks()
perInTicks = getPeriodInTicksAfterCommandsCompleted()
speedInTicks = getSpeedInTicks()

rpm  ticksToRPM  perInTicks  spdInTicks
986       992          121       121
987       992          121       121
988       992          121       121
989       992          121       121
990       992          121       121
991       992          121       121
992       1000          121       120
993       1000          121       120
994       1000          120       120
995       1000          120       120
996       1000          120       120
997       1000          120       120
998       1000          120       120
```
There is no integer value between 120 and 121. 120 translates for your motor in 1000rpm and 121 translates to 992rpm.
So the big jump is not avoidable and you have hit the limitation of the current implementation.

Even though, the esp32 would have the possibility to increase the resolution by a factor of 5 (using 160MHz as base frequency and not 32MHz), this cannot be done on AVR. As this would be a major change to the library, breaking the compatibility with avr,  I have no plan to implement this.

gmoneti, 2021-02-18:
Thank you for the generous explanation. It wasn't clear to me that the esp32 code had the same limitations as the UNO---in this regard.  That clears up all the remaining questions or thoughts I had. Thanks.

closed 2021-02-18 by gin66