gin66/FastAccelStepper #50: Why Timer4 only for Arduino Mega2560?

Nick507, 2021-02-17:
Hi! It is clearly stated in doc that for Arduino Mega2560 we can use only pins 6,7,8 (Timer4). But what is the reason? Can it be improved? In My project I have connected lcd display on mega2560, and these pins are used for display. I want to switch to Timer5 (pins 44,45,46), but it does not work :(

gin66, 2021-02-17:
The first user requesting AtMega2560 support has opted for Timer 4, so it was done. No other reason. Have you tried this from the README to change the timer ?

Similar to ATmega328, but instead of timer 1, timer 4 is used.

For users of platformio, the used timer can be changed to either 1, 3, 4 or 5. For e.g. timer module 3 add to platformio.ini under build_flags:
```
build_flags = -DFAS_TIMER_MODULE=3
```

or better:
```
build_flags = -Werror -Wall -DFAS_TIMER_MODULE=3
```

For arduino users, the same can be done by defining the flag *before* including the `FastAccelStepperEngine.h` header (as per info ixil).
e.g.
```
sketch.ino
----------
#include <Arduino.h>
#define FAS_TIMER_MODULE 3
#include <FastAccelStepper.h>
/* ... */
```

gin66, 2021-02-17:
If it does not work, then you need to put the library in your project and change the define in AVRStepperPins.h line 27 to:
`#define FAS_TIMER_MODULE 4`

I haven’t tried it myself, because I use platformio, where it works reliably.

Nick507, 2021-02-17:
Yes, I've tried #define FAS_TIMER_MODULE 5 before #include <FastAccelStepper.h>, and it does not work. I'll try to change AVRStepperPins.h, but I bet result will be the same, because there is ifndef. I think that the reason might be in ISR declaration in avr.io, which goes before any custom libraries, but this is just guess.

Nick507, 2021-02-17:
I can't believe, but change define in AVRStepperPins.h really works. But when I add define before include lib it does not work. Why???

gin66, 2021-02-17:
Good that this works. If you find a way, how to get this compiler define done via the ArduinoIDE, then I would happily use that. Cause I have no idea.

Personally I would advise to use platformio, where it can be done easily.

Pity is, that for any library update, you need to fiddle around with the code again. Sorry for that.

mikhail11111, 2021-02-17:
> I can't believe, but change define in AVRStepperPins.h really works. But when I add define before include lib it does not work. Why???

Add define After include.

Nick507, 2021-02-17:
> > I can't believe, but change define in AVRStepperPins.h really works. But when I add define before include lib it does not work. Why???
> 
> Add define After include.

Kind of joke?

mikhail11111, 2021-02-17:
> > > I can't believe, but change define in AVRStepperPins.h really works. But when I add define before include lib it does not work. Why???
> > 
> > 
> > Add define After include.
> 
> Kind of joke?

No joke. The last define will take effect, if you define before include, it will be redefined in the include.

gin66, 2021-02-18:
Apparently there is no solution besides changing the code. 
Readme has been updated.

closed 2021-02-18 by gin66