gin66/FastAccelStepper #52: setCurrentPosition(0) does not reset moveTo(##), moveTo(##) still remembers old positon

bkapadia01, 2021-02-21:
We are at the point now where we're relating the pulse counter and stepper position both the physical world. We are able to reset both counters to 0, with the marker in the physical world, but using commands such as moveTo/move() we are running into limitations or we are not using it correctly where it seems to remember the original position even though we reset the pulse and position counters to 0. The construct for the stepper are not being reset even though the position counter is reset.

In the code below we are setting moveTo(-25599). As the motor is running to position -25599, we are setting the stepper position and pulse counter to 0. Although in the output the pulse counter and stepper position counters reset to 0 the moveTo/move(-25599) functions do not reset to zero and continues to remember the old position.
As a result the stepper moves -25599 steps instead of moving -25599 from the point the stepper position and counter has been set to 0.

```
void runStepperMotor()
{

  stepper->enableOutputs();
  stepper->moveTo(-25599);
  
  while(!((stepper->readPulseCounter() - 1) == (-25600))){
//  Serial.print(" -> PULSE COUNTER -1: ");
//  Serial.print(stepper->readPulseCounter() - 1);

  currentMillis = millis();
  if (stepper->pulseCounterAttached()) {
      Serial.print(" -> Pulse Counter: [");
      Serial.print(stepper->readPulseCounter());
      Serial.print(']');
      Serial.print(" -> Stepper Positon: [");
      Serial.print(stepper->getCurrentPosition());
      Serial.println(']');
      
//      Serial.print("TIME MILLIS: ");
//      Serial.print(currentMillis);
      }

  if(Serial.available()>0){
     if (Serial.read() == 'C') {
            Serial.print("Clear Counter");
            stepper->clearPulseCounter();
            stepper->setCurrentPosition(0);
        }
   }
    
    } //end of while loop
  
  Serial.print("Done Counting");
  stepper->clearPulseCounter();
  stepper->setCurrentPosition(0);

}

gin66, 2021-02-21:
Hmmm. in the code above, the counters are cleared at the end of the revolution aka after outputting “Done Counting”. Could this be the reason ?

vepicurean, 2021-02-21:
We clear the pulse counter and setCurrentPosition(0) manually by inputting the "C" character within that "if(Serial.available()>0 {} statement. After we have started the original move or moveTO statement at the beginning (while the motor shaft is still moving), but before the the while loop ends. The "Done Counting" executes only after the while loop ends. Then the last thing is to again clear the Pulse counter, and again setCurrentPosition(0). 

Let me know if this does not make sense to you. 

Separate from the move() and moveTO() retaining their original position instruction, even after the Pulse counter clear and the stepper contruct set to 0 position, another thing that is bothering me is that we sequentially clearPulseCounter and then the next statement is setCurrentPosition(0).... there's some time change in between those two statements. So we are not executing both commands (clear Pulse counter and setCurrentPosition(0)) at the exactly same time. I can see the Pulse Counter and the stepper position pulse numbers are very close, but not 100% in sync. 

Thanks for your input and time.

gin66, 2021-02-21:
Yes, and whatever this “C” initiated counter clear has meant, it is completely overwritten after the while loop. Should be logical.

The sequential clearance of clearPulseCounter() and setCurrentPosition() is only a problem, if the motor is running. The advise is in FastAccelStepper.h:
```
  // Set the current position of the stepper - either in standstill or while
  // moving.
  //    for esp32: the implementation uses getCurrentPosition(), which does not
  //               consider the steps of the current command
  //               => recommend to use only in standstill
```
So better to perform after the while loop, when the motor is in standstill, this command to synchronize the two counters:
`
stepper->setCurrentPosition(stepper->readPulseCounter())
`
Then you can run to your physical zero by `stepper->moveTo(0)`

vepicurean, 2021-02-21:
If we have to be able to clearPulseCounter() and setCurrentPosition() while the motor is moving. This is how calibration to the physical world is usually (always, really) done with calibration point to the physical world. 

Such as using the Z index pulse on an incremental quadrature encoder or using the pulse from a proximity sensor. We're just using the C input command first (temporarily to get the code right) to simulate the pulse from the proximity sensor..... Which is made even more stringent by the fact we want the clearPulseCounter and setCurrentPosition(0) to happen only at the transition from low to high (rising edge of the pulse from the proximity sensor). 

If FastAccelStepper can't be developed to be able to do this calibration clearPulseCounter and setCurrentPosition(0) while moving, then in won't be able to work on a real world positioning system. 

Do you believe this functionality is something that can be developed?

gin66, 2021-02-21:
Apparently there is a misunderstanding. What I propose:
1. While motor is running, wait for the sensor transition from low to high. As soon as this is reached, clear the pulse counter
2. when the motor is in standstill, update the motor position from the pulse counter like this: `stepper->setCurrentPosition(stepper->clearPulseCounter())`
3. From now on, you can run the motor to any position you want with moveTo() and you get the exact rotation angle from readPulseCounter(). While the motor is running getCurrentPosition() may be off a few steps into the future, but readPulseCounter() will be always up to date.

BTW: calibration of stepper motors are normally done at a lower speed. If needed a two stage strategy can be applied:
1. run fast and stop at the transition point with few pulses overshoot.
2. Revert direction in slow speed until the sensor transition again.
3. Perfectionist revert again, to check again. This is, because the low-high and high-low transition of a sensor can deviate by a couple of steps.

If you cannot make it run with FastAccelStepper, then sorry for this. From my point of view, with the right strategy in place, it can be done very well. Just I do not know, what your application is, so you need to decide on your own.

vepicurean, 2021-02-21:
Some background. We've had the original prototype of the index positioning machine (table) in production for a long time. We are just rebuilding it now with a step motor, instead of an induction motor that was used originally long ago.  

The users want it running the same way and calibrating with the motor shaft position to the physical reference point while the motor/index table is moving as it as has always been. I understand why for important process/logistics reasons. It makes sense to do it this way. They don't want to stop the motor/index positioning table to accurately calibrate the motor shaft rotation with the physical positioning of the index table. The purpose of changing to the step motor is to increase the accuracy and stability and also make their lives easier from a UX perspective. We're on the right track. Just need to get this calibration in line with how their want things done for their long proven workflow. 

We do the calibration to the physical space reference sensor with the motor running on the rising edge of either the Z index pulse of a quadrature incremental encoder or in this case, we want to see if a pulse from a proximity sensor will achieve the same objective as the quadrature encoder in terms of accuracy (when using the Z index pulse on the quadrature encoder).

Close enough we can not do (yes, we have to be that precise in positioning accuracy). We need to get the pulse counter and the stepper position counter as close (in sync) as technically possible while the motor/index table is running. 

Our speeds are not high. 1RPM to maybe 3RPM at the max. During the calibration phase on the first rotation of the motor shaft/index table, we could go even slower within reason if needed. 

I just tried to use the statement `stepper->setCurrentPosition(stepper->clearPulseCounter());`  The issue right now is that statement produces an error because the clearPulseCounter() function returns the wrong data type for this situation.  

It would be great to have this  nested `stepper->setCurrentPosition(stepper->clearPulseCounter());` statement work when the motor is running..... and also when the motor is still  (would achieve how we do things now with the old motor and workflow logistics). 

When the table is running, all I can do right now is two sequential statements
stepper->clearPulseCounter();
stepper->setCurrentPosition(0);

The result of the two sequential statements above is close between the two counters, but they are slightly off. 

With the motor and index stable still, yes, we can achieve probably the closely sync between the two counters as follows:

stepper->stopMove();
stepper->clearPulseCounter();
stepper-setCurrentPosition(stepper->readPulseCounter();

But the above is not practical to stop the table for physical space calibration given our logistic workflow. 

Anyway we can help with or try to achieve the closely possible sync while the motor is running. We're willing to test and give detailed feedback. 

Thanks for your help and time.

gin66, 2021-02-21:
Just see a typo:
`stepper->setCurrentPosition(stepper->clearPulseCounter())` 
should be
`stepper->setCurrentPosition(stepper->readPulseCounter())`

BTW: The table is constantly running and does never stop ? So why not forget about trying to setCurrentPosition() at all and just use the stepper driver to control the speed. The absolute angle of the stepper can be retrieved at the moment from readPulseCounter().

vepicurean, 2021-02-21:
There's no typo on my part.  Sorry for the confusion....  I'll be more clear below: 

What we were trying to explain in the previous post: 

It would be great if `stepper->setCurrentPosition(stepper->clearPulseCounter());`was able to sync and reset to zero both the pulse counter and the stepper counter at the same time, instead of having to use the two sequential statements as follows while the motor is running. 
```
stepper->clearPulseCounter();
stepper->setCurrentPosition(0);
```
The single line statement `stepper->setCurrentPosition(stepper->clearPulseCounter());` will not work at any time, whether motor is running or stopped. Because of the issue with the data type.  I know that. 

What we're being forced to do because of the limitations with the FastExcelStepper library (unless there is a better way that can be developed or suggested when the motor is running) is the following:
```
stepper->clearPulseCounter();
stepper->setCurrentPosition(0);
```

The speed has to be user configurable, The users will do this within the user interface software. Their input gets translated into commands from a user interface program with in turn, these commands a translated to the program within the ESP32 MCU which drives/controls the stepper driver (the ESP32 MCU is simply now a PLC). The users only every interfaces with a simple web and/or mobile user interface. 

They never see or interact with the code we're developing/using now on this core positioning system code for the index positioning machine/table (it's all be proven in the field for a while). All this command line stuff gets hidden from the user. They don't ever touch the stepper driver, motor, etc. 

Please let us know if there is anything you can do within the FastAccelStepper library to solve the limitation on having to use the two sequential statements that we're currently having to employ to re-set the two counters to zero when seeing the proximity sensor. 

We've got the rising edge of the proximity sensor pulse programmed properly on the proximity sensor now. Initial indications in testing are it's working as it should. 

The remaining main issue is the sync and re-set to zero between the pulse counter and the stepper counter. 

Please let us know if you want us to test any new methods to eliminate the sequential statements above to sync and zero the two counters as precisely as possible. 

Thanks for your help and time.

gin66, 2021-02-22:
The only (better) way to sync the position with the pulse counter is during standstill. Even if setCurrentPosition() for esp32 would take into consideration the count of the current queue command, still this processing takes time and any pulse generated during this processing, will cause an off by one between pulse counter and position.

If there is no standstill, then the question comes up, why the stepper position is necessary at all. From what I understand about your application, that only constant speed is needed, which can be adjusted be setSpeed(), and the angle of the motor, which can be derived in real time by readPulseCounter(), to trigger whatever processing is needed. So apparently setting the position of the stepper is not needed and runForward() is more appropriate, if the stepper is not intended to stop at any position. If this is not correct and the stepper should stop, then there is the standstill for a perfect one time sync between readPulseCounter() and stepper position.

gin66, 2021-02-22:
As there is no apparent bug or need for improvement, this issue can be closed.

vepicurean, 2021-02-22:
?? Maybe we're not understanding each other. 

If the pulse counter and the stepper counter are not in sync and can not be set to zero at the same time, I'm not sure the library is useful in our application. 

I'm just doing and admitted work around with the way the library is set up right now. I said they counters are close, but there are not in sync and we agreed on that point, yes?

vepicurean, 2021-02-23:
If you want feedback on the library with respect to syncing and setting the two counters while the motor is running to make it useful for our application, we're happy to work with you. 

Your guessing and interpretation of our application (which is long field proven, we're just doing a replacement of the motor) needs are wrong. 

Thanks for all your help and time. Greatly appreciated.

gin66, 2021-02-23:
The pulse counter is additional functionality, only available on esp32. It has its purpose and use.

Syncing of pulse counter and motor position can be reliably done only in standstill. All else is from my current view quite complicated, especially if one user would like to see this work at 200kHz, too. Anyway, this library is not closed source and everybody is free to implement a fancy sync mechanism, which I am currently not able to figure out with reasonable time. If FastAccelStepper is due to this not suitable for your application, then it is like it is.

There are two causes, why getCurrentPosition() and readPulseCounter() on a running motor are not perfectly in sync:
1.) for getting those values two reads are necessary. If after the first read, a pulse is generated, then it is out of sync already.
2.) getCurrentPosition() is bit complicated, because it has to calculate backwards the current position from the end of the queue. This could be improved by spending four bytes more for each queue entry.

And yes, my guessing and interpretation may be wrong based on the provided information. Consequently I do not understand your problem fully and as such cannot support you on a solution or even see the benefit in it.

vepicurean, 2021-03-02:
I hear what you're saying. But for some reason, even from a still position (as you indicate), resetting both counters to zero, when we start again and come back around to the proximity sensor, we're are still about 10 pulses (steps) off (very close, but not exact). We're putting it down as something we have to enhance going forward. We have a work around, but it's some that we will need to find a fix for long term.  We'll send you some feedback as we experiment with it and try to find a what to get these 100% in sync. 

Thanks for your help and time.

gin66, 2021-03-05:
The position calculation is now improved for esp32 in v0.23.0. The delta between pulse counter and current position is nearly zero. My esp32 hw tests accept a max. deviation of 1. Until now looks ok.

vepicurean, 2021-03-06:
Thanks. We'll test it out this weekend. Appreciated.

gin66, 2021-03-12:
as seems now issue remaining, can close the issue

closed 2021-03-12 by gin66