gin66/FastAccelStepper #53: move() does not allow more than 1,000,000,000 steps

TwinLizzie, 2021-02-22:
```
int gearRatio = (53.575757576 * (49 / 12));
int stepsPerRev = 200;
int microsteps = 8;

int formula = (microsteps * (gearRatio * stepsPerRev));

stepper->move(2500 * formula)
```
This line of code works fine.

```
stepper->move(25000 * formula)
```
But as soon as you add another figure to it, it produces fewer steps.

Only moveTo() will give you the full number of steps. move() seems to only allow a limited amount (1 billion steps I think). 

For reference:
25000×(8×((53.575757576×(49÷12))×200)) = 8750707070.746666667
2500×(8×((53.575757576×(49÷12))×200)) = 875070707.074666667

gin66, 2021-02-22:
First a comment for the gearRatio: Does the compiler calculate `int(53.575757576 * 49 / 12) = 218` or
`int(53.575757576 * (49 / 12)) = int(53.575757576 * (4)) = 214` ?

Second, I avoid using int, because I am always not sure, what data type I get on a compiler/uC. On an avr apparently int is 16 bit. Guess on esp32 is 32bit ? Anyway, assume you are using avr and have a 16bit int (-32768...32767), then the value of formula is already overflowing and will be 14720 (in case gearRatio is 214) and not 342,400. Now I again cannot predict, how the compiler calculates 25000 * 14720 (16bit) with an expected 32bit result. In best case, the compiler will yield 368,000,000.

If the int is actually 32bit, then the result will be 342,400 * 25000 = 8,560,000,000 which overflows an int32_t and the result would be only -29,934,592 (so backward movement).

BTW: if the stepper is at position 0, then move(arg) and moveTo(arg) both must run the motor to position arg.

Just had a short test with StepperDemo and a relative move (aka move()) with 2,000,000,000 works as expected. But I have not waited 2 billion steps, so I assume it will run really to that value. A value of 3,000,000,000 is already overflowing an int32_t and the stepper starts to run backwards towards -1,294,967,296, just as expected.

Could you please - in a first step - just use numbers as arguments to move() and moveTo() and afterwards extend with the calculation ?

TwinLizzie, 2021-02-22:
Hi there,

Sorry, I believe I meant to say that yes, the move() function won't let you move more than 2,000,000,000 in one direction. (not 1,000,000,000 as I put in the title)

But this is still a very small number. It would be handy to be able to move longer distances than this, without having to use MoveTo().

Otherwise, if you have a gearbox with a ratio of around 214, you will only be able to move short angles at a time. Maybe there needs to be moveLong() function which allows unlimited steps to be moved in one particular direction without interruptions.

gin66, 2021-02-22:
Using max speed of avr: 25 kHZ, then it takes 22h to run 2,000,000,000 steps.
With max speed of esp32: 200kHz, then it is still 10000s aka approx. 3h.

Does it hurt to issue every 3h just a single move(1000000000) ?

gin66, 2021-02-22:
BTW: move() and moveTo() have the same limitation in regard to max step count.
Just moveTo() uses absolute position and move() relative position, but both will run at max +/-2,147,486,647 steps in one direction

gin66, 2021-02-22:
If you need continuous running in one direction, then runForward() and runBackward() are useful. They will run continuously in the  requested direction with the position wrapping around

TwinLizzie, 2021-02-22:
Hi again,

Issue solved by using **long** integer instead of **int** for formula variable.

Closing. Thanks for the clarifications

closed 2021-02-22 by TwinLizzie