gin66/FastAccelStepper #54: Does pulse counter take into account the setacceleration?

bkapadia01, 2021-02-25:
Hi Gin66,

It appears that the pulse counter does not take into the set acceleration. The set acceleration we are currently using is a high value of 1000000 since we do not want any delay in the acceleration. 

When running the stepper and pulse counter from a standstill position at the z-index(external reference for zero mark) we notice that the counter reads `0` for several counts approx. 10-12 counts. This is causing reduced total counts after one revolution to be 25590 instead of 25600. We are missing about 10 ticks/counts from the total.

In the case that the motor is not starting from a standstill at the z-index point where the pulse counter is reset the pulse counter behaves as expected and after one revolution we are getting total ticks of 25600.

Let me know if you have any suggestions for dealing with the pulse counter having several `0` when starting from a standstill position resulting in lost counts.

gin66, 2021-02-25:
The pulse counter is connected to the stepper pin and will count all steps generated independent from acceleration. This is done by the pulse counter hardware.

What are the counts (10-12) you refer to ?

bkapadia01, 2021-02-25:
I see, thanks for clarifying. 
10-12 counts refer to the number of initial 0's we see when starting the motor from a standstill.

Starting stepper from Standstil at z-pulse index:
```
Pulse Counter: [0] <-- starting from z-pulse index
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [-1]
Pulse Counter: [-2]
...
...
... 
Pulse Counter: [-25590] <- 1 revolution completed EXPECT -25600
Pulse Counter: [0] <-- z-pulse index triggered
Pulse Counter: [-1]
Pulse Counter: [-2]
...
..
...
Pulse Counter: [-25599] 
Pulse Counter: [-25600] <- 2nd revolution completed
Pulse Counter: [0] <-- z-pulse index triggered
Pulse Counter: [-1]

```

Starting stepper before the z-pulse index: 
RANDOM START POINT -  in this case 2583 counts before z-index
```
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [0]
Pulse Counter: [-1]
...
...
...
Pulse Counter: [-2581]
Pulse Counter: [-2582]
Pulse Counter: [-2583] 
Pulse Counter: [0]  <--- z-index triggered
Pulse Counter: [-1]
Pulse Counter: [-2]
Pulse Counter: [-3]
...
...
...
Pulse Counter: [-25598]
Pulse Counter: [-25599]
Pulse Counter: [-25600]
Pulse Counter: [0]  <--- z-index triggered
Pulse Counter: [-1]
Pulse Counter: [-2]
```

gin66, 2021-02-25:
The `Pulse Counter: [0]` are printed in a loop ? using a delay or any other physical event ?

Anyway, the pulse counter counts exactly the steps sent to the motor. If 10 steps are missing from standstill and all fine starting from a moving motor, then I would guess, that either your mechanics is not as stable as expected or the z-index trigger is toggling and resetting the pulse counter.
=> In a nutshell, the pulse counter is correct couning L->H-transitions on the step pin, so cannot be the cause

gin66, 2021-03-05:
stale

closed 2021-03-05 by gin66