gin66/FastAccelStepper #56: Several issues

SorenBNielsen, 2021-02-28:
Dear gin66

Thank you for your great ideas in this library that takes care to off load the cpu, when running more steppers. 

I have integrated v. 0.22 of the library into our real time controllers for our stepper controlled devices at DTU (Danish Technical University) and have found a couple of problems. I have wrapped your interface into what we are using (with 3 steppers) and have performed some tests of the library. 

In short are the found problems: 
 1. The current speed is wrong when the stepper is moving
 2. The current speed is wrong when the stepper is stopped (Used by runForward() and keepRunning())
 3.  The current position is wrong when the stepper is moving
 4. The current acceleration is wrong when the stepper is moving
 5. The sign of the current speed is always positive no matter what direction the motor is running
 6. The sign of the current acceleration is always positive no matter what direction the motor is running
 7. The current acceleration > 0 when the motor is not running

All problems are documented in the attached file (WrappedExample_210228.ino). I have used v. 0.22 of the library. I have chosen not to separate the problems into different issues before you have looked into them. 

I know that some of the issues (problem1 and 3) are documented to be known, but they are essential to function correctly, if the library shall be used for real applications. 

Hope you will look into the tests/problems, so we can get a superior and full functioning stepper library out of it. Please do not hesitate to contact me, in case I can help. 

Best regards
Soren

[WrappedExample_210228.ino.txt](https://github.com/gin66/FastAccelStepper/files/6057655/WrappedExample_210228.ino.txt)

gin66, 2021-02-28:
Result is from execution on my hw:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Starting ...
Acceleration test
    Error - Wrong value - A1: 40.00  80.00  120.00  
  A2: 40.00  80.00  120.00  
  A3: 40.00  80.00  120.00  
  Error - Should be 0 - A4: 40.00  80.00  120.00  
Speed test
  Error - Vrong values - S1: 39.27  81.81  122.72  
  Error - Should be 0 - S2: 39.27  81.81  122.72  
Position test
  P1: -85.78  -140.06  482.75  
  P2: -25.13  25.13  33.24  
  P3: 25.13  -25.13  -25.13  
  P4: -25.13  25.13  25.13  
```

gin66, 2021-02-28:
BTW: Thanks for the detailed report and test program. I have added some more information, so I better understand what is going on:
```
Starting ...
>>> Set speed of stepper 1 to speed in rad/s=50.00, steps/s=50929
>>> Set acceleration of stepper 1 to steps/s^2=50930
>>> Set speed of stepper 2 to speed in rad/s=50.00, steps/s=50929
>>> Set acceleration of stepper 2 to steps/s^2=50930
>>> Set speed of stepper 3 to speed in rad/s=50.00, steps/s=50929
>>> Set acceleration of stepper 3 to steps/s^2=50930
Acceleration test
>>> Set speed of stepper 1 to speed in rad/s=120.00, steps/s=122230
>>> Set speed of stepper 2 to speed in rad/s=120.00, steps/s=122230
>>> Set speed of stepper 3 to speed in rad/s=120.00, steps/s=122230
>>> Set moveByAcceleration of stepper 1 to steps/s^2=40744
>>> Set moveByAcceleration of stepper 2 to steps/s^2=-81487
>>> Set moveByAcceleration of stepper 3 to steps/s^2=122231
    Error - Wrong value - A1: printAcceleration: 40.00  80.00  120.00  
  A2: printAcceleration: 40.00  80.00  120.00  
>>> Set moveByAcceleration of stepper 1 to steps/s^2=-40744
>>> Set moveByAcceleration of stepper 2 to steps/s^2=81487
>>> Set moveByAcceleration of stepper 3 to steps/s^2=-122231
  A3: printAcceleration: 40.00  80.00  120.00  
>>> StopMove for stepper 1
>>> StopMove for stepper 2
>>> StopMove for stepper 3
  Error - Should be 0 - A4: printAcceleration: 40.00  80.00  120.00  

Speed test
>>> Set speed of stepper 1 to speed in rad/s=40.00, steps/s=40743
>>> Set speed of stepper 2 to speed in rad/s=80.00, steps/s=81487
>>> Set speed of stepper 3 to speed in rad/s=120.00, steps/s=122230
  Error - Vrong values - S1: 
printSpeed: 40.91  81.81  122.72  
>>> StopMove for stepper 1
>>> StopMove for stepper 2
>>> StopMove for stepper 3
  Error - Should be 0 - S2: 
printSpeed: 40.91  81.81  122.72  

Position test
>>> Set position of stepper 1 to 25600
>>> Set position of stepper 2 to -25600
>>> Set position of stepper 3 to -25600
  P1: printPosition: >>> Stepper 1 is still running
-72.93  >>> Stepper 2 is still running
-131.32  >>> Stepper 3 is still running
474.33  
Expected 8*PI=25.13

>>> Set position of stepper 1 to -25600
>>> Set position of stepper 2 to 25600
>>> Set position of stepper 3 to 25600
  P2: printPosition: -25.13  25.13  >>> Stepper 3 is still running
31.58  
Expected 8*PI=-25.13 25.13 25.13

>>> Set position of stepper 1 to 25600
[WrappedExample.ino.txt](https://github.com/gin66/FastAccelStepper/files/6057913/WrappedExample.ino.txt)

>>> Set position of stepper 2 to -25600
>>> Set position of stepper 3 to -25600
  P3: printPosition: 25.13  -25.13  -25.13  
Expected 8*PI=25.13 -25.13 -25.13

>>> Set position of stepper 1 to -25600
>>> Set position of stepper 2 to 25600
>>> Set position of stepper 3 to 25600
  P4: printPosition: -25.13  25.13  25.13  
Expected 8*PI=-25.13 25.13 25.13

```

gin66, 2021-02-28:
Based on the above, here are my comments to your findings:

> The current speed is wrong when the stepper is moving

Relevant part from the log:
```
Speed test
>>> Set speed of stepper 1 to speed in rad/s=40.00, steps/s=40743
>>> Set speed of stepper 2 to speed in rad/s=80.00, steps/s=81487
>>> Set speed of stepper 3 to speed in rad/s=120.00, steps/s=122230
  Error - Vrong values - S1: 
printSpeed: 40.91  81.81  122.72  
```
Here the issue is the resolution from integer calculation in unit [us]. 40473 steps/s is 24.708 us. Retrieving the speed in us yields 24. Calculating back in rad/s gives 40.91. Similar the other values. Internally the esp32 uses 16 MHz and using `setSpeedInHz()` makes use of this higher resolution. The corresponding function is `getSpeedInTicks()`.

> The current speed is wrong when the stepper is stopped (Used by runForward() and keepRunning())
```
>>> StopMove for stepper 1
>>> StopMove for stepper 2
>>> StopMove for stepper 3
  Error - Should be 0 - S2: 
printSpeed: 40.91  81.81  122.72  
```
The maximum speed is still valid. I assume, you need a function similar to `int32_t getActualSpeed()`

BTW: `keepRunning()` is not needed, because `runForward()` and `runBackward()` implies `keepRunning()`.

> The current position is wrong when the stepper is moving

Here I do not get, what the problem is..

> The current acceleration is wrong when the stepper is moving

The relevant log is:
```
Acceleration test
>>> Set speed of stepper 1 to speed in rad/s=120.00, steps/s=122230
>>> Set speed of stepper 2 to speed in rad/s=120.00, steps/s=122230
>>> Set speed of stepper 3 to speed in rad/s=120.00, steps/s=122230
>>> Set moveByAcceleration of stepper 1 to steps/s^2=40744
>>> Set moveByAcceleration of stepper 2 to steps/s^2=-81487
>>> Set moveByAcceleration of stepper 3 to steps/s^2=122231
    Error - Wrong value - A1: printAcceleration: 40.00  80.00  120.00  
  A2: printAcceleration: 40.00  80.00  120.00  
>>> Set moveByAcceleration of stepper 1 to steps/s^2=-40744
>>> Set moveByAcceleration of stepper 2 to steps/s^2=81487
>>> Set moveByAcceleration of stepper 3 to steps/s^2=-122231
  A3: printAcceleration: 40.00  80.00  120.00  
```
Except the sign, the values are equal.

> The sign of the current speed is always positive no matter what direction the motor is running

Yes. The speed in context of the ramp generator is the maximum speed in both directions. getSpeedInUs() does not retrieve the current speed but this maximum speed. As this is applicable for both directions, the speed is a scalar and not a vector. What you need is `getActualSpeed()` or similar.

> The sign of the current acceleration is always positive no matter what direction the motor is running

True. The system is not intended to provide the "current" acceleration. It uses just one acceleration as scalar in both directions. The function moveByAcceleration() translates the sign into corresponding target direction and uses the absolute value as scalar for the acceleration. If the motor is coasting at maximum speed, then the actual 
 acceleration is 0. Reading back the acceleration value just retrieves the scalar acceleration.

> The current acceleration > 0 when the motor is not running

Yes, as said before.

My takeaway is:

- Needed feature is the possibility to retrieve the current direction of the stepper e.g. `getCurrentDirection()`with results: +1, 0, -1
- Needed feature is the possibility to retrieve the current direction of the acceleration e.g. `getHeadingTo()` with +1, 0, -1

Separation in two values, because the stepper can run forward (+1), but is decelerating (-1).

Any better alternatives ?

SorenBNielsen, 2021-03-02:
Thank you for your fast feed back, and I'm sorry for my late reply (Hmmm ... too busy these days). 

I guess that the problem is, that we are using the stepper driver for real time control. That is why it is necessary to be able to read the actual position, actual speed and actual acceleration at any time and get correct values for them. We have not seen this problem in other drivers, as they always return actual values for position, speed and acceleration. 

I am not sure that your suggested current direction and heading are necessary, as these shall be included as the sign of the actual values for position, speed and acceleration. 

Re. "The current speed is wrong when the stepper is moving": My mistake. I was not aware of the truncations, so you are absolutely right on this. 

Will it be possible to add functions for getting these actual values for position, speed and acceleration correctly from the driver ?

gin66, 2021-03-02:
For speed I have an idea, how to improve this and perhaps even get rid of `getCurrentSpeedAtQueueEnd()`. For acceleration adding the sign and even return zero while coasting, should be possible, too. For current position it is a bit more complicated, but for esp32 there is another possibility to make use of available pulse counters. With three motors, there are actually five pulse counters free.

SorenBNielsen, 2021-03-02:
It sounds great. 
We are really interested in using your driver principle, as the drivers we use pt. (AccelStepper and FlexyStepper) takes one of the CPU kernels on an ESP32 just to send the pulses to the steppers, where yours practically will free up half of the CPU resources. 

I am looking forward to your findings and will intensively perform practical tests of them when they are ready.

gin66, 2021-03-03:
Please try the latest, not yet official release. This adds the functions: `getCurrentSpeed() ` and `getCurrentAcceleration()`.
Any issues, please report accordingly. Just finished the implementation and have not done any test for these two functions. 

In addition the getCurrentPosition() is improved on esp32 to take the current ongoing command's issued steps in consideration. From my (limited) test, I see at max one count off from a second attached pulse counter.

SorenBNielsen, 2021-03-04:
I have made a few tests and found one problem in the getCurrentSpeedInUs(). 
Please see the attached program, which gives the following output: 
Starting ...
Speed test0
  S: inf  inf  inf    
  S: inf  inf  inf    
  S: inf  inf  inf    
  S: 0.54  -0.54  0.54    
  S: 0.54  -0.54  0.54    
  S: 0.54  -0.54  0.54    
  S: 0.54  -0.54  0.54    
  S: 0.63  -0.63  0.63    
  S: 0.70  -0.70  0.70 

It seems as the first 3 speed readings gives getCurrentSpeedInUs() = 0. 

[WrappedExample_210304.ino.txt](https://github.com/gin66/FastAccelStepper/files/6085945/WrappedExample_210304.ino.txt)

gin66, 2021-03-04:
`getCurrentSpeedInUs()` returns 0 for stopped motor. This special case needs to be taken care for in 'readCurrentSpeedInRadianPerSecond()'. 

Why it takes three loops until the speed is not zero, I need to check.

gin66, 2021-03-05:
Actually the implementation of getCurrentSpeedInUs() is broken due to a mistake in a condition. Good proof, that every feature should be tested....

This evening can upload a fixed version.

SorenBNielsen, 2021-03-05:
Yes, right :-). 
Agree, it will be excellent to have getCurrentSpeedInHz() and in MilliHz.
I will update the wrapper now and make some more tests this afternoon. 
Looking forward to your update.

gin66, 2021-03-05:
Done. Please check on your side. There is only `getCurrentSpeedInMilliHz()` for now

SorenBNielsen, 2021-03-05:
Thank's. I have included getCurrentSpeedInMilliHz() and used setSpeedInHz() in the attached file. 
Output: 
Starting ...
Set speed in rad/s (Hz): 40.00
Set speed in rad/s (Hz): -80.00
Set speed in rad/s (Hz): 120.00
Speed test after 0 s:
Current speed in rad/s (Hz): 0.00
Current speed in rad/s (Hz): -1.25
Current speed in rad/s (Hz): 1.63
Speed test after 3 s:
Current speed in rad/s (Hz): 40.07
Current speed in rad/s (Hz): -80.14
Current speed in rad/s (Hz): 120.83

It looks much better now, though there still are significant truncation errors. 

[WrappedExample_210305.ino.txt](https://github.com/gin66/FastAccelStepper/files/6092379/WrappedExample_210305.ino.txt)

Will continue with more tests.

gin66, 2021-03-05:
the example still does this division based in us instead of using `getCurrentSpeedInMilliHz()`:
```
void readCurrentSpeedInRadianPerSecond(int stepper, double & currentSpeedInRadPerSecond) {                                                                                                                                  
  switch (stepper) {                                                                                                                                                                                                        
    case 1: { currentSpeedInRadPerSecond = 1000000.0 / (stepper1->getCurrentSpeedInUs() * stepsPerRad); }; break;                                                                                                           
    case 2: { currentSpeedInRadPerSecond = 1000000.0 / (stepper2->getCurrentSpeedInUs() * stepsPerRad); }; break;                                                                                                           
    case 3: { currentSpeedInRadPerSecond = 1000000.0 / (stepper3->getCurrentSpeedInUs() * stepsPerRad); }; break;                                                                                                           
  }                                                                                                                                                                                                                         
} 
```

SorenBNielsen, 2021-03-05:
I have attached a test file. 

Problem1: The current speed is changed though the acceleration is 0.
Notice the following lines: 
  A: 40.0003  0.0000  0.0000    S: 119.9081  -120.8305  120.8305    P: 177.6512  -268.2704  298.4248    
  A: 0.0000  0.0000  0.0000    S: 119.9081  -120.8305  120.8305    P: 178.7448  -269.3719  299.5263    
  A: 0.0000  0.0000  0.0000    S: 119.9081  -120.8305  120.8305    P: 179.8375  -270.4735  300.6278    
  A: 0.0000  0.0000  0.0000    S: 120.8305  -120.8305  120.8305    P: 180.9381  -271.5740  301.7284    
The speed of stepper1 is changing though the acceleration of stepper1 is 0.

Problem 2: The current position of the stepper is moving though the current speed is 0. 
Notice the output lines: 
  A: -40.0003  0.0000  0.0000    S: 1.0142  0.0000  0.0000    P: 436.3829  -436.5636  437.0937    
  A: 0.0000  0.0000  0.0000    S: 0.0000  0.0000  0.0000    P: 436.3898  -436.5636  437.0937    
  A: 0.0000  0.0000  0.0000    S: 0.0000  0.0000  0.0000    P: 436.3937  -436.5636  437.0937    
  A: 0.0000  0.0000  0.0000    S: 0.0000  0.0000  0.0000    P: 436.3947  -436.5636  437.0937    
The position of stepper1 is changing though the speed of stepper1 is 0.

[WrappedExample_210305b.ino.txt](https://github.com/gin66/FastAccelStepper/files/6092638/WrappedExample_210305b.ino.txt)

gin66, 2021-03-05:
Yes, the current acceleration is based on the ramp state, which is related to queue end.
The speed is for speeds <4096 us the current command's speed and for slower speeds, actually from queue end.
The difference between queue end and current command is just a couple of ms.

Any other issue identified ?

SorenBNielsen, 2021-03-06:
I expect that it is ok to have a time deviation of a few ms on reading the current values as long as the sampling time in a control loops is bigger than 10 ms. 
I have to verify this on a practical system.

SorenBNielsen, 2021-03-06:
I can see a significant deviation in setting the max speed, please see the attached wrapper + test output. 

[WrappedExample_210306b.ino.txt](https://github.com/gin66/FastAccelStepper/files/6095311/WrappedExample_210306b.ino.txt)

Test output: 
Starting ...
Test1:
  Set max speed in rad/s (Hz): 120.00  Set max speed in (INT) mHz): 122230996  Read max speed in rad/s (Hz): 122.72
Test2:
  Set max speed in rad/s (Hz): 120.00  Set max speed in (INT) Hz): 122231  Read max speed in rad/s (Hz): 120.83
Test3:
  Set max speed in rad/s (Hz): 180.00  Set max speed in (INT) mHz): 183346494  Read max speed in rad/s (Hz): 187.00
  Set max speed in rad/s (Hz): 180.00  Set max speed in (INT) Hz): 183346  Read max speed in rad/s (Hz): 180.55
Test4:
  Set max speed in rad/s (Hz): 200.00  Set max speed in (INT) mHz): 203718327  Read max speed in rad/s (Hz): 180.55
  Set max speed in rad/s (Hz): 200.00  Set max speed in (INT) Hz): 203718  Read max speed in rad/s (Hz): 180.55

There is significant deviation between the set max speed and the read max speed. When setting the max speed to 200 rad/s is the read max speed 180.55 rad/s. 

The read max speed is the one reached when reaching a target speed and reading it with readCurrentSpeed.

gin66, 2021-03-06:
Thanks for highlighting this. The internal calculation is `250*16000000/speed_in_milli_hz*4`, because 16,000,000,000 does not fit into a 32bit. I have changed the calculation by use of another 32bit multiplication and division to improve the result. In addition, the result is rounded, too.

Please check the changed version. Should be fine now within the limits of 0.0625us granularity of the time base.

SorenBNielsen, 2021-03-07:
Running the same test again now gives the following output: 
Starting ...
Test1:
  Set max speed in rad/s (Hz): 120.00  Set max speed in (INT) mHz): 122230996  Read max speed in rad/s (Hz): 119.91
Test2:
  Set max speed in rad/s (Hz): 120.00  Set max speed in (INT) Hz): 122231  Read max speed in rad/s (Hz): 119.91
Test3:
  Set max speed in rad/s (Hz): 180.00  Set max speed in (INT) mHz): 183346494  Read max speed in rad/s (Hz): 180.55
  Set max speed in rad/s (Hz): 180.00  Set max speed in (INT) Hz): 183346  Read max speed in rad/s (Hz): 180.55
Test4:
  Set max speed in rad/s (Hz): 200.00  Set max speed in (INT) mHz): 203718327  Read max speed in rad/s (Hz): 180.55
  Set max speed in rad/s (Hz): 200.00  Set max speed in (INT) Hz): 203718  Read max speed in rad/s (Hz): 180.55

There are some things to notice here: 
The speed setting of 120 rad/s has become significant better and has now a 0.1% error. 
The speed setting of 180 rad/s is unchanged and still has a -0.3% error. 
The speed setting of 200 rad/s is unchanged and has a 10.8% error, which is unusable. It seems as there still are something wrong in the calculations of the speed for higher speeds. 

The 0.1% - 0.3% error is acceptable but still quite large. 
We use the faster speeds (i.e. 200 - 300 rad/s) when using the stepper driver with an O-drive driving an AC-motor with tachometer, so these speeds should be working correctly. 

I have executed the program with low speeds with the following output: 
Starting ...
Test1:
  Set max speed in rad/s (Hz): 0.01  Set max speed in (INT) mHz): 10186  Read max speed in rad/s (Hz): 0.010000
Test2:
  Set max speed in rad/s (Hz): 0.01  Set max speed in (INT) Hz): 10  Read max speed in rad/s (Hz): 0.009817
Test3:
  Set max speed in rad/s (Hz): 0.20  Set max speed in (INT) mHz): 203718  Read max speed in rad/s (Hz): 0.200000
  Set max speed in rad/s (Hz): 0.20  Set max speed in (INT) Hz): 204  Read max speed in rad/s (Hz): 0.200278
Test4:
  Set max speed in rad/s (Hz): 1.00  Set max speed in (INT) mHz): 1018592  Read max speed in rad/s (Hz): 0.999997
  Set max speed in rad/s (Hz): 1.00  Set max speed in (INT) Hz): 1019  Read max speed in rad/s (Hz): 1.000379

This seems to be ok for all speeds set with setSpeedInMilliHz(). 

Test2 using setSpeedInHz() has a deviation of approx. 2%, so it seems as there are more deviation when using setSpeedInHz(). 

I cannot see any gains from using setSpeedInHz() relative to setSpeedInMilliHz(), so I will suggest to remove it from the interface and always use setSpeedInMilliHz() for speed setting.

gin66, 2021-03-07:
Actually test4 gives the same result at test3, which indicates, that setting of the new speed value has not been executed.

200 rad/s translates to 203718 Steps/s, which is above the library limit of 200kSteps/s.
Please check the return value of `setSpeedInHz()` and `setSpeedInMilliHz()`. For 200rad/s it will indicate an error.

The 0.1%-0.3% error is surprisingly good. The tick value are with 120-200 rad/s around ~100. The tick has a granularity of 1. So the error can be expected with proper rounding up to 0.625% at fastest tick value of 80 aka 200 kHz.

SorenBNielsen, 2021-03-07:
Sure, I will check the return value from the setSpeed... (). And I see your point. 
Yes, right, there must be some errors when the speed setting is high. 
Will continue ...

gin66, 2021-03-07:
Not sure, how complicated it would be for implementation, but there could be a scheme to go for an average speed. This means, toggling between two slightly different speeds (differ by one tick, aka approx. 1%) with the average speed being the desired one. The change between those two speeds would be around 1ms.

Not sure, if it is worth the effort and if the constant speed switch is acceptable or not.

SorenBNielsen, 2021-03-08:
We can read the actual set speed so we do know what it is. 
Getting the "constant" speed exact in average is more a matter of principle. 
I do not have any use cases where it is necessary, so I do not think it is worth the effort. 

I will include the driver in one of our systems, and verify it on a real model. It will take some time to do this. I will close this issue and open a new if I find more problems.

closed 2021-03-08 by SorenBNielsen