gin66/FastAccelStepper #58: Jitter at start and end of negative value moves

digitalbird01, 2021-03-08:
Hi
I am trying to test FastAccelStepper for use on my Camera motion Control system. I am currently trying to port the system over to the ESP32 and am running TMC2209 drivers. When I run your basic test script on the system it works perfectly but when I set negative values to the move the stepper jitters for a second before and after making the move. Have you experienced this behaviour?. I also have questions regarding the following:

- I am also struggling with lack of documentation do you have an end user command list I can follow?
- Its great that the commands are not blocking but how can I test to see if the moves are complete before I issue another move?
Any help would be much appreciated

gin66, 2021-03-08:
For the negative jitter: I have not experienced this. The only thing coming up to my mind is, that enable/disable of the stepper driver  could cause a half step jump, if micro stepping is in use. This could be with auto enable mode being in use. If this is not what you mean, then my question is: If you refer to basic test script, do you mean StepperDemo ? If yes, could you please state the commands, which reveal the problem ? 

In regard to lack of documentation, there is none besides the `FastAccelStepper.h`, which implicitly defines the API and as such the end user commands. Perhaps I will look at some of the great tools out there, to make this more readable.

`stepper->isRunning()` is available for checking, that the last ramp has completed. `stepper->isMotorRunning()` is more low level and checks, if the stepper is still executing commands from the command queue.

digitalbird01, 2021-03-10:
Thank you
I found the problem with the jitter. Very strange. I am patching the ESP32 on to an Arduino Nano shield I'm using and
forgot to patch over the 3.3V pin to the ESP32 which is required by the driver. Problem solved however it is a wonder that the driver worked at all without this being connected! but it does seem to work in one direction.... Further to the latest issue I to would love an example script showing the simplest way of driving more than one stepper but admittedly haven't tried yet given the last problem.

gin66, 2021-03-12:
Having multiple steppers is straightforward: Just duplicate the configuration in setup() for more steppers and in loop() call move()/moveTo()/.... as needed. Those operations are not blocking, so the motors are all controlled independently

gin66, 2021-03-21:
stalled

closed 2021-03-21 by gin66