gin66/FastAccelStepper #59: Multiple Motor Example

Humanoidx, 2021-03-09:
Hi 
Thanks for the library!
(Im the one that made the high speed youtube video)

I cant figure out how to run multiple motors at the same time using different pins, speeds, directions accelerations.

Do you think you can do an example where a push button triggers the movement of multiple steppers each with different parameters and a seperate push button can initiate an emergency stop ?

Thanks!

gin66, 2021-03-10:
esp32 or avr ?

Actually StepperDemo does this. If two steppers are connected, then the following will run one stepper, then start the second stepper after 300ms. After 5s will stop stepper 1 and 1s later stop M2 immediately with position reset:
ˋM1 V100 a1000 w300 M2 V100 a1000 w5000 M1 S w1000 M2 X ˋ

Humanoidx, 2021-03-11:
@gin66 Thanks for the quick response. 

I am using an ESP32.

I tried looking through StepperDemo but without comments much of it is hard for me to understand.

The goal is simply to have a Foot Pedal Button trigger a movement with 2 steppers and an emergency stop button to stop it if needed.

This is the code ive adapted from your examples but i cant figure out how to only do it on the pedal press. Right now it does it on boot and somehow reboots when i press the pedal.

Thank you so muchh for your help

`#include "FastAccelStepper.h"

// As in StepperDemo for Motor 1 on ESP32
#define enablePinStepper1 25
#define dirPinStepper1 26
#define stepPinStepper1 27

// As in StepperDemo for Motor 1 on ESP32
#define enablePinStepper2 34
#define dirPinStepper2 35
#define stepPinStepper2 32

const int footPedalPin = 33;
const int relayOut = 13;
int footPedalState = 0;         // variable for reading the pushbutton status

FastAccelStepperEngine engine = FastAccelStepperEngine();  //What Does This Do?????????????????????
FastAccelStepper *stepper1 = NULL; //What Does This Do?????????????????????
FastAccelStepper *stepper2 = NULL; //What Does This Do?????????????????????

void setup() {
  pinMode(footPedalPin, INPUT_PULLDOWN);
  pinMode(relayOut, OUTPUT);
  engine.init(); //What Does This Do?????????????????????
  stepper1 = engine.stepperConnectToPin(stepPinStepper1); //What Does This Do?????????????????????
  stepper2 = engine.stepperConnectToPin(stepPinStepper2); //What Does This Do?????????????????????

    if (stepper1) {              //What Does If (stepper1) mean?
    stepper1->setDirectionPin(dirPinStepper1);
    stepper1->setEnablePin(enablePinStepper1);
    stepper1->setAutoEnable(true);
    stepper1->setDelayToEnable(50);
    stepper1->setDelayToDisable(50);

    stepper1->setSpeedInHz(4000);  // the parameter is us/step !!!
    stepper1->setAcceleration(2000);
    stepper1->move(-2490); //-2490

  }
  if (stepper2) { //What Does If (stepper2) mean?
    stepper2->setDirectionPin(dirPinStepper2);
    stepper2->setEnablePin(enablePinStepper2);
    stepper2->setAutoEnable(true);
    stepper2->setDelayToEnable(50);
    stepper2->setDelayToDisable(50);

    stepper2->setSpeedInHz(4000);  // the parameter is us/step !!!
    stepper2->setAcceleration(2000);
    stepper2->move(-2300);

  }
  
}

void loop(){
  footPedalState = digitalRead(footPedalPin);
if (footPedalState == HIGH){
    digitalWrite(relayOut, LOW);
    //engine.init();
  stepper1 = engine.stepperConnectToPin(stepPinStepper1);
  stepper2 = engine.stepperConnectToPin(stepPinStepper2);

    stepper1->setDirectionPin(dirPinStepper1);
    stepper1->setEnablePin(enablePinStepper1);
    stepper1->setAutoEnable(true);
    stepper1->setDelayToEnable(50);
    stepper1->setDelayToDisable(50);

    stepper1->setSpeedInHz(4000);  // the parameter is us/step !!!
    stepper1->setAcceleration(2000);

    stepper2->setDirectionPin(dirPinStepper2);
    stepper2->setEnablePin(enablePinStepper2);
    stepper2->setAutoEnable(true);
    stepper2->setDelayToEnable(50);
    stepper2->setDelayToDisable(50);

    stepper2->setSpeedInHz(4000);  // the parameter is us/step !!!
    stepper2->setAcceleration(2000);
    stepper2->move(-1600);

  
  }
else{
      digitalWrite(relayOut, HIGH);

}
}`

oaklandGman, 2021-03-12:
Few things to try:

Get rid of all that setup stuff in loop() ... you only need it once under setup()

Under loop you only want to poll your switch, control your relay and send move command to your motor(s). you could also test if the motor is already running,, use isRunning and only send move command if it is stopped.

I recommend using a non-blocking debounce on the input; search google for arduino debounce, look for the example that uses millis() instead of delay() ... without debounce your each time you press your foot pedal it's going to trigger the code multiple times.

You can find an explanation of the various functions and methods in the header files eg https://github.com/gin66/FastAccelStepper/blob/master/src/FastAccelStepper.h

gin66, 2021-03-12:
Thanks oaklandGman for clarification. Just one addition for these two invocations in loop:
```
stepper1 = engine.stepperConnectToPin(stepPinStepper1);
stepper2 = engine.stepperConnectToPin(stepPinStepper2);
```

If the stepperConnectToPin() gets a valid pin, it will normally return a valid FastAccelStepper* value. If the function is called again with the same pin, then the second invitation will return NULL. So the call in the loop will overwrite the stepper1 and stepper2 variables with NULL and the following access could cause a processor reset/guru meditation.

gin66, 2021-03-21:
stalled

closed 2021-03-21 by gin66