gin66/FastAccelStepper #61: attachInterrupt on avr

etempm, 2021-03-11:
I am using rotary optical encoder to check and compensate lost steps on arduino uno. Encoder code uses hardware  "attachInterrupt" and with fastaccellstepper motor hangs / no movement. If I comment the code line "attachInterrupt" it works as expected. Anything I need to know?

By the way very good library and thanks for your effort.

gin66, 2021-03-12:
The attachInterrupt() connects to which pin and what is in the ISR?

etempm, 2021-03-12:
Hardware interrupt as I said. Pins 2 and 3 on uno. And no ISR code.

gin66, 2021-03-12:
What are the parameters to attachInterrupt() ? The second parameter is the ISR function.
```
attachInterrupt(digitalPinToInterrupt(pin), ISR, mode) (recommended)
attachInterrupt(interrupt, ISR, mode) (not recommended)
attachInterrupt(pin, ISR, mode) (Not recommended. Additionally, this syntax only works on Arduino SAMD Boards, Uno WiFi Rev2, Due, and 101.)
```

etempm, 2021-03-12:
Like this,

#define encoder0PinA 2
#define encoder0PinB 3
------

pinMode(encoder0PinA, INPUT_PULLUP);
 pinMode(encoder0PinB, INPUT_PULLUP);
 
attachInterrupt(0, doEncoderA, CHANGE);
attachInterrupt(1, doEncoderB, CHANGE);

gin66, 2021-03-12:
Looks ok. How do the doEncoderA() and doEncoderB() functions look like ?

gin66, 2021-03-12:
btw: please check for [Rotary encoder issue](https://github.com/gin66/FastAccelStepper/issues/44)

etempm, 2021-03-15:
Read methods just some fast port read code asm.

void doEncoderA(){
  // look for a low-to-high on channel A
  if (ReadPinA() == HIGH) {
    // check channel B to see which way encoder is turning
    if (ReadPinB() == LOW) { 
      encoder0Pos = encoder0Pos + 1;         // CW
    }
    else {
      encoder0Pos = encoder0Pos - 1;         // CCW
    }
  }
  else   // must be a high-to-low edge on channel A                                       
  {
    // check channel B to see which way encoder is turning 
    if (ReadPinB() == HIGH) {   
      encoder0Pos = encoder0Pos + 1;          // CW
    }
    else {
      encoder0Pos = encoder0Pos - 1;          // CCW
    }
  }
  Serial.println (encoder0Pos, DEC);         
  // use for debugging - remember to comment out
}

byte ReadPinA()//PIN2
{
  //return digitalRead(2);
  byte rtx=LOW;
  asm (
  "in __tmp_reg__, __SREG__  \n"
  "cli                       \n"                     
  "ldi %0, 1                 \n" //high 
  "sbis %1, %2               \n" //skip next if pin high
  "clr %0                    \n" //low
  "out __SREG__, __tmp_reg__ \n"
  : "=r" (rtx) : "I" (_SFR_IO_ADDR(PIND)), "I" (PIND2)  
);

      return rtx;
}

byte ReadPinB()//PIN3
{
   //return digitalRead(3);
  byte rtx=LOW;
  asm (
  "in __tmp_reg__, __SREG__  \n"
  "cli                       \n"                     
  "ldi %0, 1                 \n" //high 
  "sbis %1, %2               \n" //skip next if pin high
  "clr %0                    \n" //low
  "out __SREG__, __tmp_reg__ \n"
  : "=r" (rtx) : "I" (_SFR_IO_ADDR(PIND)), "I" (PIND3)  
);

      return rtx;
}

void doEncoderB(){
  // look for a low-to-high on channel B
  if (digitalRead(encoder0PinB) == HIGH) {   
   // check channel A to see which way encoder is turning
    if (digitalRead(encoder0PinA) == HIGH) { 
      encoder0Pos = encoder0Pos + 1;         // CW
    }
    else {
      encoder0Pos = encoder0Pos - 1;         // CCW
    }
  }
  // Look for a high-to-low on channel B
  else {
    // check channel B to see which way encoder is turning 
    if (digitalRead(encoder0PinA) == LOW) {   
      encoder0Pos = encoder0Pos + 1;          // CW
    }
    else {
      encoder0Pos = encoder0Pos - 1;          // CCW
    }
  }
}

devrim-oguz, 2021-03-16:
Your Interrupt Service Routine (ISR) uses cli which disables the interrupts for the microprocessor. Which what the stepper library needs in order to work. It can't generate interrupts while your ISR is running.

etempm, 2021-03-16:
Yes... In asm, I fogot that. Thanks. I will test it. (Devrim bey teşekkür ederim.)

etempm, 2021-03-22:
Hi,
 I disabled ISR code. But same, no change. This is all code.

#include "FastAccelStepper.h"
#include "AVRStepperPins.h" // Only required for AVR controllers
//
#define encoder0PinA 2
#define encoder0PinB 3
volatile int encoder0Pos = 0;

#define dirPinStepper    5
//#define enablePinStepper 6
#define stepPinStepper   9

// If using an AVR device use the definitons provided in AVRStepperPins
//    stepPinStepper1A
//
// or even shorter (for 2560 the correct pin on the chosen timer is selected):
//    stepPinStepperA

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;
int yol=5066;
void setup() {

   engine.init();
   stepper = engine.stepperConnectToPin(stepPinStepper);
   if (stepper) {
      stepper->setDirectionPin(dirPinStepper);
      //stepper->setEnablePin(enablePinStepper);
      //stepper->setAutoEnable(true);

      stepper->setSpeedInHz(16000);       
      stepper->setAcceleration(8000);  
      stepper->moveTo(yol);
   }
}

void loop()
{

if (stepper->getCurrentPosition()==yol)
{
   stepper->moveTo(0);
}
else if (stepper->getCurrentPosition()==0)
{
   stepper->moveTo(yol);
}
}

-------------------------------

void doEncoderA(){
  // look for a low-to-high on channel A
  if (ReadPinA() == HIGH) {
    // check channel B to see which way encoder is turning
    if (ReadPinB() == LOW) { 
      encoder0Pos = encoder0Pos + 1;         // CW
    }
    else {
      encoder0Pos = encoder0Pos - 1;         // CCW
    }
  }
  else   // must be a high-to-low edge on channel A                                       
  {
    // check channel B to see which way encoder is turning 
    if (ReadPinB() == HIGH) {   
      encoder0Pos = encoder0Pos + 1;          // CW
    }
    else {
      encoder0Pos = encoder0Pos - 1;          // CCW
    }
  }
  Serial.println (encoder0Pos, DEC);         
  // use for debugging - remember to comment out
}

byte ReadPinA()//PIN2
{
  return digitalRead(2);
}

byte ReadPinB()//PIN3
{
   return digitalRead(3);
}

void doEncoderB(){
  // look for a low-to-high on channel B
  if (digitalRead(encoder0PinB) == HIGH) {   
   // check channel A to see which way encoder is turning
    if (digitalRead(encoder0PinA) == HIGH) { 
      encoder0Pos = encoder0Pos + 1;         // CW
    }
    else {
      encoder0Pos = encoder0Pos - 1;         // CCW
    }
  }
  // Look for a high-to-low on channel B
  else {
    // check channel B to see which way encoder is turning 
    if (digitalRead(encoder0PinA) == LOW) {   
      encoder0Pos = encoder0Pos + 1;          // CW
    }
    else {
      encoder0Pos = encoder0Pos - 1;          // CCW
    }
  }
}

gin66, 2021-03-28:
In the latest code, the encoder is not in use at all. Initially the stepper has been operated, if interrupt has not been attached to the pin. So please first check your setup, if the steppers are still operable. Or go back to the old code, which has worked.

gin66, 2021-04-11:
stale issue

closed 2021-04-11 by gin66