gin66/FastAccelStepper #66: addQueueEntry ?

digitalbird01, 2021-03-26:
Hi Again
I have know ported all my code over to FastAccelStepper on the ESP32 and its working great running a 4 axis camera system with A-B moves, Thank you very much for all the hard work.
I would know like to build more complex moves involving multiple keypoints and differing speeds and accelerations which flow together often without pause between key positions. I have grasped how to change  speed and acceleration on the fly and hope I am on the correct path using the 'addQueueEntry' command to stack my move requests but have been struggling to work out the syntax  for the command. 
stepper1->addQueueEntry(moveTo(Key6_position),true);  Clearly does not work!
Can you enlighten me?

gin66, 2021-03-26:
Happy to hear, that the lib works for you. addQueueEntry() is quite low level, so no acceleration/deceleration by the ramp generator. An example is e.g. RawAccess.ino. The key part is:

```
    struct stepper_command_s cmd = {
        .ticks = curr_ticks, .steps = steps, .count_up = direction};
    while (true) {
      int rc = stepper->addQueueEntry(&cmd);
      if (rc == AQE_OK) {
        break;
      }
      // adding a delay(1) causes problems
      delayMicroseconds(1000);
    }
```

It is advised to first fill the queue with second parameter false, and then followed by true after one or two queue entries have been added.

Just from your description I understand, you want to chain several move/moveTo requests. For this addQueueEntry() is not intended. If you like to stack such commands, then you need to build your own queue and issue the next move/moveTo as soon as isRunning() returns false.

gin66, 2021-03-28:
Hope the answer has helped. for the lib currently nothing to do

closed 2021-03-28 by gin66