gin66/FastAccelStepper #67: Deceleration when using runForward/runBackwards fails

ixil, 2021-03-27:
I'm using the latest release: 0.23

However, it doesn't seem to occur for every speed from one to another. 
e.g. From speed 1 to 0.1 nothing changes.
From 0 or other direction, to 0.1 it *does* change.
From 5 to 0.1 it changes to something in between.

I set the speed to slower value `setSpeedInUs`, and call `applySpeedAcceleration`, but I do not seem to actually decelerate.
The ramp state is set to `RAMP_STATE_DECELERATE`

Some logs that I am dumping (the `stat` and `evel 1 x` are commands sent over the serial, the speed x mm/s is after some gearing etc)

```
stat
 ExtruderTargetSpeed set to: 0.00mm/s
 Retract speed: 30.00mm/s
 Speed Factor:1.00 Heaters : 0 TEMP : 31.57 Extrudable: 1 Enable?: 1 speeds: INACTIVE_RAMP  Is running? 0 0us Target=0 STOP   IDLE  
evel 1 0.1
0.10
191114.06Sent us/step. 191114us/step 0curr us/step 
5232.99Sent mHz. 5232mHz 0curr mHz.
3057824Ticks.
updated speed to: 0.10
stat
 ExtruderTargetSpeed set to: 0.10mm/s
 Retract speed: 30.00mm/s
 Speed Factor:1.00 Heaters : 0 TEMP : 31.60 Extrudable: 1 Enable?: 1 speeds: ACTIVE_RAMP  Is running? 1 191114us nonstop RUN   COAST 
evel 1 1
1.00
19111.86Sent us/step. 19111us/step 191114curr us/step 
52325.41Sent mHz. 52324mHz 5232curr mHz.
305776Ticks.
updated speed to: 1.00
evel 1 0.1
0.10
191114.06Sent us/step. 191114us/step 19111curr us/step 
5232.99Sent mHz. 5232mHz 52324curr mHz.
3057824Ticks.
updated speed to: 0.10
stat
 ExtruderTargetSpeed set to: 0.10mm/s
 Retract speed: 30.00mm/s
 Speed Factor:1.00 Heaters : 0 TEMP : 31.65 Extrudable: 1 Enable?: 1 speeds: ACTIVE_RAMP  Is running? 1 19111us nonstop RUN   RED   
evel 1 5
5.00
3822.77Sent us/step. 3822us/step 19111curr us/step 
261625.04Sent mHz. 261640mHz 52324curr mHz.
61152Ticks.
updated speed to: 5.00
evel 1 1
1.00
19111.86Sent us/step. 19111us/step 4352curr us/step 
52325.41Sent mHz. 52324mHz 229776curr mHz.
305776Ticks.
updated speed to: 1.00
stat
 ExtruderTargetSpeed set to: 1.00mm/s
 Retract speed: 30.00mm/s
 Speed Factor:1.00 Heaters : 0 TEMP : 31.62 Extrudable: 1 Enable?: 1 speeds: ACTIVE_RAMP  Is running? 1 9760us nonstop RUN   RED   
evel 1 0.1
0.10
191114.06Sent us/step. 191114us/step 9760curr us/step 
5232.99Sent mHz. 5232mHz 102456curr mHz.
3057824Ticks.
updated speed to: 0.10
evel 1 2
2.00
9556.18Sent us/step. 9556us/step 9760curr us/step 
104650.32Sent mHz. 104644mHz 102456curr mHz.
152896Ticks.
updated speed to: 2.00

```

ixil, 2021-03-27:
I bisected it to commit e938cda . It has the (appears to) have the correct behavior prior to that (up to the tag 0.21.0)

gin66, 2021-03-27:
Thanks for the detailed info and bisect-ing down to a commit.
In order to repeat this, the acceleration value is needed.
Can you please provide that value ?

gin66, 2021-03-27:
Two more questions:
- the curr us/step value uses which lib-call ?
- how much time approx. between speed changes ? Or is it just longer than deceleration or acceleration would need ?

gin66, 2021-03-27:
don‘t bother. Seems I can repeat the issue

gin66, 2021-03-27:
The issue, which I have identified based on your description, is fixed in 0.23.1 (with an appropriate test case). Please check, if this solves your problem, too.

Background:
The calculation uses this formula s = 1/2 * a * t² = v² / 2a for the ramp.
Thus the speed at step s = 1 should be v = sqrt(2*a).
If the acceleration is high enough, then the calculated speed is faster than the setSpeed value.
Thus the deceleration would stay stuck with too high speed.
The fix is to detect this situation and just set the speed to the setSpeed value.

ixil, 2021-03-28:
Seems to work! Thanks for the fixes

ixil, 2021-03-28:
> Thanks for the detailed info and bisect-ing down to a commit.
>  The acceleration value is needed.
Whoops, 5243 s/s^2

```

closed 2021-03-28 by ixil