gin66/FastAccelStepper #69: I2C not working

Stipena, 2021-04-17:
I've connected SSD1306 OLED display via I2C to Arduino Nano but it just doesn't work when FastAccelStepper library is included. Also after first loop Serial Monitor stops printing `Serial.print("@");` and `Serial.println(stepper->getCurrentPosition());`, but stepper seems to work ok. Apparently FastAccelStepper library doesn't like to work with "Wire", "Adafruit_GFX" and Adafruit_SSD1306" libraries.
SSD1306 OLED is connected to SDA and SCL pins.
Code I tried:
```
#include <Wire.h>  //  Include I2C library
#include<Adafruit_GFX.h>   //  include Graphical Library
#include<Adafruit_SSD1306.h>  // Oled library

// Make a oled class
Adafruit_SSD1306 oled(128, 64, &Wire, -1);  //  182= Oled width, 64 = oled height, 4 is reset pin

#include "FastAccelStepper.h"

// As in StepperDemo for Motor 1 on AVR
#define dirPinStepper 5
#define enablePinStepper 6
#define stepPinStepper 9

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {
  Serial.begin(115200);
  oled.begin(SSD1306_SWITCHCAPVCC, 0x3C);  // SSD1306_SWITCHCAPPVCC is generate 3.3volt internaly,  0x3c is I2c slave(oled) address
  oled.clearDisplay();  // Celar display function
  Serial.println("Starting");
  Serial.flush();
  engine.init();
  stepper = engine.stepperConnectToPin(stepPinStepper);
  Serial.println("Starting");
  Serial.print("Stepper Pin:");
  Serial.println(stepPinStepper);
  Serial.flush();
  Serial.println((unsigned int)stepper);
  Serial.println((unsigned int)&engine);
  if (stepper) {
    stepper->setDirectionPin(dirPinStepper);
    stepper->setEnablePin(enablePinStepper);
    stepper->setAutoEnable(true);

    stepper->setSpeedInUs(500);  // the parameter is us/step !!!
    stepper->setAcceleration(100);
    // stepper->move(7000);
    stepper->runForward();
  } else {
    Serial.println("Stepper Not initialized!");
    delay(1000);
  }
  Serial.print("    F_CPU=");
  Serial.println(F_CPU);
  Serial.print("    TICKS_PER_S=");
  Serial.println(TICKS_PER_S);
  Serial.flush();
}

void loop() {
  delay(100);
  if (stepper) {
    if (stepper->isRunning()) {
      Serial.print("@");
      Serial.println(stepper->getCurrentPosition());
    }
  } else {
    Serial.println("Stepper died?");
    Serial.flush();
    delay(10000);
  }
  oled.setTextSize(1);    // first we have to decide font size
  oled.setTextColor(WHITE);     //we have to decide back ground color
  oled.setCursor(0, 0);   //then set print location or printing cursor
  oled.println(F("Hello, World!"));   // print conmmand
  oled.display();  // oled display command
}
```

gin66, 2021-04-17:
I assume, that perhaps the stack is overflowing or the oiled is not initialized. Could you please try something like this?

```
#include <Wire.h>  //  Include I2C library
#include<Adafruit_GFX.h>   //  include Graphical Library
#include<Adafruit_SSD1306.h>  // Oled library

// Make a oled class
Adafruit_SSD1306 oled(128, 64, &Wire, -1);  //  182= Oled width, 64 = oled height, 4 is reset pin

#include "FastAccelStepper.h"

// As in StepperDemo for Motor 1 on AVR
#define dirPinStepper 5
#define enablePinStepper 6
#define stepPinStepper 9

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {
  Serial.begin(115200);
  bool res = oled.begin(SSD1306_SWITCHCAPVCC, 0x3C);  // SSD1306_SWITCHCAPPVCC is generate 3.3volt internaly,  0x3c is I2c slave(oled) address

  if (!res) {
     Serial.println("Initialization error");
     for(;;) {}
  }
  oled.clearDisplay();  // Celar display function
  Serial.println("Starting");
  Serial.flush();
  engine.init();
  stepper = engine.stepperConnectToPin(stepPinStepper);
  Serial.println("Starting");
  Serial.print("Stepper Pin:");
  Serial.println(stepPinStepper);
  Serial.flush();
  Serial.println((unsigned int)stepper);
  Serial.println((unsigned int)&engine);
  if (stepper) {
    stepper->setDirectionPin(dirPinStepper);
    stepper->setEnablePin(enablePinStepper);
    stepper->setAutoEnable(true);

    stepper->setSpeedInUs(500);  // the parameter is us/step !!!
    stepper->setAcceleration(100);
    // stepper->move(7000);
    stepper->runForward();
  } else {
    Serial.println("Stepper Not initialized!");
    delay(1000);
  }
  Serial.print("    F_CPU=");
  Serial.println(F_CPU);
  Serial.print("    TICKS_PER_S=");
  Serial.println(TICKS_PER_S);
  Serial.flush();
}

void loop() {
  delay(100);
  if (stepper) {
    if (stepper->isRunning()) {
      Serial.print("@");
      Serial.println(stepper->getCurrentPosition());
    }
    else {
       Serial.println("Stepper stopped“);
     }
  } else {
    Serial.println("Stepper died?");
    Serial.flush();
    delay(10000);
  }
  oled.setTextSize(1);    // first we have to decide font size
  oled.setTextColor(WHITE);     //we have to decide back ground color
  oled.setCursor(0, 0);   //then set print location or printing cursor
  oled.println(F("Hello, World!"));   // print conmmand
  oled.display();  // oled display command
}
```

Stipena, 2021-04-17:
Got initialization error.

Stipena, 2021-04-18:
From google search it looks like I2C and interrupts aren't friends on Arduino.

gin66, 2021-04-20:
I have not analyzed in detail the OLED driver, but the first possible cause for an initialization error is coming from the malloc() call, trying to allocate 1kByte of RAM, which is a lot on a 2kB RAM device. So I assume, that FastAccelStepper mem usage (I assume 230 Bytes) + Arduino core + I2C + OLED driver does not fit in this device.

If it is really the malloc(), then the initialization error should vanish, if you reduce the size of the oled in pixels for a quick test.

sennalb, 2021-04-20:
Hi, 
I confirm that the standard library for oled is voracious of memory and essentially unusable in the atmega328 architectures

gin66, 2021-04-20:
ok. Then there is nothing to do in the FastAccelStepper library. So issue can be closed

closed 2021-04-20 by gin66