gin66/FastAccelStepper #70: Compiling error - VS Code - Platform IO - Teensy 3.6

FlipEngineering, 2021-04-25:
Hey gin66,
your library looks amazing. It would be great for our project, but I'm not a pro in this,... I'm running our code at the moment on a Teensy 3.6 and would like to use your lib, but I can't get it to work. 

If I understand right I need to create an alternative StepperISR for the MK66FX1M0 Chip on the Teensy 3.6.
Now I've tried that before for another library,... and I ended up getting another microcontroller. Now I'm to far down the road with the project. 

I know that TeensyStep is a library for teensy also using the same timers. [TeensyStep - Timer](https://github.com/luni64/TeensyStep/tree/master/src/timer/teensy3)
But I must admit this is a bit past my understanding. 
TeensyStep is nice, but doesn't work well for my case. 

The outlines of the project are TMC2130 drivers controlled by Teensy 3.6 via Ethernet Artnet/DMX controlling 25 steppers in sync to music for a circus performance.

Hope one can point me in the right direction.
Greetings
Flip

gin66, 2021-04-25:
SPI-Interface is not supported, which the TMC2130 needs for being configured. So the app would need to do this.

The MK66FX1M0 is too not supported. I had a look at the data sheet and perhaps the tpm module can be used. Apparently up to 16 steppers could be controlled with one interrupt per step and stepper. This means quite high interrupt load at high speeds. As it would be quite an effort to add support for this uC, there is no plan to extend the library.

devrim-oguz, 2021-05-04:
You need to use the TMC library to initiate the driver, and use this library after that to drive it.

gin66, 2021-05-08:
nothing to do here

closed 2021-05-08 by gin66