gin66/FastAccelStepper #71: Any plans to bring this to SAMD21 chips? (Arduino Nano 33 IoT)

theelims, 2021-05-18:
Hi,

your nice library is exactly what I'm looking for. Unfortunately for an Arduino Nano 33 IoT sporting an 32-bit ARM SAMD21 µC. Any plans to port your lib to this chip?

I was briefly looking through the sources and they are fairly complex. I'm not sure whether my coding skills are sufficient to implement that on my own. I would be willing to help, if I can understand what is happening on your code. 

Kind regards,
elims

gin66, 2021-05-19:
I have checked the datasheet [samd21](https://community.atmel.com/sites/default/files/forum_attachments/SAM-D21-Family-Datasheet-DS40001882B.pdf) and have found out, that this device has no mcpwm module or pulse counter like the esp32. But the timers (TCC) could be used, just like the avr solution. This means, the implementation from `StepperISR_avr.cpp` could be a good starting point. Unfortunately this means, for every generated step one interrupt has to be serviced. The implementation will require still a remarkable effort. So the question is, if you could switch over to an Arduino Nano/Mega or even an esp32 ?

theelims, 2021-05-19:
Maybe redesigning the board for ESP32 might indeed be the solution with the least effort. Any particular board you can recommend?

gin66, 2021-05-19:
The Arduino Nano 33 IoT has a small size. There will be no drop in replacement. If the form factor can change, then there are plenty of options.

closed 2021-06-08 by gin66