gin66/FastAccelStepper #72: How to stopMove without disabling steppers?

MahOraibi, 2021-05-19:
I really love this library as it does everything I need except for this small thing. I'm using it to move 2 steppers with a joystick however I couldn't find any function to stop moving without putting the motors in idle state or freewheeling state. This is really important because one of the motors is used for tilting and whenever I use stopMove() function the z axis falls down. Is there a way to stop movement without disabling the steppers?

gin66, 2021-05-19:
Apparently you are using `setAutoEnable(true)`. So you have two options:
1. `setAutoEnable(false)`: in this case, the stepper needs to be manually enabled
2. `setDelayToDisable(uint16_t delay_ms)` with a large number for delay_ms

gin66, 2021-06-08:
stale issue

closed 2021-06-08 by gin66