gin66/FastAccelStepper #73: FastAccelStepper for Teensy 4.1

MahOraibi, 2021-06-09:
Hello gin66,

I've tried using this library on teensy 4.1 but it doesn't seem to be compatible with it as it uses attachinterrupt functions which are different for this board (or at least this is what the debug section in the IDE shows). Is there a way to use this on teensy 4.1? because I find it really easy to implement and work with despite the more advanced examples which are complete jargon for a newbie like me, but I find it very easy for basic things like accel value and maxspeed.

gin66, 2021-06-09:
Hi, teensy4.1 uC is neither an esp32, nor an Atmel. It's an ARM Cortex. Consequently, this board is not supported. Sorry

gin66, 2021-06-27:
stale

closed 2021-06-27 by gin66