gin66/FastAccelStepper #75: Interrupt safety

Llaves, 2021-06-26:
There are quite a few places in the code where a small section of a routine is wrapped with noInterrupts/Interrupts to insure atomic operation. This renders these routines and anything that relies on them unsafe for use in an ISR. (Let's please try to avoid discussion of what an ISR should contain for another place.) Would there be any problems with instead wrapping these code sections with save/restore of the status register (SREG)? A quick look suggests that the Atmega328 and 2560 as well as the ESP32 all use the literal SREG for the status register, so this would not have to explicitly account for architecture.

gin66, 2021-06-27:
This means, the noInterrupt()/interrupt() macros shall be replaced with a reentrant version in the API-callable routines.

For avr:
```
  uint8_t sreg_save = SREG;
  cli();
```
and `SREG = sreg_save;`.

For esp32, I am not sure, if this is needed, because the api side routines shall have a lower task priority than the FastAccelStepper internal routines. I just kept the `noInterrupt()/interrupt()` for being on the safe side - e.g. one app lifting the priority to max before calling api routines…

Llaves, 2021-06-27:
That sounds like what I had in mind. I didn't realize that interrupt()/noInterrupt() were macros, though. 
It might be nice if the interrupt macro also saved the SREG and a restoreInterrupts() macro was provided in addition to interrupts(), but I'm sure someone would complain about the extra instruction and byte of storage. 

Do you plan to make these changes to the library, or is your reply just a clarification for anyone wanting to do this in their own branch?

gin66, 2021-06-27:
To me this is not a high priority topic aka bug. So I will do it, when I can squeeze it. I will keep this issue open till it’s done. The code change is not the problem. The main effort is for manual/semi-automatic tests.

So I would expect this to be implemented some day in July.

Llaves, 2021-06-28:
thanks. I'm not in a rush, but it's nice to know that an upcoming version will have this feature.

gin66, 2021-09-05:
Haven't forgotten this, but I am waiting for the due support.

gin66, 2021-09-11:
implemented in 0.23.3

closed 2021-09-11 by gin66