gin66/FastAccelStepper #77: How is Positioning Using a GearHead Handled Within FastAccelStepper

vepicurean, 2021-07-15:
Hi: 

I know this is question item. But these no place to put questions. I'm hoping someone has run into this before with FastAccelStepper with using a gearhead with a step motor and has some suggestions on how to handle it. 

We've been working with FastAccelStepper and reading the PulseCounter to count the pulses from 24599 to 0 through to 24599  (using register 7) microsteps works well in the way  we were using the step motor previously. 

The issue we've run into is when we added a 100:1 planetary gearhead for more power to manage the load on the step motor. And now our 1:1 positioning approach to convert micrrosteps to degrees from 0 to 259 degrees does not work for us. 

Adjusting the code using FastAccelStepper for the faster speed using the ticks was easy. 

But the positioning adjustments we've been at for a few days now and nothing we think of is accurate. 

Ideally we'd have just read the pulse counter and divide by a factor of 100 to reflect the fact the 100:1 gearhead shaft only moves 1/100 as far as the motor shaft. Because the pulse counter resets every 25600 steps to the zero value that logic can only apply for the first (just 1) revolution of the motor shaft (which is nothing as far as the displacement of the gearhead shaft is concerned).   

Does anyone have any suggestions on how to deal with this 100:1 gearhead positioning challenge using FastAccelStepper?

gin66, 2021-07-18:
Thanks for the question und use of FastAccelStepper. Nice to hear, that you can use it. 

Reading through your post, I wonder, if switching to the position reported from the driver could be a solution. This is a 32bit signed integer value and would fulfill your purpose. Either `getPositionAfterCommandsCompleted()` or `getCurrentPosition()`

Besides   for the gearbox - if you change direction - do you have an estimate, how many steps to perform until the 100:1 shaft starts rotating in opposite direction? Or is there a constant load, so that the gearbox slack is not occurring?

vepicurean, 2021-07-19:
Thanks. Greatly appreciate the input. 

We were able to use the `getCurrentPosition` function in our logic to create the virtual microstep positioning for the gearhead shaft.  Then combined with the speed of the gearhead, we were about to re-edit our previous logic. We just need to work with the timings more to get things as accurate as possible. 

Because we're just moving in once direction, the backlash from the gearhead is not too bad. We do notice that we need to do our times using a range of microsteps depending on the resolution of the gearhead. This 100:1 gearhead gives a virtual resolution of 25600 microsteps x 100 = 2,560,000 virtual microsteps in 360 degrees. It hard to say execute a specific event at virtual microstep position 1,920,000.  We have to use some logic to execute the target event at say 1,919,975 to 1,920,030.   

I hope this makes sense. 

Thanks for all the help. We're moving now and just a matter of some experiments to get the timing under the gearhead driven virtual positioning as exact as possible.

gin66, 2021-07-20:
As the ramp generation is relatively well predictable, the time to run to the position 1,919,975….1,920,030 can be calculated. Only wondering, if it is worth the effort for implementation, because polling the position at 1ms ready is not too costly on an esp32 and less complex

Anyway, happy to hear, that you have found a working solution

closed 2021-08-02 by gin66