gin66/FastAccelStepper #78: Seeeduino XIAO (SAMD21G18 Cortex-M0+)

courtens, 2021-07-19:
Is this library compatible with the [Seeeduino XIAO](https://www.seeedstudio.com/Seeeduino-XIAO-Arduino-Microcontroller-SAMD21-Cortex-M0+-p-4426.html)? 
[SAMD21G18 Cortex-M0+](https://www.microchip.com/wwwproducts/en/ATsamd21g18)

courtens, 2021-07-19:
I replaced `defined(ARDUINO_ARCH_AVR)` with `defined(ARDUINO_ARCH_AVR) || defined(ARDUINO_ARCH_SAMD) ` to add my Seeeduino. Now I am getting: 
````
error: 'MAX_STEPPER' was not declared in this scope
   FastAccelStepper* _stepper[MAX_STEPPER];

gin66, 2021-07-20:
From the datasheet the Seeduino Xiao could drive three steppers:

_Three 24-bit Timer/Counters for Control (TCC), with extended functions:
• Up to four compare channels with optional complementary output
• Generation of synchronized pulse width modulation (PWM) pattern across port pins_

But it is quite an effort to implement. If there are a couple of comments wanting this chip to be supported, then I will look into it. So I keep it open, for others to register their wish. A patch (MIT licensed) would be welcomed, too

courtens, 2021-07-20:
Thank you -- **very much appreciated**. 

I am still new to the github concept and I am not sure what this would entail: "_A patch (MIT licensed) would be welcomed, too_"; I plan on using it privately, and found ([url](https://github.com/Stan-Reifel/FlexyStepper)) to do the job for now. It would be nice if I could run it faster so I can use 1/8 or 1/16 steps.

clazarowitz, 2021-07-25:
You are in so much luck....I'm not sure what the seeduino xiao is, so I'm not sure if it is an atmel sam3x8e like the DUE, but I'm pretty close to forking/committing/pull-requesting Due support.  You're addition of ARDUINO_ARCH_SAM makes me wonder.

I didn't look too closely at the full line.  My code is using the PWM generator peripheral, not synchronous, to generate up to 8 channels...but not all pins are easily accessible, and some share functions with other pins you're likely to want.  I arbitrarily capped it at 6.  gin66 is correct, this was a non-trivial effort.  It probably would have been easier for him, and I suspect he will need to "fix" some of my code (or tell me what to fix).

I chose NOT to use the timers because I need them to read servo PWM signals.  It looks like those timers have a rise time capture, as well as a fall time capture, so no ISR delay on pulses that are synchronous and end within a few microseconds of one another.  I'm working pretty hard to get this committed tonight....as evident by the fact I just registered a github account (I havent had need/want to contribute/collaborate until very recently).  Anyhow, off to fork...

clazarowitz, 2021-07-25:
You can give my fork a shot and see if it works...I didn't look at the datasheet for the SAMD21G18, so I honestly have no idea.
https://github.com/clazarowitz/FastAccelStepper

clazarowitz, 2021-07-25:
So, I did just have a look at the data sheet for the SAMD21G18.  Sorry, my fork will definatly not work.  The M0 and M3 from atmel are just too different.  Looks like my #defines will need to be updated.  ARDUINO_ARCH_SAM is far too broad.

gin66, 2026-09-12:
samd51 is supported

closed 2026-09-12 by gin66