gin66/FastAccelStepper #85: How to use AS5048a degree to move stepper to position?

MahOraibi, 2021-09-13:
Hello gin66,

I have hard time trying to figure out how to make this work in an easy step and I hope you can help me with it.

Basically what I'm trying to do is moving the stepper to the specified degree without using relying on the stepper stepping feature. More of in case of misstepping or anything like it. I see everyone uses PID for this purpose but it's way over my ability so basically I'm trying to compare the current degree of the AS5048a and ask the motor to move continuously until the AS5048a says it reaches the same degree.

until now it somewhat works to a certain degree but it is terribly slow, that's because I'm moving it 1 step each loop, if I increase the number of steps I'll be in the risk of not getting to the degree for positions closer than the number of steps specified. 

Here's a snip it of the code:
```

#include <SPI.h>
/*PINS
   Arduino SPI pins
   MOSI = 11, MISO = 12, SCK = 13, CS = 10
   STM32 SPI pins
   MOSI = PA7, MISO = PA6, SCK = PA5, CS = PA4
*/
#include "FastAccelStepper.h"
#include <TMCStepper.h>
#include <TMCStepper_UTILITY.h>

#define dirPinStepper 6
#define enablePinStepper 5
#define stepPinStepper 7
#define SERIAL_PORT Serial1 // TMC2208/TMC2224 HardwareSerial port
#define DRIVER_ADDRESS 0b00 // TMC2209 Driver address according to MS1 and MS2
#define DRIVER_ADDRESS2 0b01 // TMC2209 Driver address according to MS1 and MS2
#define xJoy A0
#define yJoy A1
#define preset_btn 4
#define RSENSE 0.11f // Match to your driver
TMC2209Stepper fDriver(&SERIAL_PORT, RSENSE, DRIVER_ADDRESS);
TMC2209Stepper zDriver(&SERIAL_PORT, RSENSE, DRIVER_ADDRESS2);
bool btnFlag;
volatile long currentPos;
long distance;
FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

const byte CS_pin = 53; //Chip select pin for manual switching

uint16_t rawData = 0; //bits from the encoder (16 bits, but top 2 has to be discarded)
float degAngle = 0; //Angle in degrees
float startAngle = 0; //starting angle for reference
float correctedAngle = 0; //tared angle value (degAngle-startAngle)
float totalAngle = 0; //total accumulated angular displacement
float numberofTurns = 0;
float rpmCounter = 0; //counts the number of turns for a certain period
float revsPerMin = 0; //RPM
int quadrantNumber = 0;
int previousquadrantNumber = 0;
 int new_degree = 135;
uint16_t command = 0b1111111111111111; //read command (0xFFF)

float timer = 0; //timer for updating the LCD and sending the data through the serial port
float rpmTimer = 0; //timer for estimating the RPM

void setup()
{
  SPI.begin();
  Serial.begin(19200);
  Serial.println("AS5048A - Magnetic position encoder");
  pinMode(CS_pin, OUTPUT); //CS pin - output
  pinMode(preset_btn, INPUT);
  digitalWrite(preset_btn, LOW);
  //Checking the initial angle
  readRegister();
  startAngle = degAngle;
  SERIAL_PORT.begin(115200);      // HW UART drivers
  //fDriver.beginSerial(115200);
  fDriver.begin();                 //  SPI: Init CS pins and possible SW SPI pins
  // UART: Init SW UART (if selected) with default 115200 baudrate
  fDriver.toff(2);                 // Enables driver in software
  fDriver.rms_current(600);        // Set motor RMS current
  fDriver.microsteps(16);
  fDriver.en_spreadCycle(false);   // Toggle spreadCycle on TMC2208/2209/2224
  fDriver.pwm_autoscale(true);

  Serial.println("Starting");
  Serial.flush();
  engine.init();
  stepper = engine.stepperConnectToPin(stepPinStepper);
  if (stepper) {
    stepper->setDirectionPin(dirPinStepper);
    stepper->setEnablePin(enablePinStepper, true);
    stepper->enableOutputs();
    stepper->setAutoEnable(false);
    stepper->setSpeedInUs(200);  // the parameter is us/step !!!
    stepper->setAcceleration(4000);
  } else {
    Serial.println("Stepper Not initialized!");
    delay(1000);
  }
}

void loop()
{
  joystick ();
  callPreset();
  readRegister(); //read the position of the magnet
  correctAngle(); //normalize the previous reading
  checkQuadrant(); //check the direction of the rotation and calculate the final displacement
  
    if (millis() - timer > 250)
    {
      Serial.print("Turns: ");
      Serial.println(numberofTurns);

      Serial.print("Total Angle: ");
      Serial.println(totalAngle, 2);

      timer = millis();
    }

  if (millis() - rpmTimer > 15000) //check and calculate RPM every 15 sec
  {
    revsPerMin = 4 * rpmCounter; //60000/15000 = 4 (we assume the same speed for the whole minute)

    rpmCounter = 0; //reset
    rpmTimer = millis();
  }
}

void readRegister()
{
  SPI.beginTransaction(SPISettings(3000000, MSBFIRST, SPI_MODE1));

  //--sending the command
  digitalWrite(CS_pin, LOW);
  SPI.transfer16(command);
  digitalWrite(CS_pin, HIGH);

  delay(10);

  //--receiving the reading
  digitalWrite(CS_pin, LOW);
  rawData = SPI.transfer16(command);
  digitalWrite(CS_pin, HIGH);

  SPI.endTransaction();

  rawData = rawData & 0b0011111111111111; //removing the top 2 bits (PAR and EF)

  degAngle = (float)rawData / 16384.0 * 360.0; //16384 = 2^14, 360 = 360 degrees

}

void correctAngle()
{
  //recalculate angle
  correctedAngle = degAngle - startAngle; //this tares the position

  if (correctedAngle < 0) //if the calculated angle is negative, we need to "normalize" it
  {
    correctedAngle = correctedAngle + 360; //correction for negative numbers (i.e. -15 becomes +345)
  }
  else
  {
    //do nothing
  }
  //Serial.print("Corrected angle: ");
  //Serial.println(correctedAngle, 2); //print the corrected/tared angle
}

void checkQuadrant()
{
  /*
    //Quadrants:
    4  |  1
    ---|---
    3  |  2
  */

  //Quadrant 1
  if (correctedAngle >= 0 && correctedAngle <= 90)
  {
    quadrantNumber = 1;
  }

  //Quadrant 2
  if (correctedAngle > 90 && correctedAngle <= 180)
  {
    quadrantNumber = 2;
  }

  //Quadrant 3
  if (correctedAngle > 180 && correctedAngle <= 270)
  {
    quadrantNumber = 3;
  }

  //Quadrant 4
  if (correctedAngle > 270 && correctedAngle < 360)
  {
    quadrantNumber = 4;
  }
  //Serial.print("Quadrant: ");
  //Serial.println(quadrantNumber); //print our position "quadrant-wise"

  if (quadrantNumber != previousquadrantNumber) //if we changed quadrant
  {
    if (quadrantNumber == 1 && previousquadrantNumber == 4)
    {
      numberofTurns++; // 4 --> 1 transition: CW rotation
      rpmCounter++;
    }

    if (quadrantNumber == 4 && previousquadrantNumber == 1)
    {
      numberofTurns--; // 1 --> 4 transition: CCW rotation
      rpmCounter--;
    }
    //this could be done between every quadrants so one can count every 1/4th of transition

    previousquadrantNumber = quadrantNumber;  //update to the current quadrant

  }

  totalAngle = (numberofTurns * 360) + correctedAngle; //number of turns (+/-) plus the actual angle within the 0-360 range
}

void joystick () {

  if (analogRead(xJoy) > 600) {
      stepper->runForward();
  } else if (analogRead(xJoy) < 450) {
     stepper->runBackward();
  } else {
    stepper->stopMove();
  }

}

void callPreset() {
  
        if ( (int)totalAngle> new_degree) { // move counter-clocwise 
          stepper->move(-1);
          //  stepper->runBackward(); // works but when it reaches the location moves back few steps
          Serial.print("new position is");
          Serial.println(totalAngle);
         
        } else if((int)totalAngle< new_degree) { // move clocwise 
         stepper->move(10);
         // stepper->runForward(); // works but when it reaches the location moves back few steps
          Serial.print("new position is");
          Serial.println(totalAngle);
      
        } else{
          stepper->stopMove();
          }
}
```

gin66, 2021-09-13:
From what I understand from the code, `loop()` does not exactly this: _I'm trying to compare the current degree of the AS5048a and ask the motor to move continuously until the AS5048a says it reaches the same degree_

in `joystick()` the code asks the stepper to run forward or backward. But immediately afterwards in _callPreset()_ the stepper gets either command `move(-1)`, `move(10)` or even `stopMove()`. Then   `readRegister(); correctAngle(); checkQuadrant();` is executed and then in `joystick()` the stepper is asked again to run forward/backward or stop, which overwrites the `move()` command. Apparently the code shall run somehow to `new_degree = 135` !?

Anyway - as you say - the code is "single stepping" the motor and this is the reason, why the ramp generator is not able to increase speed to maximum, but rather has to stay at minimal speed.

Perhaps try this approach:
1. Via joystick determine a new target position/angle
2. From the target position/angle calculate the number of steps to run - assuming no lost steps/angle deviation
3. Ask the stepper to move those steps with `stepper->move(+/- steps)`
4. **Wait (!)** for the stepper to stop using `isRunning()`
5. Measure the angle
6. If the angle is identical to the target angle, then finished
7. Otherwise calculate the deviation and translate this in steps and go back to 3

A more fancy implementation could - while running - measure the angle, derive the deviation to the expected angle based on actual number of steps and calculate a prediction value, how many steps less/more for the **end** position. Just this correction should be intelligent enough to not add the corrected number of steps again and again and again on each measurement of the angle.

Besides, it is better to not call `move()` or `moveTo()` on every `loop()`

MahOraibi, 2021-09-13:
Thank you so very much for your prompt reply. This really means a lot. I'll try and change the code based on your suggestion but there are few points which I understood but maybe don't know how to practically apply but will give them a try tomorrow. However, your comments have raised few questions in my mind, can I do this using moveForward or Backward functions instead of move or move to? Is that what you meant when you said to not use moveto or move in every loop? The problem I had with moveforward and backward is it reaches 135 degree and then move backs few steps. 

The idea with joystick is to move the motor forward and backward based on the joystick and I'm using stopMove because otherwise it will keep moving after I leave the joystick. other buttons will be used to recall presets but I wanted to figure out how to recall the position before dealing with button press and so on. Do you think this is what causing all this issue?

gin66, 2021-09-13:
`runForward()` and `runBackward()` are intended to let the motor run for a while without taking care for the position too much. If `runForward()` and `runBackward()` is used, then the application would need to poll periodically the position and take action e.g. `stopMove()`, if a specific position is reached. Should be feasible, but perhaps not the best approach. I have meant to use `move()`.

In the code above, the `runForward()` and `runBackward()` has been constantly overwritten with `move()` and vice versa. If the joystick is left, so that the motor shall stop at the current position, then there is no use to read the angle and issue a move, as the current code does. And that is overwritten by `stopMove()` immediately afterwards. So perhaps once a while a step is generated with joystick left alone.

So better to use the joystick for target setting (angle) and use as a switch (to stop).

BTW: In the above code `joystick()` only calls `runForward()` or `runBackward()`, but actually new_degree should be modified (,too).

MahOraibi, 2021-09-18:
Hello again gin66,

I've been thinking about the actual code to do what you suggested but honestly I couldn't get my head around everything in your comment, and I felt embarrassed asking for clarification which implies that this whole thing is not meant for me. However, we all started at some point and I feel although I'm not awesome at math or coding but I still can learn. 

I'll try and repeat your suggestion just to make sure that I can translate it into coding, but first it's important to make it clear that at this point I'm not testing if the joystick moves at all or not, or if the encoder is reading the motor's new angle after the joystick is moved. This is a battle for another day. What I'm concerned with at present is finding (and testing) a way that when I start up the program which means the encoder is in 0 degree, it knows that and moves the motor regardless of the number of steps it knows it has to travel, it just moves until the encoder reads it reaches the goal angle (135 degree) and it stops. so here what I think you meant:

```
void callPreset() {
  long distance = new_degree - (int)totalAngle; //measuring the distance from the current position to 135 degree
        if ( (int)totalAngle> new_degree||(int)totalAngle< new_degree) { //if the current position is higher or lower than the final degree
          stepper->move(distance); // move the same number of steps measured
            Serial.print("distance is");
          Serial.print(distance);
          Serial.print("\tnew position is");
          Serial.println(totalAngle);
          
        }
        while (stepper->isRunning()){
  delay(10);
 }
}
```

this is the result of this code:
https://www.youtube.com/watch?v=Qq7-85rFmbk

gin66, 2021-09-19:
From your info I have collected:
* AS5048a outputs 16384 values/revolution and your code translates this into degrees (360°/revolution)
* Your stepper motor is a NEMA17 with 200 steps/revolution
* Your stepper driver is set to 16 microsteps/step

So the relation from AS5048A ang to position is:
```
ang/360 = position/3200
position = ang * 3200/360 = ang * 80 / 9 = ang * 8.888
```

So the target position for 135° is 1200 steps. In your code, you use steps = ang. So the stepper runs only 9/80 of the required steps. Your code will let the stepper run these movements:
```
At 0°:  135 steps to pos=135, ~15°
At 15°: 120 steps to pos=255, ~29°
At 29°: 106 steps to pos = 361, ±41°
At 41°: 94 steps to pos = 455, ±51°
and so on
```
And for each move the stepper will accelerate from and decelerate to speed = 0. So what is seen in the video fits to the analysis.

Using the relation from angle to position, your code becomes:
```
void callPreset() {
  float deltaAngle = (float)new_degree - measuredAngle;
  int32_t deltaSteps = (int32_t)(deltaAngle * 8.888);
  if (deltaSteps != 0) {
    stepper->move(deltaSteps); // move the same number of steps measured
    Serial.print("distance is");
    Serial.print(distance);
    Serial.print("\tnew position is");
    Serial.println(totalAngle);
    while (stepper->isRunning()){
      delay(10);
    }
  }
}
```

MahOraibi, 2021-09-19:
I'm really grateful. It finally works. I thought it had something to do with the calculation but I'd never thought of these formulas. 

there are 2 issues that came up with this solution though:
1- with this, I cannot go beyond 360 degree for some reason.
2- I tried applying the formula using variables (in case I needed to change microstepping or add a gearbox) but the result is not the same. This is what I used:

`stepsCal = 200*microstepping*gear/360`

it works but it overshoots by few steps after reaching the correct position.

other than that everything is working perfectly. Thanks again, we couldn't have pulled this off without you.

gin66, 2021-09-19:
Glad to be of help. The proposed approach is definitely not the best solution, but may work for you. If the position is overshooting, then the predicted amounts of steps is too high and the formula/factors may need some tweaking. Or there is some play in the whole mechanics, or….

Plenty of possibilities and plenty of ways to work on a better implementation. Starting with the question, why is it limited to 360°….

MahOraibi, 2021-09-19:
So what do you suggest as a smarter and better way to achieving this?

gin66, 2021-09-19:
The drawback of this solution is, that the motor comes to a full stop first - and if there is a position mismatch, need to add few more steps - more or less in single stepping.

Better would be to predict the deviation (if any) while running or at least shortly before the stop. But this needs a more complex algorithm and good understanding of timing relation between angle measurement and current stepper position. And in addition I would expect, that the float calculation is too slow for this purpose.

MahOraibi, 2021-09-20:
I see what you mean. This is why I initially thought of a way to make the motor move blindly like runForward or backward and stop when the new angle reaches the desired angle. But this in its own require a way to stop the program at this function until it reaches the desired angle without going into different loops to work, which is something I don't think can be done. I tried while loop for isRunning and maybe it's not an ideal solution considering if I need to alter another function while the motor is moving to a position especially for long distances with slower speeds

closed 2021-09-22 by MahOraibi