gin66/FastAccelStepper #88: I have a problem when I click runZ

quangbinhhvac, 2021-09-17:
Hi, gin66
I have a problem when I click runZ
If there is an impact, the running of the step is broken.

void runZ() {
  int microstep = 800;
  long x = a*microstep; //xung/s
  long y = a*10*microstep; //xung/s^2 tăng tốc trong 100 miliseconds
  long z = (b/10)*microstep; // số xung hành trình

  if (stepperZ) {
     stepperZ->setDirectionPin(dirPin1);
     stepperZ->setEnablePin(enablePin1);
     stepperZ->setAutoEnable(true);
     
     stepperZ->setSpeedInHz(x);       // steps/s
     stepperZ->setAcceleration(y);    // steps/s²
     
     stepperZ->move(z);
     
     kichZ();
     delay((c/10) * 1000); //trễ (c/10)*1000 miliseconds
     digitalWrite(kich, LOW);
     
     stepperZ->move(-z);
    }
  
}

gin66, 2021-09-17:
Could you please provide more info ?

Have you been able to drive the motor at all ?
Is the hw ok ? Is the pin assignment ok ?
What is the full code, because the initialization is not seen.
Can you add couple of print(), so it gets more clear, which path the program takes.

quangbinhhvac, 2021-09-17:
Thank you so very much for your prompt reply. I added a new issue in the first entry. Looking forward to helping

closed 2021-09-17 by gin66