gin66/FastAccelStepper #90: I have problems with speed and distance traveled

quangbinhhvac, 2021-09-17:
Hi gin66,

my distance is set to constant 2 laps. but when speed change, it is not wrong. 
and it also no longer holds true when i change the delay time so i have to add the function "delay(100);" I do not understand why

void runZ() {
  int microstep = 800;
  long x = a*microstep; //xung/s
  long y = a*10*microstep; //xung/s^2 tăng tốc trong 100 miliseconds
  long z = (b/10)*microstep; // số xung hành trình
  if (stepperZ) {
     stepperZ->setDirectionPin(dirPin1);
     stepperZ->setEnablePin(enablePin1);
     stepperZ->setAutoEnable(true);
     
     stepperZ->setSpeedInHz(x);       // steps/s
     stepperZ->setAcceleration(y);    // steps/s²
     stepperZ->moveTo(-z);
     delay(100);
     kichZ();
     delay((c/10) * 1000); //trễ (c/10)*1000 miliseconds
     digitalWrite(kich, LOW);
     stepperZ->moveTo(0);
  }
}

gin66, 2021-09-17:
the `move()` only starts the process, while the application in `loop()` is continue to run. So `move()` or `moveTo()` are NOT blocking. If there is the need to wait for the stepper running, then check for `isRunning()`

quangbinhhvac, 2021-09-17:
it mean:
1. problem 1:
microstep 800, step 1.8
I put moveTo(1600) 
run speed 900 RPM, it's so good
but run speed 40 RPM, it doesn't go to the location
2. problem 2:
if there is no function delay(100), it doesn't go to the location

gin66, 2021-09-17:
yes. Without delay(), the code executes moveTo(1600) and then immediately moveTo(0). So the stepper runs towards 0, where it already is. So will not run to 0.
You have to wait after moveTo(1600), that the stepper reaches that position. And only then issue moveTo(0)

gin66, 2021-09-17:
I see, there is no simple usage example, which shows simple running forward and backward.

quangbinhhvac, 2021-09-17:
moveTo(1600) and moveTo(0) it is overwritten.  it doesn't run in order

gin66, 2021-09-17:
That's what I mean: moveTo() is NOT blocking.

gin66, 2021-09-17:
Try this (hope it compiles):
```
void loop() {
     if (!stepper->isRunning()) {
         uint32_t pos = stepper->getCurrentPosition();
         stepper->moveTo(1600-pos);
     }
}
```

quangbinhhvac, 2021-09-17:
#include <Wire.h> //khai báo thư viện
#include <EEPROM.h> //khai báo thư viện
#include "FastAccelStepper.h"
#include "AVRStepperPins.h" // Only required for AVR controllers
FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepperZ = NULL;
#include <LiquidCrystal_I2C.h>     //Thư viện LiquidCrystal_I2C: https://github.com/ArduinoKIT/Luc-Nguyen/raw/master/LiquidCrystal_I2C-1.1.2.zip
LiquidCrystal_I2C lcd(0x27, 16, 2); //kích thước LCD
#define up A0
#define down A1
#define runn A2
#define CTHT A3
#define outputA 11
#define outputB 12
#define sw 13
#define stepPin1 9
#define dirPin1 8
#define enablePin1 A6
#define kich 10
int counter = 0;
int aState;
int aLastState;
double a = EEPROM.read(0); //Speed đọc giá trị tại ô nhớ 0 
double b = EEPROM.read(1); //H.Tr đọc giá trị tại ô nhớ 1
double c = EEPROM.read(2); //Tre đọc giá trị tại ô nhớ 2
//...............................................
void setup() {
  Serial.begin(9600);
  pinMode(up , INPUT_PULLUP);
  pinMode(down , INPUT_PULLUP);
  pinMode(runn , INPUT_PULLUP);
  pinMode(CTHT , INPUT_PULLUP);
  pinMode(outputA , INPUT);
  pinMode(outputB , INPUT);
  pinMode(sw , INPUT_PULLUP);
  pinMode(stepPin1, OUTPUT);
  pinMode(dirPin1, OUTPUT);
  pinMode(enablePin1, OUTPUT);
  pinMode(kich, OUTPUT);
  aLastState = digitalRead(outputA); //gán giá trị xung outputA cuối là aLastState
  lcd.init();
  lcd.backlight();
  lcd.setCursor(0, 0);
  lcd.print("Speed");
  lcd.setCursor(7, 0);
  lcd.print("H.Tr");
  lcd.setCursor(13, 0);
  lcd.print("Tre");
  engine.init();
  stepperZ = engine.stepperConnectToPin(stepPin1);
}
//...............................................
void loop() {
ok:
  lcd.setCursor(1, 1);
  lcd.print(a); //in ra LCD giá trị tốc độ
  lcd.setCursor(3, 1);
  lcd.print("   ");
  lcd.setCursor(7, 1);
  lcd.print(b/10); //in ra LCD giá trị hành trình
  lcd.setCursor(10, 1);
  lcd.print("  ");
  lcd.setCursor(13, 1);
  lcd.print(c/10); //in ra LCD giá trị trễ
  if ((a < 0) || (b < 0) || (c < 0))
  {
    lcd.setCursor(0, 0);
    lcd.print("                ");
    lcd.setCursor(0, 1);
    lcd.print("                ");
[... 167 lines cut ...]
  long y = a*10*microstep; //xung/s^2 tăng tốc trong 100 miliseconds
  long z = (b/10)*microstep; // số xung hành trình
  if (stepperZ) {
     stepperZ->setDirectionPin(dirPin1);
     stepperZ->setEnablePin(enablePin1);
     stepperZ->setAutoEnable(true);
     
     stepperZ->setSpeedInHz(x);       // steps/s
     stepperZ->setAcceleration(y);    // steps/s²
     stepperZ->moveTo(-z);
     delay(100);
     kichZ();
     delay((c/10) * 1000); //trễ (c/10)*1000 miliseconds
     digitalWrite(kich, LOW);
     stepperZ->moveTo(0);
  }
}
void kichZ() {
    digitalWrite(kich, HIGH);
}
void kichTAYHAN() {
    digitalWrite(kich, HIGH);
    delay(500);
    digitalWrite(kich, LOW);
}

quangbinhhvac, 2021-09-17:
help me make it move better

gin66, 2021-09-17:
Move this to` setup()`:
```
stepperZ->setDirectionPin(dirPin1);
stepperZ->setEnablePin(enablePin1);
stepperZ->setAutoEnable(true);
```

Besides, I have no idea, what your code shall do, but you can try this:
```
 delay(100);
 kichZ();
 while(stepperZ->isRunning()) { /* wait */ }
 digitalWrite(kich, LOW);
 stepperZ->moveTo(0);
```

quangbinhhvac, 2021-09-17:
thank you so much. I made it run as I wanted. Your library is amazing. Thank you for taking the time to help me

quangbinhhvac, 2021-09-19:
Hi gin66,
Have a good day
I have another problem that is:
when i call the function "stepperZ->setEnablePin(enablePin1, true);" i see the motor shaft is not held, so is enablePin1 high. and when I set it to low "digitalWrite(enablePin1, LOW);" then "moveTo" doesn't work

gin66, 2021-09-19:
Please check for `stepperZ->setAutoEnable(true);`
If set to false, enable is always on.
Alternatively, call `setDelayToDisable(uint16_t delay_ms);` with a suitable long delay for your needs.

quangbinhhvac, 2021-09-19:
i mean when i set stepperZ->setAutoEnable(true) the axis can be rotated manually i.e Enable is in HIGH it should be LOW

gin66, 2021-09-19:
not clear. If you have called `moveTo()` before, then the motor is running.Right ?

So I assume, the problem of „not held“ is after the move is completed, or before the very first move.

With `autoEnable()`, the driver will enable the motor with configured polarity - just before the move starts. After the move, the driver will disable the motor again. Any application’s `digitalWrite(enablePin1, LOW)` will be overwritten. As there is no call to `setDelayToDisable(uint16_t delay_ms);`, the driver will nearly immediately turn off the driver after a move is completed and the motor will not be held.

So apparently the auto enable feature may not work for you, so you could use this  sequence instead:
```
stepperZ->setAutoEnable(false);
stepperZ->enableOutputs();
```

quangbinhhvac, 2021-09-19:
i called  "stepperZ->setEnablePin(enablePin1, true);
                stepperZ->setDirectionPin(dirPin1, true);
                stepperZ->setAutoEnable(true);" at void setup. 
then the motor has not rotated, moveTo has not been called, I tried turning the motor shaft manually, but the motor doesn't hold

gin66, 2021-09-19:
If auto enable is set to true, then the driver will not touch the enable pin at all before any move()/....

quangbinhhvac, 2021-09-19:
when i call so enablePin1 will be high or low?

gin66, 2021-09-19:
If you call setEnablePin(), the following code is executed (some error detection code removed for better clarity):
```
void FastAccelStepper::setEnablePin(uint8_t enablePin,
                                    bool low_active_enables_stepper) {
  if (low_active_enables_stepper) {
    if (enablePin != PIN_UNDEFINED) {
      digitalWrite(enablePin, HIGH);
      pinMode(enablePin, OUTPUT);
    }
  } else {
    if (enablePin != PIN_UNDEFINED) {
      digitalWrite(enablePin, LOW);
      pinMode(enablePin, OUTPUT);
    }
  }
}
```

So after this call, the enable pin is set to OFF.
=> If the motor is fixed, then the polarity given is low_active_enables_stepper wrong. If the motor is free, all is ok.
=> If the motor is free, then it works as expected.

quangbinhhvac, 2021-09-19:
I don't understand it very well maybe because of my ability, I will try to find out more what you say

quangbinhhvac, 2021-09-19:
So when I call the function "stepperZ->setAutoEnable(true)" what will happen.
stepperZ->setEnablePin(enablePin1, true) and stepperZ->setAutoEnable(true) are superfluous

gin66, 2021-09-19:
The purpose of `setEnablePin()` and `setAutoEnable()` are not superfluous at all. 

`setEnablePin()` tells the library **HOW** to enable/disable the stepper motor output drivers.

`setAutoEnable()` tells the library **WHEN** to enable/disable the stepper motor output drivers. If false, then the stepper library does not enable/disable the output drivers automatically and the application programmer has to enable/disable explicitly with `enableOutputs()`/`disableOutputs()`. If true, then the stepper library will enable the output drivers automatically before a move and disable them after a move. The exact timing is defined by the enable/disable delays.

quangbinhhvac, 2021-09-20:
as far as i understand setAutoEnable will set Enalble to HIGH right before the first move and LOW at the start of the move, right after stopping it goes back to HIGH
I think so right?

gin66, 2021-09-20:
If low_active_enables_stepper=true  as parameter to setEnablePin() has been set, then  yes. The „right after“ can be changed by these calls:
```
  // In auto enable mode, the stepper is enabled before stepping and disabled
  // afterwards. The delay from stepper enabled till first step and from
  // last step to stepper disabled can be separately adjusted.
  // The delay from enable to first step is done in ticks and as such is limited
  // to MAX_ON_DELAY_TICKS, which translates approximately to 120ms for
  // esp32 and 60ms for avr at 16 MHz). The delay till disable is done in period
  // interrupt/task with 4 or 10 ms repetition rate and as such is with several
  // ms jitter.
  void setAutoEnable(bool auto_enable);
  int8_t setDelayToEnable(uint32_t delay_us);
  void setDelayToDisable(uint16_t delay_ms);
```

quangbinhhvac, 2021-09-20:
I want it to always be LOW is the engine operating level, what functions should I call?  i don't want it to turn off/on/off like with the autoenable function doing because it's out of sync with some function i call from stepper default library

gin66, 2021-09-20:
as I have written above:
```
stepperZ->setAutoEnable(false);
stepperZ->enableOutputs();
```

quangbinhhvac, 2021-09-20:
can you help me a command with push button at A0,A1 and step will rotate reversibly if button is pressed, release push button step will stop

quangbinhhvac, 2021-09-20:
Thank you very much. You helped me a lot, thanks

quangbinhhvac, 2021-09-20:
the commands are in the file FastAccelStepper.h, where can I find a detailed explanation of it

quangbinhhvac, 2021-09-21:
I find it not very detailed.  Is there any more detail?

quangbinhhvac, 2021-09-21:
Thank you very much.

closed 2021-09-21 by quangbinhhvac