gin66/FastAccelStepper #91: stepMotor->isMotorRunning() doesn't give correct results

alba-ado, 2021-09-24:
Hello, I have a program in which I call stepMotor->isMotorRunning() like so;

  int32_t holeCenterCorrection = HOLE_MIDDLE_DISTANCE;
  stepMotor->move(holeCenterCorrection);

  while(stepMotor->isMotorRunning()) {
      delay(50);
  }
  delay(500);
  SystemState.holeCentered = true;
  return true;

However, as I use this in a FreeRTOS task the function returns before the stepper motor stops. I tried blinking an LED when the isMotorRunning turns false, and it turns false before the motor stops. So I think there might be an issue in the implementation of this function with ESP32.

gin66, 2021-09-24:
On return from `move()`, it is not ensured, that the motor is already running. So `isMotorRunning()` can still return false for a while until it returns true. Better alternative is `isRunning()`, which returns true directly after `move()`

gin66, 2021-09-24:
Till now I have not observed, that during running, the function `isMotorRunning()` returns false.  And from implementation looks ok.

alba-ado, 2021-09-24:
> On return from `move()`, it is not ensured, that the motor is already running. So `isMotorRunning()` can still return false for a while until it returns true. Better alternative is `isRunning()`, which returns true directly after `move()`

Ohhh... Thank you very much! I didn't know that function existed. I will try using it.

closed 2021-09-24 by alba-ado