gin66/FastAccelStepper #92: Step pin polarity should be configurable

ttait-vantim, 2021-09-29:
Depending on how the stepper driver is wired (common anode or cathode) the polarity of the step pin needs to change. The documentation states the the step is taken on the low-to-high transition of the pulse, but in common anode mode the stepper driver is expecting to step when the cathode is driven low, lighting the optocoupler LED. 

One side effect of this with a common-anode driver configuration is that when stepping is paused, the step pin is left low continuously driving the optocoupler, which prevents the driver from switching into a low power idle state (if capable).

There may also be other side effects regarding the minimum high/low pulse widths.

gin66, 2021-09-29:
From application point of view, this requirement is understandable. On implementation side, this is very difficult to do. There are many places, where either the port pin is read or the polarity is hardcoded in writes/ to/reads from special io registers. For avr this is timer compare module, for esp32 this is mcpwm and pcnt module, for sam clazarowitz would need to look into it.

So in near term this will not be done from my side.

ttait-vantim, 2021-10-01:
OK, I understand. I appreciate the work you put into this, it works well on on my ESP32. I have added an inverter NPN for the now.

alba-ado, 2021-11-10:
> Depending on how the stepper driver is wired (common anode or cathode) the polarity of the step pin needs to change. The documentation states the the step is taken on the low-to-high transition of [...]

I have experienced the same issue, but really the fix is not easy as I also looked at it in the code. Just try to use common cathode configurations and be happy. If not, you need to include an inverter to the pulse pin, as other pins are already reversible.

closed 2022-01-27 by gin66